arendst/Tasmota #17301: DS3231 on ESP32

gfn256, 2022-12-07:
### PROBLEM DESCRIPTION
Used TasmoCompiler to compose ESP32 generic to use with DS3231 RTC
Added:
#define USE_I2C
#define USE_DS3231

Compiles status 0 (all good)

Configured module GPIO4/SCL GPIO5/SDA

Device found: {"I2CScan":"Device(s) found at 0x68"}

STATUS 4 shows "Drivers":"1,2,4,8,9,10,12,24,26,27,29,35,38,50,52,62"  (I believe 26 is the DS3231)

BUT no RTC showing!

i2cdriver doesn't show driver 26 at all

I have tried tasmota versions 12.0 and 12.1 (available on Tasmocompiler)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WEMOS Lolin32 lite
- [x] Tasmota binary firmware version number used: 12.1 and 12.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: TasmoCompiler
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:41:31.872 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E3A3B4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:41:31.909 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:00:27","StartupUTC":"2022-12-07T11:41:04","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-12-07T12:20:50","SaveCount":17}}
12:41:31.935 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2(TasmoCompiler-esp32generic)","BuildDateTime":"2022-12-07T11:07:01","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"392/699"}}
12:41:31.961 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000","00000000"]}}
12:41:31.993 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1167,"Free":1856,"Heap":165,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F8045CE","00040001","00000003","00000400","C0000981","00000080","00200000","14000008","00000000"],"Drivers":"1,2,4,8,9,10,12,24,26,27,29,35,38,50,52,62","Sensors":"1,2,127"}}
12:41:32.032 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E3A3B4-0948","IPAddress":"192.168.1.136","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"xxxxxxxxx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:41:32.058 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E3A3B4","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:41:32.075 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-12-07T11:41:32","Local":"2022-12-07T12:41:32","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:29","Sunset":"16:53"}}
12:41:32.104 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-12-07T12:41:32","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
12:41:32.115 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-12-07T12:41:32","Uptime":"0T00:00:28","UptimeSec":28,"Heap":164,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":32},"Wifi":{"AP":1,"SSId":"xxxxxxxxx","BSSId":"xxxxxxxxx","Channel":11,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Recognize DS3231 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
used same board with same DS3231 in Arduino and it works fine.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-12-07:
As I already told you in the other thread https://github.com/arendst/Tasmota/discussions/16765 your build does not include the driver for DS3231.

sfromis, 2022-12-07:
You need `#define USE_RTC_CHIPS` too