arendst/Tasmota #17312: ZBBridge: IKEA buttons (E1743) doesn't report after pressing repeatedly

xjvi, 2022-12-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After pressing the I of O button on de Ikea E1743 repeatedly (>5x) faster than once every 3 seconds, the ZBBridge is not reporting any more. Then you have to wait a few seconds before it works again. If you pressing slower than once every 3 seconds everything is ok.
I have tried this with several E1743 buttons with firmware version 2.2.010.
When I tried this with a Xiaomi WXKG11LM button I didn't have this problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-12-08:
Please paste logs. Supposing that you are using Zigbee2Tasmota and not ZHA, Tasmota reports everything from the Zigbee device. So it looks like it's a behavior from the device. But again, without logs it's just wild guessing and low confidence

xjvi, 2022-12-08:
This is the consoles log, after 5 times pressing the "I" button on the IKEA E1743 you see no response. Then wait a few seconds and again 5 times a response. 

00:42:37.429 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":84}}}
00:42:38.868 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":76}}}
00:42:40.679 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":71}}}
00:42:42.690 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":68}}}
00:42:44.439 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":66}}}
00:42:55.524 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":89}}}
00:42:56.882 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":79}}}
00:42:58.457 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":71}}}
00:42:59.978 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":76}}}
00:43:01.969 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":81}}}

xjvi, 2022-12-08:
I have used this same IKEA E1743 with Deconz without any problem

xjvi, 2022-12-09:
I'm using Zigbee2Tasmota on a Sonoff ZBBridge (not the Pro). Flashed and programmed a few days ago.
[image]

xjvi, 2022-12-13:
[user.log](https://github.com/arendst/Tasmota/files/10218517/user.log)
This is a complete log after pressing 6 times

xjvi, 2022-12-13:
Now with content!
[user.log](https://github.com/arendst/Tasmota/files/10218553/user.log)

xjvi, 2022-12-13:
STATUS 0 output here:

14:56:19.950 CMD: STATUS 0
14:56:19.955 MQT: stat/ST-ZBridge1/STATUS = {"Status":{"Module":75,"DeviceName":"ST-ZBridge1","FriendlyName":["ST-ZBridge1"],"Topic":"ST-ZBridge1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:56:19.960 MQT: stat/ST-ZBridge1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"4T14:17:25","StartupUTC":"2022-12-08T23:38:54","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2022-12-06T18:19:50","SaveCount":42,"SaveAddress":"F7000"}}
14:56:19.966 MQT: stat/ST-ZBridge1/STATUS2 = {"StatusFWR":{"Version":"12.2.0(zbbridge)","BuildDateTime":"2022-10-17T08:42:26","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"373/699"}}
14:56:19.971 MQT: stat/ST-ZBridge1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VCS-54-UNI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000080","40006082","00004000","00000000"]}}
14:56:19.980 MQT: stat/ST-ZBridge1/STATUS4 = {"StatusMEM":{"ProgramSize":714,"Free":1076,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","04000800","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50,59,62","Sensors":"1"}}
14:56:19.985 MQT: stat/ST-ZBridge1/STATUS5 = {"StatusNET":{"Hostname":"ST-ZBridge1","IPAddress":"192.168.1.180","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"A4:CF:12:DC:06:85","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:56:19.990 MQT: stat/ST-ZBridge1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DC0685","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:56:19.997 MQT: stat/ST-ZBridge1/STATUS7 = {"StatusTIM":{"UTC":"2022-12-13T13:56:19","Local":"2022-12-13T14:56:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:35","Sunset":"16:52"}}
14:56:20.001 MQT: stat/ST-ZBridge1/STATUS10 = {"StatusSNS":{"Time":"2022-12-13T14:56:19"}}
14:56:20.008 MQT: stat/ST-ZBridge1/STATUS11 = {"StatusSTS":{"Time":"2022-12-13T14:56:20","Uptime":"4T14:17:26","UptimeSec":397046,"Vcc":3.543,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"VCS-54-UNI","BSSId":"F2:9F:C2:F4:EE:8E","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}
15:01:02.430 MQT: tele/ST-ZBridge1/STATE = {"Time":"2022-12-13T15:01:02","Uptime":"4T14:22:08","UptimeSec":397328,"Vcc":3.552,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"VCS-54-UNI","BSSId":"F2:9F:C2:F4:EE:8E","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}

s-hadinger, 2022-12-13:
I don't understand. The log you send does contain 5 messages, one for each press.

Also from what you copied earlier:
```
00:42:37.429 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":84}}}
00:42:38.868 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":76}}}
00:42:40.679 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":71}}}
00:42:42.690 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":68}}}
00:42:44.439 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":66}}}
00:42:55.524 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":89}}}
00:42:56.882 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":79}}}
00:42:58.457 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":71}}}
00:42:59.978 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":76}}}
00:43:01.969 MQT: tele/ST-ZBridge1/TDK2/SENSOR = {"ZbReceived":{"TDK2":{"Device":"0xC0F9","Name":"TDK2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":81}}}
```

We can see a first series of 5 presses, then a second series of 5 presses. The timestamps do match.

The fact that the console dis not update can we caused by a network glitch.

xjvi, 2022-12-14:
During the test I press continuously without interruptions.
And there is always an interruption of a few seconds after every presses before messages come again.
I can do this a hundred times but the result is always the same. I can press 5x then he will not respond to the E1743 for a few seconds. But in the meantime, it still responds to the XDK1 (Xiaomi WXKG11LM).
If I press them simultaneously, I see the same effect.
It's not that he's dead for those few seconds, he's just not responding to the E1743????

What I will do, is buying a few complete new E1743's and see if there is any difference.

MattWestb, 2022-12-16:
If you is have not the latest firmware on your IKEA controller they is sending commands to Zigbee groups = broadcast.
In the underlying 154 network have logic for blocking brodcast storms and is blocking braodcast if sending more then 8 packages in 9 second its dropping the frames.
Its one foundation of the 154 network and we cant doing anything about it its only dropping the package.

Also is your system is working  in real Zigbee 3 mode is all the router doing "passive acks" and only sending one broadcast / command. If not Zigbee 3 mode the routers is sending all broadcast commands 3 times for being sure its being sent OK to all routers in the system.

ouinouin, 2022-12-16:
@xjvi to my rememberance, on some ikea buttons, pressig 6 times in a row the central button does a reset of the ikea button and it tries to reassoc, if you keep the pairing mode open ... just check that you are not plagued by this issue (another common issue on those buttons is the infinitre reboot loop while using some bad quality button batteries, the button reboots many times in a row then loses its config.

xjvi, 2022-12-16:
@ouinouin Thanks for your reply. But I use a IKEA E1743. My problem only occurs in combination with Zigbee2Tasmota and not with Deconz or Zigbee2MQTT. So we can rule out battery problems, apart from the fact that I have already replaced the batteries a number of times to rule this out.

[image]

xjvi, 2022-12-16:
@MattWestb The software version of the IKEA E1743  button is Version 2.2.010. As far as I know this is the latest version, if not then please enlighten me.

MattWestb, 2022-12-16:
@xjvi Nop the latest is 2.3.079 but then you is loosing the group binding for binding it  to Zigbee light groups and its have many fixes for battery draining and losing network settings.
If you is not binding to light groups you can upgrading it the the latest and if need it you shall staying on version 2.2.010 !!

Be aware all IKEA controllers is blocked for firmware downgrading and can only being done with SWD probe.

More information of IKEA OTA firmware and firmware flashing https://github.com/zigpy/zigpy/discussions/660.

Jason2866, 2022-12-18:
Closong since it is not a Tasmota issue

closed 2022-12-18 by Jason2866

xjvi, 2022-12-18:
Happy news, After updating the IKEA Button E1743 to version 2.3.079, the problem no longer occurred.
Thanks for all the help!

barbudor, 2022-12-18:
@xjvi basic question: how do you recognize this E1743 button on ikea's web site?

sfromis, 2022-12-18:
https://www.ikea.com/gb/en/p/tradfri-wireless-dimmer-smart-white-00468432/

If you look in Product Details, Other Documents, the PDF file says 
> Type: E1743 wireless ON/OFF button

barbudor, 2022-12-18:
Thanks @sfromis