arendst/Tasmota #17346: BLE: Mi Scale  wrong weight (missing decimal places)

michapr, 2022-12-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
In the reported weight are missing the decimal places - only the integer value will be reported (as sample 88 instead of 88.35)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32
- [x] Tasmota binary firmware version number used: 12.2.0.4(bluetooth)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web Flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:12:18.242 SRC: WebConsole from 192.168.178.83
10:12:18.244 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
10:12:18.267 SRC: Backlog
10:12:18.270 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
10:12:18.277 MQT: stat/tasmota_190F70/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
10:12:18.497 SRC: Backlog
10:12:18.500 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
10:12:18.515 MQT: stat/tasmota_190F70/RESULT = {"Module":{"1":"ESP32-DevKit"}}
10:12:18.713 SRC: Backlog
10:12:18.715 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:16:28.603 SRC: WebConsole from 192.168.178.83
10:16:28.606 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
10:16:28.613 MQT: stat/tasmota_190F70/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_190F70","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:16:28.655 MQT: stat/tasmota_190F70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"25T17:30:37","StartupUTC":"2022-11-15T15:45:51","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2022-11-03T16:56:11","SaveCount":31}}
10:16:28.697 MQT: stat/tasmota_190F70/STATUS2 = {"StatusFWR":{"Version":"12.2.0.4(bluetooth)","BuildDateTime":"2022-11-13T18:07:38","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"452/699"}}
10:16:28.716 MQT: stat/tasmota_190F70/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"192.168.178.200","LogPort":514,"SSId":["FRITZ!Box Fon WLAN 7170",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004002","00000000"]}}
10:16:28.751 MQT: stat/tasmota_190F70/STATUS4 = {"StatusMEM":{"ProgramSize":1593,"Free":1286,"Heap":66,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7C7","00148001","000000CF","010013C0","C0000981","00004080","00200000","5400082C","00000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,85","Sensors":"1,2,3,5,6,52,62,127"}}
10:16:28.794 MQT: stat/tasmota_190F70/STATUS5 = {"StatusNET":{"Hostname":"tasmota-190F70-3952","IPAddress":"192.168.178.107","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"78:E3:6D:19:0F:70","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:16:28.821 BRY: GC from 7472 to 4017 bytes, objects freed 9/40 (in 1 ms) - slots from 47/122 to 41/122
10:16:28.830 MQT: stat/tasmota_190F70/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.92","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_190F70","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:16:28.859 MQT: stat/tasmota_190F70/STATUS7 = {"StatusTIM":{"UTC":"2022-12-11T09:16:28","Local":"2022-12-11T10:16:28","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"16:52"}}
10:16:28.880 MQT: stat/tasmota_190F70/STATUS10 = {"StatusSNS":{"Time":"2022-12-11T10:16:28"}}
10:16:28.887 BRY: GC from 4512 to 3734 bytes, objects freed 3/40 (in 1 ms) - slots from 42/122 to 41/122
10:16:28.896 MQT: stat/tasmota_190F70/STATUS11 = {"StatusSTS":{"Time":"2022-12-11T10:16:28","Uptime":"25T17:30:37","UptimeSec":2223037,"Heap":68,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7170","BSSId":"DC:15:C8:B7:83:7D","Channel":6,"Mode":"11n","RSSI":100,"Signal":-16,"LinkCount":2,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Dec 10 23:43:24 tasmota-190F70-3952 ESP-M32: MI32: new MISCALEV1 at slot: 16
Dec 10 23:43:24 tasmota-190F70-3952 ESP-MQT: tele/tasmota_ble/MISCALEV1f842c1 = {"Time":"2022-12-10T23:43:23","mac":"c8478cf842c1","WeightUnit":"kg","Weight":88,"RSSI":-89}
Dec 10 23:43:24 tasmota-190F70-3952 ESP-MQT: tele/tasmota_ble/MISCALEV1f842c1 = {"Time":"2022-12-10T23:43:23","mac":"c8478cf842c1","WeightUnit":"kg","Weight":88,"RSSI":-77}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

use MISCALE ;)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

transfer the full received value with decimals

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

using the OpenMQTTGateway I receive:
Dec 10 23:43:15 192.168.178.207  - O_MQTT_GW-1 O_MQTT_GW-1 - - - Received json : {"id":"C8:47:8C:F8:42:C1","name":"MI SCALE2","manufacturerdata":"5701c8478cf842c1","rssi":-82,"distance":11.4949,"model":"XMTZC04HM","weight":88.35,"service":"a20645b2070b0d0b2a2f"}

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-12-11:
https://tasmota.github.io/docs/Commands/#weightres

michapr, 2022-12-11:
SOLVED ! , thanks for fast reply.

Maybe would be good idea to set it for weight by default to "2" ?

Thanks!

closed 2022-12-11 by Jason2866