arendst/Tasmota #17356: ZBBridge: IKEA Devices dont report back their state

stenub, 2022-12-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
IKEA Tradfri Devices (LED Driver, Outlet, LED Bulb) do not report their status back after pressing a button on a remote E1743.
I removed the LED Driver with zbleave and zbforget, added it again with zbpermitjoin 1 and pressing the reset button with a needle.
After that i removed the switch with zbleave and zbforget and added it again with zbpermitjoin 1 and 4 short binding button presses.

`00:53:51.687 CMD: zbpermitjoin 1
00:53:51.702 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
00:53:51.710 MQT: stat/zbbridgepro/RESULT = {"ZbPermitJoin":"Done"}
00:54:07.831 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xB4E3F9FFFE88ED9D","ShortAddr":"0xD9B8","ParentNetwork":"0x7292"}}
00:54:07.877 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xB4E3F9FFFE88ED9D","ShortAddr":"0xD9B8","ParentNetwork":"0x7292"}}
00:54:09.936 ZIG: Zigbee Devices Data saved in File System (504 bytes)
00:54:10.060 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xB4E3F9FFFE88ED9D","ShortAddr":"0xD9B8","ParentNetwork":"0x7292"}}
00:54:10.107 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xB4E3F9FFFE88ED9D","ShortAddr":"0xD9B8","ParentNetwork":"0x7292"}}
00:54:10.261 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0xB4E3F9FFFE88ED9D","ShortAddr":"0xD9B8","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
00:54:11.309 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
00:54:12.419 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","ModelId":"TRADFRI on/off switch","Manufacturer":"IKEA of Sweden","Endpoint":1,"LinkQuality":69}}}
00:54:13.567 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":33,"Device":"0xD9B8","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0820","DeviceVersion":1,"InClusters":["0x0000","0x0001","0x0003","0x0009","0x0020","0x1000","0xFC7C"],"OutClusters":["0x0003","0x0004","0x0006","0x0008","0x0019","0x0102","0x1000"]}}
00:54:15.196 ZIG: Zigbee Devices Data saved in File System (543 bytes)
00:54:15.556 ZIG: auto-bind `ZbBind {"Device":"0xD9B8","Endpoint":1,"Cluster":"0x0006"}`
00:54:16.513 MQT: tele/zbbridgepro/RESULT = {"ZbBind":{"Device":"0xD9B8","Status":0,"StatusMessage":"SUCCESS"}}
00:54:17.558 ZIG: auto-bind `ZbBind {"Device":"0xD9B8","Endpoint":1,"Cluster":"0x0008"}`
00:54:18.331 ZIG: Battery auto-probe `ZbSend {"Device":"0xD9B8","Read":{"BatteryPercentage":true,"BatteryVoltage":true}}`
00:54:18.346 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0003!01":"","IdentifyQuery":true,"Endpoint":1,"LinkQuality":0}}}
00:54:19.136 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0xB701,0AD76996635364FB4196630E05010F0067070080B115FF310080"}
00:54:19.162 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0x03C7,0A9D6A96635E61D55396630E05010F00AEFE00800F27FF380080"}
00:54:19.173 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0xE7F0,0A196C96630A64BA3F9663"}
00:54:19.181 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0x0258,0A6E2196630E573C219663"}
00:54:19.190 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0xBA66,0A9924966307FFFFFFFFFF0E01010F0002FEFFFF7201BA751D690506010F0001"}
00:54:19.201 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0xBCB1,0A896A9663834B705C96630E05010F005B0800802615FF360080"}
00:54:19.207 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0xB148,0AC55C966350FFFFFFFFFF0506010F0001"}
00:54:19.217 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0x7292,0A536C966348FFFFFFFFFF"}
00:54:19.224 MQT: stat/zbbridgepro/RESULT = {"ZbData":"ZbData 0xD9B8,0AAB6D96634864AB6D9663"}
00:54:19.236 ZIG: ZbData - 198 bytes written to File System
00:54:19.258 MQT: tele/zbbridgepro/RESULT = {"ZbBind":{"Device":"0xD9B8","Status":0,"StatusMessage":"SUCCESS"}}
00:54:19.465 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","BatteryVoltage":0,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":72}}}
00:54:19.604 ZIG: auto-bind `ZbBind {"Device":"0xD9B8","Endpoint":1,"Cluster":"0x0102"}`
00:54:20.822 MQT: tele/zbbridgepro/RESULT = {"ZbBind":{"Device":"0xD9B8","Status":0,"StatusMessage":"SUCCESS"}}
00:54:51.725 MQT: tele/zbbridgepro/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}`

Then i went on as the docs say:

- Add the LED driver to group 101:
`01:04:46.832 CMD: ZbSend {"device":"0x7292","Send":{"AddGroup":101}}
01:04:46.850 MQT: stat/zbbridgepro/RESULT = {"ZbSend":"Done"}
01:04:47.115 MQT: tele/zbbridgepro/7292/SENSOR = {"ZbReceived":{"0x7292":{"Device":"0x7292","0004?00":"006500","AddGroup":101,"AddGroupStatus":0,"AddGroupStatusMsg":"SUCCESS","Endpoint":1,"LinkQuality":72}}}
01:04:47.456 ZIG: Auto-responder: ZbSend {"Device":"0x7292","Cluster":"0x0006","Endpoint":1,"Response":"0006/0000":0}
01:04:47.473 MQT: tele/zbbridgepro/7292/SENSOR = {"ZbReceived":{"0x7292":{"Device":"0x7292","Cluster":6,"Read":[0],"ReadNames":{"Power":true},"Endpoint":1,"Group":101,"LinkQuality":0}}}
01:04:48.935 ZIG: Auto-responder: ZbSend {"Device":"0x7292","Cluster":"0x0008","Endpoint":1,"Response":"0008/0000":100}
01:04:48.953 MQT: tele/zbbridgepro/7292/SENSOR = {"ZbReceived":{"0x7292":{"Device":"0x7292","Cluster":8,"Read":[0],"ReadNames":{"Dimmer":true},"Endpoint":1,"Group":101,"LinkQuality":0}}}`

- Binding the E1743 remote to the group 101:
`01:08:25.549 CMD: ZbBind {"Device":"0xD9B8","ToGroup":101}
01:08:25.566 MQT: stat/zbbridgepro/RESULT = {"ZbBind":"Done"}`

Now i can switch on/off the light, dimming is also possible, but in the console i only see reports of the switch, but nothing regarding the led driver.

`zblisten` does not work as i have ZNP Firmware. `setoption85 1` is on, but it seems it has no effect on this problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-12-13:
Can you please add the logs when you use the button. What type of device is the led driver?

stenub, 2022-12-13:
here you have a zbstatus3, the exact model is icpshc24-10eu-il-1

```
23:51:33.802 CMD: zbstatus3 0x7292
23:51:33.816 MQT: stat/zbbridgepro/RESULT = {"ZbStatus3":[{"Device":"0x7292","IEEEAddr":"0x804B50FFFED2AE8E","ModelId":"TRADFRI Driver 10W","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":false,"LastSeen":168405,"LastSeenEpoch":1670803488,"LinkQuality":0}]}
```
What seems wrong is that Reachable is "false" because it can get switched on and off. zbping is also possible:

```
23:54:46.074 CMD: zbping 0x7292
23:54:46.085 MQT: stat/zbbridgepro/RESULT = {"ZbPing":"Done"}
23:54:46.431 MQT: tele/zbbridgepro/RESULT = {"ZbPing":{"Device":"0x7292","IEEEAddr":"0x804B50FFFED2AE8E"}}
```

As you requested, here are the logs when i am using the button:

```
00:02:08.400 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0006!01":"","Power":1,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:12.153 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0006!00":"","Power":0,"Endpoint":1,"Group":101,"LinkQuality":36}}}
00:02:19.556 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0006!01":"","Power":1,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:23.659 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0008!01":"01530000","DimmerMove":1,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:25.710 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0008!07":"","DimmerStop":true,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:27.813 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0008!05":"0053","DimmerMove":0,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:30.013 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0008!07":"","DimmerStop":true,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:31.865 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0006!01":"","Power":1,"Endpoint":1,"Group":101,"LinkQuality":32}}}
00:02:34.064 MQT: tele/zbbridgepro/D9B8/SENSOR = {"ZbReceived":{"0xD9B8":{"Device":"0xD9B8","0006!00":"","Power":0,"Endpoint":1,"Group":101,"LinkQuality":32}}}
```
Perhaps i should mention that all worked until i updated to 12.2.0 and the battery of the remote went empty and was replaced. After the replacement i had to pair the remote with the zbbridge pro again and the whole problem started.
At the same time a battery of another E1743 was empty. This remote controlled a tradfri wall outlet:

```
00:08:44.852 CMD: zbstatus3 0xB148
00:08:44.869 MQT: stat/zbbridgepro/RESULT = {"ZbStatus3":[{"Device":"0xB148","Name":"Tradfri-Steckdose-WoZi","IEEEAddr":"0x804B50FFFE103264","ModelId":"TRADFRI control outlet","Manufacturer":"IKEA of Sweden","Endpoints":[1,242],"Config":["O01"],"Power":1,"Reachable":true,"LastSeen":13123,"LastSeenEpoch":1670959801,"LinkQuality":72}]}
```

The wall outlet also does not report back anymore. Therefore the problem must be on the tasmota side, not on the LED driver or outlet side.
Perhaps the zbbridge pro is not "subscribed" to the device groups? I must admit that i do not fully understand the device group concept in tasmota, as the docs only describe device groups in the context of multiple tasmota devices in a group, but here we have multiple zigbee devices in a group (switch and driver respectively switch and outlet).

closed 2023-02-04 by github-actions[bot]

Roukie686868, 2023-03-05:
> ok, perhaps this information is useful for further troubleshooting. I did a complete factory reset of the zbbridge pro and deleted the zbfiles in filesystem (zbdata and zbv4). Then i joined the led driver. With `ZbSend {"Device":"LED-Küche","Send":{"Power":0}}` and `...{"Power":1}}` via console i can switch the LED stripe on and off. Everytime i do this, i get a response by the driver:
> 
> ```
> 02:05:54.396 CMD: ZbSend {"Device":"led-küche","Send":{"Power":1}}
> 02:05:54.402 RSL: RESULT = {"ZbSend":"Done"}
> 02:05:55.397 RSL: SENSOR = {"ZbReceived":{"0xF693":{"Device":"0xF693","Name":"LED-Küche","Power":1,"Endpoint":1,"LinkQuality":7}}}
> 02:06:01.848 CMD: ZbSend {"Device":"led-küche","Send":{"Power":0}}
> 02:06:01.855 RSL: RESULT = {"ZbSend":"Done"}
> 02:06:02.747 RSL: SENSOR = {"ZbReceived":{"0xF693":{"Device":"0xF693","Name":"LED-Küche","Power":0,"Endpoint":1,"LinkQuality":7}}}
> ```
> 
> equivalent thing for the switch. i can join it, and with every button press i get a response:
> 
> ```
> 02:07:36.320 RSL: SENSOR = {"ZbReceived":{"0x123E":{"Device":"0x123E","Name":"Switch-Küche","0006!01":"","Power":1,"Endpoint":1,"Group":101,"LinkQuality":0}}}
> 02:07:38.019 RSL: SENSOR = {"ZbReceived":{"0x123E":{"Device":"0x123E","Name":"Switch-Küche","0006!00":"","Power":0,"Endpoint":1,"Group":101,"LinkQuality":0}}}
> ```
> 
> but when i join the led-küche to the group 101
> 
> ```
> ZbSend {"device":"led-küche","Send":{"AddGroup":101}}
> ```
> 
> and bind the switch to this group:
> 
> ```
> ZbBind {"Device":"switch-küche","ToGroup":101,"Endpoint":1,"Cluster":6}
> ```
> 
> and press a button, i get only logs regarding the switch-küche, but nothing from led-küche. BUT when i `ZbSend {"Device":"led-küche","Send":{"Power":1}}` via console, the led reports back as it should!
> 
> As mentioned above it "feels" like the coordinator is not subscribed to messages destined to the group.
> 
> Has this zigbee group 101 something to do with devGroups enabled via SO85 or are those two completely separate worlds?

> here you have a zbstatus3, the exact model is icpshc24-10eu-il-1
>
> ```
> 23:51:33.802 CMD: zbstatus3 0x7292
> 23:51:33.816 MQT: stat/zbbridgepro/RESULT = [...]

Were yo able to fix this problem? I have exactly the same issue where I can switch the ikea led light on and off but I can't get a proper report back from the lamp it just reports back {"ZbReceived":{"0x34C1":{"Device":"0x34C1","Reachable":false}}}

Roukie686868, 2023-03-05:
Were yo able to fix this problem? I have exactly the same issue where I can switch the ikea led light on and off but I can't get a proper report back from the lamp it just reports back {"ZbReceived":{"0x34C1":{"Device":"0x34C1","Reachable":false}}}