arendst/Tasmota #17389: Zbbrigge Tuya commands issues with Tasmota 12.2.0

jenshenk, 2022-12-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After updating from 12.0.2 to 12.2.0 the tuya specific commands (as listed here https://github.com/arendst/Tasmota/discussions/11386) are not recognized anymore (error message command unknown). After downgrading to 12.0.2 everything works as expected again. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ X] Tasmota binary firmware version number used: 12.2.0
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: Seial
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
I'm sorry but I already downgraded again so I cannot provide this info with the version number where the error occured

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
See above

```
- [ x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade to Tasmota ZB Bridge Version 12.2.0
Try to send e.g. command ZbSend {"Device":"0x715A","Write":{"TuyaTempTarget":10.0}}
Get error message command unknown

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
ZBsend successful, with a status report of the zigbee device (in this case a thermostat on a radiator) that received the format

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jenshenk, 2022-12-14:
> Hi, sorry for the inconvenience. The support for Tuya has been made
> customizable. If you want to get back to the previous behavior, you need to
> copy tasmota/zigbee/legacy_tuya.zb in the file system.
>
> More details here:
> https://tasmota.github.io/docs/Zigbee/#advanced-topic-zigbee-device-plugin

Thanks for your response! Seems this zigbee tuya stuff is a deep rabbit
hole to get into :) and after following the link I totally get the
reasoning behind the change.
For now I'm happy that my heating works again, but gonna look into the
extra steps for being able to use newer versions when it's not freezing
outside anymore.

Jason2866, 2022-12-18:
Closing since answered. Reopen if there is still a problem

closed 2022-12-18 by Jason2866