arendst/Tasmota #17397: Sonoff ZB Bridge Exception 3

Mights001, 2022-12-14:
### PROBLEM DESCRIPTION
Hello, I have ZB Bridge from Sonoff with Zigbee 3.0. I have the latest version of Tasmota on it. Firmware for Zigbee too. The bridge starts again and again. Tried all versions of the firmware with Tasmota 12.3.0.
Always the same, exception 3 and reboot.

It works with Tasmota version 12.2.0. Maybe you found a solution. 

KR 
Mights 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: Tasmota 12.3.0 Firmware Zigbee ncp-uart-nsw_6.7.9_115200.ota
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool on Raspberry 3
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

``` {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:00:04.794 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E18328","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:00:04.796 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz","RestartReason":"Exception","Uptime":"0T00:00:06","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":101,"BCResetTime":"2022-12-14T18:50:44","SaveCount":113,"SaveAddress":"1000"}}
00:00:04.798 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.0(zbbridge)","BuildDateTime":"2022-12-14T12:57:52","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"394/699"}}
00:00:04.800 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TheMaterix",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000080","00006000","00004000","00000000"]}}
00:00:04.805 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":717,"Free":1072,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","04000800","00000080"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50,59,62","Sensors":"1"}}
00:00:04.807 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E18328-0808","IPAddress":"10.0.0.116","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"192.168.10.1","Mac":"C4:5B:BE:E1:83:28","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:00:04.809 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E18328","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:00:04.812 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:00:04","Local":"1970-01-01T00:00:04","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:00:04.814 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:04"}}
00:00:04.818 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:04","Uptime":"0T00:00:06","UptimeSec":6,"Vcc":3.516,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TheMaterix","BSSId":"08:55:31:BA:18:5B","Channel":9,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:05"}}}
00:00:04.825 RSL: STATUS12 = {"StatusSTK":{"Exception":3,"Reason":"Exception","EPC":["4000bf64","00000000","00000000"],"EXCVADDR":"402a5779","DEPC":"00000000","CallChain":["4021808d","40003176","4024d164","402a62e2","4024d529","4024c2da","40242d00","40101507","402a5779","40293270","4021c679","40255f50","402a62e2","40293270","4021c6a8","40255f50","40225db8","40225cb4","40255f50","402505f4","40222215","40293268","402226c6","40225db8","40225dc0","402226a0","40232094","40100df2","4023843d","4021bf5e","402394c7"]}}
00:00:05.049 RSL: INFO1 = {"Info1":{"Module":"Sonoff ZbBridge","Version":"12.3.0(zbbridge)","FallbackTopic":"cmnd/DVES_E18328_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:05.051 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-E18328-0808","IPAddress":"10.0.0.116"}}
00:00:05.057 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":3,"Reason":"Exception","EPC":["4000bf64","00000000","00000000"],"EXCVADDR":"402a5779","DEPC":"00000000","CallChain":["4021808d","40003176","4024d164","402a62e2","4024d529","4024c2da","40242d00","40101507","402a5779","40293270","4021c679","40255f50","402a62e2","40293270","4021c6a8","40255f50","40225db8","40225cb4","40255f50","402505f4","40222215","40293268","402226c6","40225db8","40225dc0","402226a0","40232094","40100df2","4023843d","4021bf5e","402394c7"]},"BootCount":102}}
19:26:04.436 QPC: Reset
19:26:07.404 RSL: STATE = {"Time":"2022-12-14T19:26:07","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.514,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TheMaterix","BSSId":"08:55:31:BA:18:5B","Channel":9,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:26:13.055 ZIG: Resetting EZSP device
19:26:14.278 RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 EZSP booted","RestartReason":"Power-on","Code":2}}
19:26:14.330 RSL: RESULT = {"ZbState":{"Status":55,"Version":"6.7.9.0","Protocol":8,"Stack":2}}
19:26:14.332 RSL: RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Same on 2 Devices 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-12-14:
Does your device have a file-system?

Mights001, 2022-12-14:
https://www.berrybase.at/sonoff-zb-bridge-zigbee-bridge

This is what a get.

gui28347, 2022-12-14:
I had the same problem. Workaround roll back to 12.2.0 all configurations still be saved and everything works fine.

In new [Sonoff ZIgbee Bridge Pro ](https://itead.cc/product/sonoff-zigbee-bridge-pro/)(newer version) 12.3.0 is working just fine. 

This issue only affects the first version of Sonoff Zigbee Bridge.

sfromis, 2022-12-15:
Until fixed, to OTA downgrade to an older binary, you may have to first do a `Module 18` to make it a plain ESP8266 device. Return to normal config after the downgrade.

s-hadinger, 2022-12-15:
Will work on it. Sorry for the inconvenience

moodyblue, 2022-12-15:
I have the same problem. By the way, EZSP is still at 6.7.9 (marked as test), is there a newer version ?

sfromis, 2022-12-15:
Yeah, everyone being quick to upgrade likely has the issue.

6.7.9 is the newest provided for Tasmota, and is really the currently recommended one, has been in use for a long time.

s-hadinger, 2022-12-15:
The fix is merged. It will be available very soon in the builds of the `development` branch. I have asked Theo if it's possible to make a new release with the fix. Sorry again, I thought I had tested it on ESP8266, but I must have missed something.

closed 2022-12-15 by s-hadinger

sfromis, 2022-12-15:
No more crash-cycling :ok_hand:

moodyblue, 2022-12-16:
> The fix is merged. It will be available very soon in the builds of the `development` branch. I have asked Theo if it's possible to make a new release with the fix. Sorry again, I thought I had tested it on ESP8266, but I must have missed something.

Thank you very much

PR #17405 (2022-12-15): Zigbee fix crash on ESP8266 #17397
## Description:

Fix crash on Zigbee ESP8266 based Tasmota.

**Related issue (if applicable):** fixes #17397

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.5
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CO [... cut]