arendst/Tasmota #17403: tasmota 12.3 does not reacts on touch buttons and gui buttons. gui slider also does not work - roller shutter

pawol, 2022-12-15:
### PROBLEM DESCRIPTION
tasmota 12.3 does not reacts on touch buttons and gui buttons. gui slider also does not work - roller shutter

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:12:57.874 CMD: Backlog Rule1; Rule2; Rule3
14:12:57.876 SRC: WebConsole from 192.168.2.8
14:12:57.879 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
14:12:57.952 SRC: Backlog
14:12:57.955 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
14:12:57.961 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:12:58.169 SRC: Backlog
14:12:58.172 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
14:12:58.181 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:12:58.420 SRC: Backlog
14:12:58.424 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
14:12:58.433 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:13:05.415 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-15T14:13:05","Uptime":"0T00:03:40","UptimeSec":220,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}
14:13:05.420 MQT: domoticz/in = {"idx":10,"nvalue":1,"svalue":"100","Battery":100,"RSSI":6}
14:13:05.424 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-15T14:13:05","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:13:24.576 CMD: Status 0
14:13:24.578 SRC: WebConsole from 192.168.2.8
14:13:24.581 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
14:13:24.587 MQT: stat/roleta_w_pokoju_lewa/STATUS = {"Status":{"Module":0,"DeviceName":"Roleta w pokoju (lewa)","FriendlyName":["Roleta w pokoju (lewa)","Roleta w pokoju (lewa)"],"Topic":"roleta_w_pokoju_lewa","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:13:24.593 MQT: stat/roleta_w_pokoju_lewa/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release-12.3.0/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:59","StartupUTC":"2022-12-15T13:09:25","Sleep":50,"CfgHolder":4617,"BootCount":221,"BCResetTime":"2020-05-20T10:16:11","SaveCount":7435,"SaveAddress":"F5000"}}
14:13:24.600 MQT: stat/roleta_w_pokoju_lewa/STATUS2 = {"StatusFWR":{"Version":"12.3.0(tasmota)","BuildDateTime":"2022-12-14T12:56:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"451/699"}}
14:13:24.608 MQT: stat/roleta_w_pokoju_lewa/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Network 4G",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["8000E009","2805C80001000600003C5A0A002800000000","40008080","00006000","00004000","00000000"]}}
14:13:24.622 MQT: stat/roleta_w_pokoju_lewa/STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
14:13:24.631 MQT: stat/roleta_w_pokoju_lewa/STATUS5 = {"StatusNET":{"Hostname":"roleta-w-pokoju-lewa-5252","IPAddress":"192.168.2.222","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:C9:34:84","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:13:24.638 MQT: stat/roleta_w_pokoju_lewa/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.3","MqttPort":1883,"MqttClientMask":"DVES_roleta_w_pokoju_lewa","MqttClient":"DVES_roleta_w_pokoju_lewa","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:13:24.649 MQT: stat/roleta_w_pokoju_lewa/STATUS7 = {"StatusTIM":{"UTC":"2022-12-15T13:13:24","Local":"2022-12-15T14:13:24","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"08:37","Sunset":"16:53"}}
14:13:24.658 MQT: stat/roleta_w_pokoju_lewa/STATUS10 = {"StatusSNS":{"Time":"2022-12-15T14:13:24","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}}
14:13:24.665 MQT: stat/roleta_w_pokoju_lewa/STATUS11 = {"StatusSTS":{"Time":"2022-12-15T14:13:24","Uptime":"0T00:03:59","UptimeSec":239,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}}
14:13:24.670 MQT: stat/roleta_w_pokoju_lewa/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":310,"Close":290,"50perc":61,"Delay":0,"Opt":"0000","Calib":[1370,483,3223,900,1000],"Mode":"1"}}}
14:13:25.412 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-15T14:13:25","Uptime":"0T00:04:00","UptimeSec":240,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}
14:13:25.418 MQT: domoticz/in = {"idx":10,"nvalue":1,"svalue":"100","Battery":100,"RSSI":6}
14:13:25.425 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-15T14:13:25","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
14:13:26.409 WIF: Sending Gratuitous ARP

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N08
00:00:00.050 CFG: Loaded from flash at F6, Count 7434
00:00:00.055 CFG: CR 451/699, Busy 0
[... 33 lines cut ...]
14:09:31.083 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"POWER1":"OFF"}
14:09:31.085 MQT: stat/roleta_w_pokoju_lewa/POWER1 = OFF
14:09:31.087 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"POWER2":"OFF"}
14:09:31.089 MQT: stat/roleta_w_pokoju_lewa/POWER2 = OFF
14:09:32.001 HTP: Console
14:09:34.379 APP: Boot Count 221
14:09:34.649 CFG: Saved to flash at F5, Count 7435, Bytes 4096
14:09:35.387 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-15T14:09:35","Uptime":"0T00:00:10","UptimeSec":10,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}
14:09:35.391 MQT: domoticz/in = {"idx":10,"nvalue":1,"svalue":"100","Battery":100,"RSSI":6}
14:09:35.395 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-15T14:09:35","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}

```

### TO REPRODUCE
just update onto 12.2 -> 12.3

### EXPECTED BEHAVIOUR
get touch buttons and gui buttons working 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-12-15:
You find all development builds for every commit [here](https://github.com/arendst/Tasmota-firmware)

pawol, 2022-12-16:
@Jason2866 
Tasmota 12.2.0.6 was former development version with roller shutter fix. At the moment development version is overwritteb by 12.3.0.1. So, I had to revert onto stable 11.1 where roller shutter worked okay.

arendst, 2022-12-16:
I've just installed v12.3.1 (as v12.3.0 is deleted) and see another issue where the relay doesn't stop and real position is way over stop position.

I think it's fair to say something is fishy with the latest PR #17380 @stefanbode

pawol, 2022-12-16:
In my case relays does not click when should after slider, buttons, etc.

Jason2866, 2022-12-16:
This firmwares https://github.com/arendst/Tasmota-firmware/tree/d81cb0e77499db7f965f8c28a04a576431b91adc are before the change with PR 17380. You can download and install via file upgrade.

pawol, 2022-12-16:
BTW. Hard to say that "release" are stable versions. From my POV 12.2 was unusable, The same with 12.3.

Jason2866, 2022-12-16:
As with every upgrade https://tasmota.github.io/docs/Upgrading/

JackTalisker, 2022-12-17:
My 4ch pro configured for 2 roller shutters seems to work well with gui sliders and buttons. 
12.3.1 installed

stefanbode, 2022-12-17:
Can you send some log files with the gui and slider issue. I tested the shutter code with 12.2 and there was no problem. Also quite strange at this point. Buttons are the same. I do not tested the buttons, but they call the openstop and closestop command and this is tested. I also see something that a later Develoment version is working.

pawol, 2022-12-17:
standard logs were delivered. I think there is no traces of buttons inside logs. How to collect logs you need?

stefanbode, 2022-12-18:
There are two things changes by designs. First shuttermode 1 with one relay for up and one relay for down now mandatory requires working interlock on the relay. There is an error message and also the Doku Has changed. Second change if you operate first relay on/off and second relay up/down you have to set shuttermode 2 manually. I tried to automate this to ensure Backward compatibility,but that may be broken.

stefanbode, 2022-12-18:
I see in you logs you have shuttermode 1. Therefore check the interlock. After this pls. Set SHUTTERRELAY1 1 again.

stefanbode, 2022-12-18:
@arendst : if this gets more common, it might be required to set a red banner on the webpage to inform users of the change.

arendst, 2022-12-18:
What change?

pawol, 2022-12-18:
So finally, what commands I have to set on my switch?

arendst, 2022-12-18:
What I understand from the latest replies you'll need to execute the following commands:
```
interlock 1,2
interlock 1
shutterrelay1 1
```
and see what happens ;-)

stefanbode, 2022-12-18:
I checked the current release version (`STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:16"`) with this settings:
`14:13:24.670 MQT: stat/roleta_w_pokoju_lewa/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":310,"Close":290,"50perc":61,"Delay":0,"Opt":"0000","Calib":[1370,483,3223,900,1000],"Mode":"1"}}}`

and everything is working no problems here.

stefanbode, 2022-12-18:
@arendst : If the interlock is missing on shuttermode 1 i send a ERROR message to the logs. Maybe it is required to pop a message on the WEBPAGE. I was not expecting people NOT setting interlock because this is part of the setup description since the beginning.

`15:55:36.906 SHT: ERROR: Shtr1 Relays are not in INTERLOCK. Pls read documentation. Shutter DISABLE. Fix and REBOOT`

arendst, 2022-12-18:
Thx. But I suppose the OP is using interlock from the beginning too didn't he?

slowfinger, 2022-12-18:
Hello 
same problem for me. My 5x sonoff dual (nonR2) are not working properly in 12.3.0 and in 12.3.1. 
all is set (like it was before) and checked again
```
SetOption80 1
ShutterRelay1 1
Backlog PulseTime1 0; PulseTime2 0
Backlog Interlock 1,2; Interlock ON
SHUTTERINVERT 1
ShutterEnableEndStopTime 1
```
Gui is non responsive, sliders are not working, sonoffs loose frequent connections and Gui is not shown at all. Now reverted back to 12.0.2 and all my errors are gone. In my desperation I already bought new sonoff dual R3 cause I thought after 5 years memory or flash is gone, now I have some spare at hand :-)

stefanbode, 2022-12-18:
Gentlemen please can you set Shutterrelay1 1 and send the log files. We have to find out what wrong there and need logs

pawol, 2022-12-18:
I shall try it late evening

pawol, 2022-12-18:
21:28:37.784 CMD: SHUTTERRELAY
21:28:37.786 SRC: WebConsole from 192.168.2.8
21:28:37.789 CMD: Grp 0, Cmd 'SHUTTERRELAY', Idx 1, Len 0, Pld -99, Data ''
21:28:37.795 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"ShutterRelay1":"1"}
21:28:39.388 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-18T21:28:39","Uptime":"0T00:01:30","UptimeSec":90,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:28:39.394 MQT: domoticz/in = {"idx":10,"nvalue":0,"svalue":"0","Battery":100,"RSSI":6}
21:28:39.402 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-18T21:28:39","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
21:28:49.393 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-18T21:28:49","Uptime":"0T00:01:40","UptimeSec":100,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:28:49.401 MQT: domoticz/in = {"idx":10,"nvalue":0,"svalue":"0","Battery":100,"RSSI":6}
21:28:49.406 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-18T21:28:49","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
21:28:53.788 WIF: Checking connection...
21:28:59.396 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-18T21:28:59","Uptime":"0T00:01:50","UptimeSec":110,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:28:59.403 MQT: domoticz/in = {"idx":10,"nvalue":0,"svalue":"0","Battery":100,"RSSI":6}
21:28:59.409 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-18T21:28:59","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
21:29:03.172 CMD: interlock
21:29:03.174 SRC: WebConsole from 192.168.2.8
21:29:03.177 CMD: Grp 0, Cmd 'INTERLOCK', Idx 1, Len 0, Pld -99, Data ''
21:29:03.182 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Interlock":"ON","Groups":"1,2"}
21:29:09.412 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-12-18T21:29:09","Uptime":"0T00:02:00","UptimeSec":120,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:29:09.422 MQT: domoticz/in = {"idx":10,"nvalue":0,"svalue":"0","Battery":100,"RSSI":6}
21:29:09.428 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-12-18T21:29:09","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
21:29:10.405 WIF: Sending Gratuitous ARP

stefanbode, 2022-12-19:
ok fine. I do see` domoticz/in`. This let me assume you use the buttons in domoticz to control your shutter. Can you please login to the webpage of the shutter and check if the bottons there are working. In this case we can narrow down the problem to "domoticz integration". Also please try in the console the "shutteropen" and "shutterclose". Just to double confirm the rest is working. @arendst : Any changes in 12.3 regarding domoticz that might cause the issue. I did not touch it.

stefanbode, 2022-12-19:
from the shutterdriver perspective all the above logs look ok. No reason why it should not work.

arendst, 2022-12-19:
Here you'll find a log where it doesn't work:
```
User click GUI button:
10:25:04.846 SRC: WebGui from 192.168.2.1
10:25:04.848 SRC: WebGui from 192.168.2.1
10:25:04.851 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 8, Pld -99, Data 'StopOpen'
10:25:04.854 SRC: Shutter
10:25:04.860 MQT: stat/t1uk3/RESULT = {"POWER1":"ON"}
10:25:04.864 MQT: stat/t1uk3/POWER1 = ON
10:25:04.868 SHT: Shtr1 Real 0, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
10:25:04.873 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
10:25:05.448 SHT: Shtr1 Real 10000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 0.55 [s], Freq 0, PWM 0, Tilt 0
10:25:05.453 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":5,"Direction":1,"Target":100,"Tilt":0}}
10:25:06.441 SHT: Shtr1 Real 30000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 1.55 [s], Freq 0, PWM 0, Tilt 0
10:25:06.446 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":15,"Direction":1,"Target":100,"Tilt":0}}
10:25:07.454 SHT: Shtr1 Real 51000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 2.60 [s], Freq 0, PWM 0, Tilt 0
10:25:07.459 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":26,"Direction":1,"Target":100,"Tilt":0}}
10:25:08.455 SHT: Shtr1 Real 71000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 3.60 [s], Freq 0, PWM 0, Tilt 0
10:25:08.459 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":36,"Direction":1,"Target":100,"Tilt":0}}
10:25:09.458 SHT: Shtr1 Real 91000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 4.60 [s], Freq 0, PWM 0, Tilt 0
10:25:09.464 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":46,"Direction":1,"Target":100,"Tilt":0}}
10:25:10.479 SHT: Shtr1 Real 111000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 5.60 [s], Freq 0, PWM 0, Tilt 0
10:25:10.484 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":56,"Direction":1,"Target":100,"Tilt":0}}
10:25:11.482 SHT: Shtr1 Real 131000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 6.60 [s], Freq 0, PWM 0, Tilt 0
10:25:11.487 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":66,"Direction":1,"Target":100,"Tilt":0}}
10:25:12.451 SHT: Shtr1 Real 150000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 7.55 [s], Freq 0, PWM 0, Tilt 0
10:25:12.456 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":75,"Direction":1,"Target":100,"Tilt":0}}
10:25:13.451 SHT: Shtr1 Real 170000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 8.55 [s], Freq 0, PWM 0, Tilt 0
10:25:13.456 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":85,"Direction":1,"Target":100,"Tilt":0}}
10:25:14.454 SHT: Shtr1 Real 191000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 9.60 [s], Freq 0, PWM 0, Tilt 0
10:25:14.459 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":96,"Direction":1,"Target":100,"Tilt":0}}
10:25:14.918 SHT: Stop 1 Mode 0 time 201
10:25:14.920 SRC: Shutter
10:25:14.926 MQT: stat/t1uk3/RESULT = {"POWER1":"OFF"}
10:25:14.931 MQT: stat/t1uk3/POWER1 = OFF
10:25:14.933 SHT: Tilt not match 0 -> 46
10:25:15.036 SRC: Shutter
10:25:15.041 MQT: stat/t1uk3/RESULT = {"POWER1":"ON"}
10:25:15.045 MQT: stat/t1uk3/POWER1 = ON
10:25:15.049 SHT: Shtr1 Real 200000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
10:25:15.054 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:25:15.485 SHT: Shtr1 Real 208000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 0.45 [s], Freq 0, PWM 0, Tilt 0
10:25:15.489 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:25:16.467 SHT: Shtr1 Real 228000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 1.45 [s], Freq 0, PWM 0, Tilt 0
10:25:16.472 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
Continues until...
10:26:50.477 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:26:51.476 SHT: Shtr1 Real 2128000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 96.45 [s], Freq 0, PWM 0, Tilt 0
10:26:51.481 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:26:52.478 SHT: Shtr1 Real 2148000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 97.45 [s], Freq 0, PWM 0, Tilt 0
10:26:52.482 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:26:53.479 SHT: Shtr1 Real 2168000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 98.45 [s], Freq 0, PWM 0, Tilt 0
10:26:53.484 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
User clicks some buttons...
10:26:54.081 SRC: WebGui from 192.168.2.1
10:26:54.083 SRC: WebGui from 192.168.2.1
10:26:54.086 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 8, Pld -99, Data 'StopOpen'
10:26:54.089 SHT: Stop 1: dir: 1
10:26:54.094 MQT: stat/t1uk3/RESULT = {"ShutterPosition":"Done"}
10:26:54.479 SHT: Shtr1 Real 2188000, Start 200000, Stop 2180000, Dir 1, Delay 0, Rtc 99.45 [s], Freq 0, PWM 0, Tilt 0
10:26:54.486 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:26:55.473 SRC: WebGui from 192.168.2.1
10:26:55.475 SRC: WebGui from 192.168.2.1
10:26:55.478 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 9, Pld -99, Data 'StopClose'
10:26:55.482 SHT: Stop 1: dir: 1
10:26:55.486 MQT: stat/t1uk3/RESULT = {"ShutterPosition":"Done"}
10:26:55.577 SHT: Shtr1 Real 2210000, Start 200000, Stop 2208000, Dir 1, Delay 0, Rtc 100.55 [s], Freq 0, PWM 0, Tilt 0
10:26:55.582 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
[... cut ...]
10:41:50.473 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:41:51.469 SHT: Shtr1 Real 20128000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 996.45 [s], Freq 0, PWM 0, Tilt 0
10:41:51.473 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
10:41:52.471 SHT: Shtr1 Real 20148000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 997.45 [s], Freq 0, PWM 0, Tilt 0
10:41:52.476 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
```

arendst, 2022-12-19:
I haven't debugged what happens above but I think you have an issue with using global variables (XdrvMailbox) on async functions.

I think you will have to rewrite your current commands by making a difference between real commands (from users like buttons or commandline) and your code executing async commands to move the shutter. You cannot rely on the XdrvMailbox variable staying stable when you code does not have sole ownership of these variables; they are being changed by rule processing or user commands when the shutter code is waiting for it's next time to process.

To solve this you will need to add specific shutter global vars to copy the initial XdrvMailbox variables once at execution time. 

The whole idea of XdrvMailbox vars is to transfer information between synchrone code parts. Async code can not rely on unchanged XdrvMailbox vars.

arendst, 2022-12-19:
In the meantime my virtual shutter is still rolling along:
```
10:58:50.481 SHT: Shtr1 Real 15961000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 798.10 [s], Freq 0, PWM 0, Tilt 0
10:58:50.486 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
10:58:51.483 SHT: Shtr1 Real 15981000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 799.10 [s], Freq 0, PWM 0, Tilt 0
10:58:51.488 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
10:58:52.435 SHT: Shtr1 Real 16000000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 800.05 [s], Freq 0, PWM 0, Tilt 0
10:58:52.439 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
10:58:53.441 SHT: Shtr1 Real 16020000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 801.05 [s], Freq 0, PWM 0, Tilt 0
10:58:53.446 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
10:58:54.444 SHT: Shtr1 Real 16040000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 802.05 [s], Freq 0, PWM 0, Tilt 0
10:58:54.449 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
10:58:55.445 SHT: Shtr1 Real 16060000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 803.05 [s], Freq 0, PWM 0, Tilt 0
10:58:55.449 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":0,"Tilt":0}}
```

arendst, 2022-12-19:
Done some more tests. It seems it always fails when using the GUI. It works when OK using the buttons on the device.

It fails with the tilt mismatch:
```
11:49:31.468 SHT: Shtr1 Real 195000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 9.80 [s], Freq 0, PWM 0, Tilt 0
11:49:31.473 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":98,"Direction":1,"Target":100,"Tilt":0}}
11:49:31.745 SHT: Stop 1 Mode 0 time 201
11:49:31.747 SRC: Shutter
11:49:31.752 MQT: stat/t1uk3/RESULT = {"POWER1":"OFF"}
11:49:31.756 MQT: stat/t1uk3/POWER1 = OFF
11:49:31.759 SHT: Tilt not match 0 -> 46
11:49:31.861 SRC: Shutter
11:49:31.866 MQT: stat/t1uk3/RESULT = {"POWER1":"ON"}
11:49:31.871 MQT: stat/t1uk3/POWER1 = ON
11:49:31.874 SHT: Shtr1 Real 200000, Start 200000, Stop 200000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
11:49:31.879 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":0}}
```

arendst, 2022-12-19:
And more tests with DEBUG_MORE enabled (and shortenend logging to catch as much logging a spossible):
```
12:08:05.563 HTP: Main Menu
12:08:11.154 SRC: WebGui from 192.168.2.1
12:08:11.156 SRC: WebGui from 192.168.2.1
12:08:11.159 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 9, Pld -99, Data 'StopClose'
12:08:11.163 SHT: Pos. payload <StopClose> (9), payload -99, idx 1 (1), src 7
12:08:11.166 SHT: Pos. payload <StopClose> (0), payload 0, idx 1 (1), src 7
12:08:11.168 SHT: lastsource 7:, real 200000, target 0, tiltreal: 0, tilttarget: 108, payload 0
12:08:11.170 SHT: Delay Start? var1 <99>=<>, max10s?
12:08:11.172 SRC: Shutter
12:08:11.173 SHT: Switched relay 2 by Shutter
12:08:11.179 MQT: stat/t1uk3/RESULT = {"POWER2":"ON"}
12:08:11.182 MQT: stat/t1uk3/POWER2 = ON
12:08:11.185 SHT: Shtr1 Real 200000, Start 200000, Stop 0, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
12:08:11.188 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0,"Tilt":0}}
12:08:11.252 SHT: 1 Tm 2(2), St 0, Ve 0/0/1000, Ru 0, Po 0/199000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.306 SHT: 1 Tm 3(0), St 0, Ve 0/0/1000, Ru 0, Po 0/198000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.357 SHT: 1 Tm 4(0), St 0, Ve 0/0/1000, Ru 0, Po 0/197000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.406 SHT: 1 Tm 5(0), St 0, Ve 0/0/1000, Ru 0, Po 0/196000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.457 SHT: 1 Tm 6(0), St 0, Ve 0/0/1000, Ru 0, Po 0/195000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.506 SHT: 1 Tm 7(0), St 0, Ve 0/0/1000, Ru 0, Po 0/194000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.528 SHT: Shtr1 Real 194000, Start 200000, Stop 0, Dir -1, Delay 0, Rtc 0.35 [s], Freq 0, PWM 0, Tilt 0
12:08:11.534 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":97,"Direction":-1,"Target":0,"Tilt":0}}
12:08:11.557 SHT: 1 Tm 8(0), St 0, Ve 0/0/1000, Ru 0, Po 0/193000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.606 SHT: 1 Tm 9(0), St 0, Ve 0/0/1000, Ru 0, Po 0/192000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.657 SHT: 1 Tm 10(0), St 0, Ve 0/0/1000, Ru 0, Po 0/191000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.706 SHT: 1 Tm 11(0), St 0, Ve 0/0/1000, Ru 0, Po 0/190000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.757 SHT: 1 Tm 12(0), St 0, Ve 0/0/1000, Ru 0, Po 0/189000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.806 SHT: 1 Tm 13(0), St 0, Ve 0/0/1000, Ru 0, Po 0/188000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.857 SHT: 1 Tm 14(0), St 0, Ve 0/0/1000, Ru 0, Po 0/187000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.906 SHT: 1 Tm 15(0), St 0, Ve 0/0/1000, Ru 0, Po 0/186000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:11.957 SHT: 1 Tm 16(0), St 0, Ve 0/0/1000, Ru 0, Po 0/185000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.007 SHT: 1 Tm 17(0), St 0, Ve 0/0/1000, Ru 0, Po 0/184000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.057 SHT: 1 Tm 18(0), St 0, Ve 0/0/1000, Ru 0, Po 0/183000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.106 SHT: 1 Tm 19(0), St 0, Ve 0/0/1000, Ru 0, Po 0/182000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.157 SHT: 1 Tm 20(0), St 0, Ve 0/0/1000, Ru 0, Po 0/181000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.206 SHT: 1 Tm 21(0), St 0, Ve 0/0/1000, Ru 0, Po 0/180000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.257 SHT: 1 Tm 22(0), St 0, Ve 0/0/1000, Ru 0, Po 0/179000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:12.306 SHT: 1 Tm 23(0), St 0, Ve 0/0/1000, Ru 0, Po 0/178000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.007 SHT: 1 Tm 37(0), St 0, Ve 0/0/1000, Ru 0, Po 0/164000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.056 SHT: 1 Tm 38(0), St 0, Ve 0/0/1000, Ru 0, Po 0/163000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.107 SHT: 1 Tm 39(0), St 0, Ve 0/0/1000, Ru 0, Po 0/162000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.156 SHT: 1 Tm 40(0), St 0, Ve 0/0/1000, Ru 0, Po 0/161000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.207 SHT: 1 Tm 41(0), St 0, Ve 0/0/1000, Ru 0, Po 0/160000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.256 SHT: 1 Tm 42(0), St 0, Ve 0/0/1000, Ru 0, Po 0/159000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.307 SHT: 1 Tm 43(0), St 0, Ve 0/0/1000, Ru 0, Po 0/158000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.356 SHT: 1 Tm 44(0), St 0, Ve 0/0/1000, Ru 0, Po 0/157000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.407 SHT: 1 Tm 45(0), St 0, Ve 0/0/1000, Ru 0, Po 0/156000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.456 SHT: 1 Tm 46(0), St 0, Ve 0/0/1000, Ru 0, Po 0/155000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.507 SHT: 1 Tm 47(0), St 0, Ve 0/0/1000, Ru 0, Po 0/154000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.520 SHT: Shtr1 Real 154000, Start 200000, Stop 0, Dir -1, Delay 0, Rtc 2.35 [s], Freq 0, PWM 0, Tilt 0
12:08:13.525 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":77,"Direction":-1,"Target":0,"Tilt":0}}
12:08:13.557 SHT: 1 Tm 48(0), St 0, Ve 0/0/1000, Ru 0, Po 0/153000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.669 SHT: 1 Tm 50(1), St 0, Ve 0/0/1000, Ru 0, Po 0/151000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.725 SHT: 1 Tm 51(0), St 0, Ve 0/0/1000, Ru 0, Po 0/150000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.775 SHT: 1 Tm 52(0), St 0, Ve 0/0/1000, Ru 0, Po 0/149000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.825 SHT: 1 Tm 53(0), St 0, Ve 0/0/1000, Ru 0, Po 0/148000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
12:08:13.875 SHT: 1 Tm 54(0), St 0, Ve 0/0/1000, Ru 0, Po 0/147000, NS 0, Ta 0, mV 0, D -1, Ti 0/108/0
[... cut ...]
12:08:26.482 SHT: 1 Tm 100(0), St 0, Ve 0/0/1000, Ru 0, Po 0/99000, NS 0, Ta 0, mV 0, D 1, Ti 0/108/1
12:08:26.496 SHT: Shtr1 Real 99000, Start 0, Stop 0, Dir 1, Delay 0, Rtc 5.00 [s], Freq 0, PWM 0, Tilt 0
12:08:26.500 MQT: stat/t1uk3/RESULT = {"Shutter1":{"Position":50,"Direction":1,"Target":0,"Tilt":0}}
12:08:26.532 SHT: 1 Tm 101(0), St 0, Ve 0/0/1000, Ru 0, Po 0/100000, NS 0, Ta 0, mV 0, D 1, Ti 0/108/1
12:08:26.582 SHT: 1 Tm 102(0), St 0, Ve 0/0/1000, Ru 0, Po 0/101000, NS 0, Ta 0, mV 0, D 1, Ti 0/108/1
```
You see the 108 is present from the beginning as tilttarget.

arendst, 2022-12-19:
The above log shows nothing different from a normal restart log.

It would be much more interresting to see why the device restarts. So logging BEFORE the restart. Try to setup a syslog server and start logging with ``syslog 3`` or enable ``mqttlog 3`` and fish the log from your MQTT server.

pawol, 2022-12-19:
Not possible to set up external syslog server

pawol, 2022-12-19:
mqtt logs there
https://file.io/nURjuiDiSkI9

link valid 3 days

arendst, 2022-12-19:
Alas. File.io reports: The transfer you requested has been deleted.

Is it that big?

stefanbode, 2022-12-19:
I am wondering why the tilt is not changing after relay start again. there should be a change recorded. Can you send me your shuttertiltconfig to reproduce. I also checked with shuttertiltconfig 0 90 20 0 90 and this seems to work. But for sure agree that there are changes in the xcommand area.

pawol, 2022-12-19:
give me the command to printout shutter tilt config, please

stefanbode, 2022-12-19:
I do not have the issue of the wrong tilt.... my tilt config is 0 0 0 0 0 in this case. 

```
15:10:25.868 HTP: Main Menu
15:10:32.197 SRC: WebGui from 192.168.178.46
15:10:32.199 SRC: WebGui from 192.168.178.46
15:10:32.201 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 9, Pld -99, Data 'StopClose'
15:10:32.205 SHT: Pos. payload <StopClose> (9), payload -99, idx 1 (1), src 7
15:10:32.210 SHT: Pos. payload <StopClose> (0), payload 0, idx 1 (1), src 7
15:10:32.217 SHT: lastsource 7:, real 620000, target 0, tiltreal: 0, tilttarget: 0, payload 0
15:10:32.225 SHT: Delay Start? var1 <99>=<>, max10s?
15:10:32.230 SRC: Shutter
15:10:32.232 SHT: Switched relay 2 by Shutter
15:10:32.238 RSL: RESULT = {"POWER2":"ON"}
15:10:32.241 RSL: POW
```ER2 = ON

pawol, 2022-12-19:
@arendst 
renewed link
https://file.io/ty0yjlAeD3rj

pawol, 2022-12-19:
taken on 11.1 but should be the same after the update onto 12.3.1
15:13:26.601 CMD: shuttertiltconfig1
15:13:26.603 SRC: WebConsole from 192.168.2.8
15:13:26.606 CMD: Grp 0, Cmd 'SHUTTERTILTCONFIG', Idx 1, Len 0, Pld -99, Data ''
15:13:26.610 SHT: TiltConfig 1, min: 0, max 0, runtime 0, close_pos: 0, open_pos: 0
15:13:26.614 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"ShutterTiltConfig1":"SHT:0 0 0 0 0 0"}

stefanbode, 2022-12-19:
@arendst : with your config I have a guess why it try so set 48. This is a bug in the code. @pawol : everything is fine. he should not run into the error.

stefanbode, 2022-12-19:
@arendst : Normally your setting should be rejected. Because open and close are not in the limits of min and max. This is the reason why it is NOT running at your side. I understand your problem with the tilt

arendst, 2022-12-19:
Confirmed. Setting ``ShutterTiltConfig 0 0 0 0 0`` solves my issue.

stefanbode, 2022-12-19:
> @arendst renewed link https://file.io/ty0yjlAeD3rj

This still does not work. We do not need the repeated lines during the run. only the beginning and the end are very interesting

pawol, 2022-12-19:
tele/roleta_w_salonie_lewa/STATE {"Time":"2022-12-19T14:02:11","Uptime":"14T20:13:08","UptimeSec":1282388,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":5,"Downtime":"0T00:41:45"}}
domoticz/in {"idx":8,"nvalue":1,"svalue":"100","Battery":100,"RSSI":8}
tele/roleta_w_salonie_lewa/SENSOR {"Time":"2022-12-19T14:02:11","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
tele/rf433_bridge/RESULT {"Time":"2022-12-19T14:02:13","RfReceived":{"Sync":12960,"Low":410,"High":1220,"Data":8142886,"RfKey":"None"}}
domoticz/in {"idx":19,"nvalue":0,"svalue":"8142886","Battery":100,"RSSI":10}
domoticz/in {"idx":178,"nvalue":1,"svalue1":"0"}
domoticz/in {"idx":19,"nvalue":0,"svalue":"0","Battery":100,"RSSI":10}
tele/roleta_w_salonie_prawa/STATE {"Time":"2022-12-19T14:02:13","Uptime":"14T20:12:57","UptimeSec":1282377,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":6,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":5,"Downtime":"0T00:41:51"}}
domoticz/in {"idx":9,"nvalue":1,"svalue":"100","Battery":100,"RSSI":5}
tele/roleta_w_salonie_prawa/SENSOR {"Time":"2022-12-19T14:02:13","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"POWER1":"ON"}
stat/roleta_w_pokoju_lewa/POWER1 ON
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Button1":1}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.402 SRC: Shutter
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.413 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"POWER1":"ON"}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.419 MQT: stat/roleta_w_pokoju_lewa/POWER1 = ON
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.421 SHT: Shtr1 Real 32230, Start 32230, Stop 620000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.425 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.430 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Button1":1}}
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
tele/roleta_w_pokoju_prawa/STATE {"Time":"2022-12-19T14:02:16","Uptime":"0T00:03:51","UptimeSec":231,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:05"}}
domoticz/in {"idx":11,"nvalue":1,"svalue":"100","Battery":100,"RSSI":4}
tele/roleta_w_pokoju_prawa/SENSOR {"Time":"2022-12-19T14:02:16","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
tele/swiatlo_w_piwnicy_z_wyjsciem/STATE {"Time":"2022-12-19T14:02:17","Uptime":"17T00:01:56","UptimeSec":1468916,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":10,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":8,"Downtime":"0T00:42:03"}}
tele/swiatlo_w_piwnicy_z_wyjsciem/SENSOR {"Time":"2022-12-19T14:02:17","Switch1":"ON"}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.461 SHT: Shtr1 Real 32230, Start 32230, Stop 620000, Dir 1, Delay 0, Rtc 0.05 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.465 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":2,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"POWER1":"OFF"}
stat/roleta_w_pokoju_lewa/POWER1 OFF
stat/roleta_w_pokoju_lewa/SHUTTER1 2
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":2,"Direction":0,"Target":2,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Button1":1}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.377 SHT: Shtr1 Real 50230, Start 32230, Stop 620000, Dir 1, Delay 0, Rtc 0.95 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.382 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":2,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.439 SHT: Stop 1: dir: 1
tele/rf433_bridge/RESULT {"Time":"2022-12-19T14:02:17","RfReceived":{"Sync":13140,"Low":410,"High":1220,"Data":15219254,"RfKey":"None"}}
domoticz/in {"idx":19,"nvalue":0,"svalue":"15219254","Battery":100,"RSSI":10}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.443 SHT: Stop 1 Mode 0 time 20
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.445 SRC: Shutter
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.452 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"POWER1":"OFF"}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.456 MQT: stat/roleta_w_pokoju_lewa/POWER1 = OFF
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.459 SHT: Shtr1 Real 51230, Start 51230, Stop 51230, Dir 0, Delay 0, Rtc 1.00 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.464 MQT: stat/roleta_w_pokoju_lewa/SHUTTER1 = 2
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.470 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":2,"Direction":0,"Target":2,"Tilt":0}}
[... cut ...]
domoticz/in {"idx":11,"nvalue":1,"svalue":"100","Battery":100,"RSSI":4}
tele/roleta_w_pokoju_prawa/SENSOR {"Time":"2022-12-19T14:02:26","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
domoticz/out {
	"Battery" : 255,
	"LastUpdate" : "2022-12-19 14:02:26",
	"RSSI" : 12,
	"description" : "",
	"dtype" : "Light/Switch",
	"hwid" : "3",
	"id" : "000xxxxx",
	"idx" : 95,
	"name" : "Czujnik ruchu (kuchnia)",
	"nvalue" : 0,
	"stype" : "Switch",
	"svalue1" : "0",
	"switchType" : "Motion Sensor",
	"unit" : 1
}

arendst, 2022-12-19:
@pawol gather logging might be easier when you open a second GUI to ``roleta_w_pokoju_lewa`` and open the console there.

Then perfrom your action on the first GUI main screen using the buttons and see the reponses in the console. 

I reduced your previous log to this:
```
stat/roleta_w_pokoju_lewa/RESULT {"POWER1":"ON"}
stat/roleta_w_pokoju_lewa/POWER1 ON
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Button1":1}}

stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.402 SRC: Shutter
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.413 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"POWER1":"ON"}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.419 MQT: stat/roleta_w_pokoju_lewa/POWER1 = ON
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.421 SHT: Shtr1 Real 32230, Start 32230, Stop 620000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.425 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.430 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Button1":1}}

stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}

stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.461 SHT: Shtr1 Real 32230, Start 32230, Stop 620000, Dir 1, Delay 0, Rtc 0.05 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:16.465 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}

stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":2,"Direction":1,"Target":100,"Tilt":0}}

stat/roleta_w_pokoju_lewa/RESULT {"POWER1":"OFF"}
stat/roleta_w_pokoju_lewa/POWER1 OFF
stat/roleta_w_pokoju_lewa/SHUTTER1 2
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Position":2,"Direction":0,"Target":2,"Tilt":0}}
stat/roleta_w_pokoju_lewa/RESULT {"Shutter1":{"Button1":1}}

stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.377 SHT: Shtr1 Real 50230, Start 32230, Stop 620000, Dir 1, Delay 0, Rtc 0.95 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.382 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":2,"Direction":1,"Target":100,"Tilt":0}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.439 SHT: Stop 1: dir: 1

stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.443 SHT: Stop 1 Mode 0 time 20
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.445 SRC: Shutter
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.452 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"POWER1":"OFF"}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.456 MQT: stat/roleta_w_pokoju_lewa/POWER1 = OFF
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.459 SHT: Shtr1 Real 51230, Start 51230, Stop 51230, Dir 0, Delay 0, Rtc 1.00 [s], Freq 0, PWM 0, Tilt 0
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.464 MQT: stat/roleta_w_pokoju_lewa/SHUTTER1 = 2
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.470 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Position":2,"Direction":0,"Target":2,"Tilt":0}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.475 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Shutter1":{"Button1":1}}
stat/roleta_w_pokoju_lewa/LOGGING 14:02:17.652 CFG: Saved to flash at F6, Count 7522, Bytes 4096

tele/roleta_w_pokoju_lewa/STATE {"Time":"2022-12-19T14:02:18","Uptime":"0T00:01:11","UptimeSec":71,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":2,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}
tele/roleta_w_pokoju_lewa/SENSOR {"Time":"2022-12-19T14:02:18","Shutter1":{"Position":2,"Direction":0,"Target":100,"Tilt":0}}
```

stefanbode, 2022-12-19:
What I see what might be causing the issue: There is a SWITCH defined: /SENSOR {"Time":"2022-12-19T14:02:17","Switch1":"ON"}
And shutters only work with buttons. @pawol : Can you send an IMAGE on your config page?

pawol, 2022-12-19:
which page exactly you need?

stefanbode, 2022-12-19:
This one: 
[image]

pawol, 2022-12-19:
[image]
[image]

arendst, 2022-12-19:
Notice the OP is currently running v11.1.0

arendst, 2022-12-19:
If I change the calibration to OPs calibration my shutter also doesn't work as expected. I think the OPs values are wrong.

stefanbode, 2022-12-19:
cannot confirm. He uses `shuttersethalfway 61`. then you get the same values and it work. I have this also here on my bench with this config.

arendst, 2022-12-19:
OK. Then I leave it to you ;-)

stefanbode, 2022-12-19:
Sorry, I had the bad tiltconfig from theo still "active". This causes the issue. Now back to tiltconfig 0 0 0 0 0 and all fine.....

stefanbode, 2022-12-19:
`17:14:38.106 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":310,"Close":290,"50perc":61,"Delay":0,"Opt":"0100","Calib":[1370,483,3223,900,1000],"Mode":"1"}}}` All fine.

I have NO output regarding a SWITCH:
`17:15:58.445 RSL: SENSOR = {"Time":"2022-12-19T17:15:58","Shutter1":{"Position":3,"Direction":0,"Target":3,"Tilt":0}}`

arendst, 2022-12-19:
There is no switch configured in ``roleta_w_pokoju_lewa``. The OPs log contains more than just the ``roleta_w_pokoju_lewa``.

stefanbode, 2022-12-19:
hi @pawol can we make a skype_call with screen sharing in 40min at 6pm german time? Account: stefan.bode

pawol, 2022-12-19:
at the miment i am out of home. wllt contact by priv

pawol, 2022-12-19:
@stefan. I sent you invitation by skype

stefanbode, 2022-12-19:
issue found.` shuttermotorstop 500`. Values was 65535. Will commit a fix that checks on this number and reset to default

[14 more replies]

closed 2022-12-22 by Jason2866