arendst/Tasmota #17464: Tasmota 12.3.1 - Shutter starts moving on his own

OppiM, 2022-12-21:
### PROBLEM DESCRIPTION
After updating Tasmota from 12.2.0 to 12.3.1 my shutters started to act weird. The shutter (**no** venetian blind) is closed, Tasmota device restarted. After a SHUTTEROPEN POWER1 is set ON till 100% is reached, then POWER1 is turned OFF (as expected).

New in 12.3.1 (and 12.3.1.1) then there is then a message "Tilt not match 0 -> 46" in the log. 10 seconds later POWER1 is turned on again, Position starts counting up till 255 when it flips to 0 again. SHUTTERSTOP and POWER1 OFF don't have an impact, Position keeps on changing. Only way to stop is a restarting the device.

Same is the opposite direction, after SHUTTERCLOSE again the "SHT: Tilt not match 0 -> 108" message is displayed, followed by POWER1 on again (so the shutter opens again...).

Flashed back to 12.2.0 and the shutter works normal again  .

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Teepao Shutter
- [x] Tasmota binary firmware version number used: 12.3.1 and 12.3.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:51:48.425 MQT: stat/Shutter_06/RESULT = {"NAME":"Teepao Shutter","GPIO":[288,0,0,33,225,34,0,0,0,224,289,0,32,0],"FLAG":0,"BASE":18}
11:51:48.637 MQT: stat/Shutter_06/RESULT = {"Module":{"0":"Teepao Shutter"}}
11:51:48.843 MQT: stat/Shutter_06/RESULT = {"GPIO0":{"288":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"33":"Button2"},"GPIO4":{"225":"Relay2"},"GPIO5":{"34":"Button3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"224":"Relay1"},"GPIO14":{"289":"Led2"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:53:07.378 MQT: stat/Shutter_06/RESULT = {"Rule1":{"State":"on","Once":"off","StopOnError":"off","Length":37,"Free":474,"Rules":"on button3#state do SHUTTERSTOP endon"}}
11:53:07.598 MQT: stat/Shutter_06/RESULT = {"Rule2":{"State":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}}
11:53:07.848 MQT: stat/Shutter_06/RESULT = {"Rule3":{"State":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:53:42.346 MQT: stat/Shutter_06/STATUS = {"Status":{"Module":0,"DeviceName":"Flur EG","FriendlyName":["Flur EG","Shutter_02_Down"],"Topic":"Shutter_06","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:53:42.384 MQT: stat/Shutter_06/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:34","StartupUTC":"2022-12-21T10:50:08","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2020-03-22T09:10:44","SaveCount":131,"SaveAddress":"F8000"}}
11:53:42.420 MQT: stat/Shutter_06/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"443/699"}}
11:53:42.440 MQT: stat/Shutter_06/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["My-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5AFF002800000000","C0000080","00006000","00004000","00000000"]}}
11:53:42.473 MQT: stat/Shutter_06/STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
11:53:42.514 MQT: stat/Shutter_06/STATUS5 = {"StatusNET":{"Hostname":"Shutter-06","IPAddress":"192.168.xx.xx","Gateway":"192.168.xx.xx","Subnetmask":"255.255.254.0","DNSServer1":"192.168.xx.xx","DNSServer2":"192.168.xx.xx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:53:42.539 MQT: stat/Shutter_06/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xx.xx","MqttPort":1883,"MqttClientMask":"Shutter_06","MqttClient":"Shutter_06","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:53:42.566 MQT: stat/Shutter_06/STATUS7 = {"StatusTIM":{"UTC":"2022-12-21T10:53:42","Local":"2022-12-21T11:53:42","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"08:41","Sunset":"16:55"}}
11:53:42.585 MQT: stat/Shutter_06/STATUS10 = {"StatusSNS":{"Time":"2022-12-21T11:53:42","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
11:53:42.597 MQT: stat/Shutter_06/STATUS11 = {"StatusSTS":{"Time":"2022-12-21T11:53:42","Uptime":"0T00:03:34","UptimeSec":214,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"off","POWER2":"off","Wifi":{"AP":1,"SSId":"My-IoT","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:24"}}}
11:53:42.634 MQT: stat/Shutter_06/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":220,"Close":210,"50perc":60,"Delay":0,"Opt":"0000","Calib":[880,352,2112,900,1000],"Mode":"1"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 131
00:00:00.056 QPC: Count 1
00:00:00.063 SHT: ShutterMode: 1
00:00:00.072 Project tasmota - Flur EG Version 12.3.1(tasmota)-2_7_4_9(2022-12-16T10:28:17)
00:00:00.564 WIF: Connecting to AP1 My-IoT Channel 11 BSSId xx:xx:xx:xx:xx:xx in mode 11n as Shutter-06...
00:00:01.757 WIF: Connected
00:00:02.009 HTP: Web server active on Shutter-06 with IP address 192.168.xx.xx
11:58:09.005 MQT: Attempting connection...
11:58:11.243 MQT: Connected
11:58:11.247 MQT: tele/Shutter_06/LWT = Online (retained)
11:58:11.249 MQT: cmnd/Shutter_06/POWER = 
11:58:11.254 MQT: tele/Shutter_06/INFO1 = {"Info1":{"Module":"Teepao Shutter","Version":"12.3.1(tasmota)","FallbackTopic":"cmnd/Shutter_06_fb/","GroupTopic":"cmnd/tasmotas/"}}
[... 278 lines cut ...]
11:59:09.289 SHT: Shtr1 Time 94(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 533000, nStop 0, Trgt 440000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 46, Tiltmove: 1
11:59:09.339 SHT: Shtr1 Time 95(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 534000, nStop 0, Trgt 440000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 46, Tiltmove: 1
11:59:09.389 SHT: Shtr1 Time 96(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 535000, nStop 0, Trgt 440000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 46, Tiltmove: 1
11:59:09.393 SHT: Shtr1 Real 535000, Start 440000, Stop 440000, Dir 1, Delay 0, Rtc 4.80 [s], Freq 0, PWM 0, Tilt 0
11:59:09.401 MQT: stat/Shutter_06/RESULT = {"Shutter1":{"Position":127,"Direction":1,"Target":100,"Tilt":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Tasmota 12.3.1
SHUTTEROPEN

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shutter should work link in version 12.2.0.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Maybe the Tilt function should be controlled by an Option so it could be enabled for venetian blinds but would not affect normal shutter.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-12-21:
Sounds familiar: https://github.com/arendst/Tasmota/issues/17403#issuecomment-1357755901

OppiM, 2022-12-21:
OK, checked the setting:

`15:19:25.399 CMD: ShutterTiltConfig`
`15:19:25.405 SHT: TiltConfig 1, min: 0, max 0, runtime 48, close_pos: 46, open_pos: 108`
`15:19:25.409 MQT: stat/Shutter_06/RESULT = {"ShutterTiltConfig1":"0 0 0 48 46"}`

Changed the setting as advised : `ShutterTiltConfig 0 0 0 0 0`

Restarted the Tasmota device and it worked as before.

I checked my other switches that I had flashed back to 12.2.0 and they all had the same ShutterTiltConfig "0 0 0 48 46". Changed their config to "0 0 0 0 0", flashed them to 12.3.1 again and the config stayed at "0 0 0 0 0".

So I'm not sure where the "0 0 0 48 46" was coming from.

But issue is fixed for me. Thanks for the quick help. 

Maybe worth adding to the shutter docu to set `ShutterTiltConfig 0 0 0 0 0` for non-Venitial Blinds.

closed 2022-12-21 by OppiM

stefanbode, 2022-12-21:
In my development build I have changed the ShutterInit procedure to avoid wrong tiltconfig. Will submitt later this week. Thx for jumping in. Anyhow quite strange setting...

arendst, 2022-12-21:
Well it was the exact same setting as I had. Probably another re-use of previous Settings variable(s).

stefanbode, 2022-12-22:
Fixed by latest change #17484