arendst/Tasmota #17530: MCP230xx - sensor29 commands error out

pin32, 2022-12-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._the problem is identical to the case already covered in this case #5535

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pin32, 2022-12-27:
il firmware personalizzato è stato compilato con TASMOCOMPILER aggiungendo le seguenti define:
#define USE_MCP230xx
#define USE_MCP230xx_ADDR 0x20
#define USE_MCP230xx_OUTPUT
#define USE_MCP230xx_DISPLAYOUTPUT

sfromis, 2022-12-27:
You need to fill in the template, in English, and be very clear about symptoms.

pin32, 2022-12-28:
Board ESP32 WROOM 32

I would like to use MCP23017

Compiled with Visual Studio Code using env tasmota32 and enabled the following:

```
#define USE_I2C
#define USE_MCP230xx
#define USE_MCP230xx_ADDR 0x20
#define USE_MCP230xx_OUTPUT
#define USE_MCP230xx_DISPLAYOUTPUT 
```

The chip is available:
```

20:14:22 CMD: i2cscan
20:14:22 RSL: stat/remote-test/RESULT = {"I2CScan":"Device(s) found at 0x20"}
```

Using i2cdriver:
```

23:20:53.319 CMD: i2cdriver
23:20:53.328 MQT: stat/SolarControl/RESULT = {"I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,22,24,29,31,36,41,42,44,46,48,58,62,69"}
```

But any sensor29 command just returns an error:

```
0:11:52 CMD: sensor29
20:11:52 RSL: stat/remote-test/RESULT = {"Command":"Error"}
20:12:03 CMD: sensor29 reset
20:12:03 RSL: stat/remote-test/RESULT = {"Command":"Error"}
20:21:15 CMD: sensor29 0,?
20:21:15 RSL: stat/remote-test/RESULT = {"Command":"Error"}

```

I tried to compile the master and dev branch of github code but the result is the same.

Hope in your help

barbudor, 2022-12-29:
You're likely have too many i2cdriver enabled
So another driver may believe the chip at 0x20 if for him
Disable this you don't need

Jason2866, 2023-01-04:
Hardware setup? Pull downs on i2c bus? Without it will not work.
Breadboard used? Try without. I2C often do not work when using a breadboard.

sfromis, 2023-01-04:
Strictly speaking I2C bus needs pull **up** resistors, not down. While these are commonly already on I2C breakout boards, not always. And yes, brea**k**boards is a risk factor, like other dangling antenna/capacitor wires, along with the the plausibility of bad connections.

Jason2866, 2023-01-05:
Closing, solved in Discord. Bad hardware setup.

closed 2023-01-05 by Jason2866