arendst/Tasmota #17542: Tuya MCU V2 does not complete initialisation on WIFI-101 door/window contact switch.

hydra, 2022-12-30:
### PROBLEM DESCRIPTION

The TuyaMCU V2 initialisation sequence does not complete, it gets stuck at stage 8, the log reports:

```
00:00:03.711 TYA: Protocol timeout state 8 -> 0
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [WIFI-101](https://expo.tuya.com/product/322001)
- [x] Tasmota binary firmware version number used: a605d4b6723cacdaebe20a5fe91bb8111de2c28d from jason2866/special_build (tasmota-battery)
  - [x] Pre-compiled (same issue as when building locally) 
  - [x] Self-compiled
- [x] Flashing tools used: esptool or pio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
11:55:01.939 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
11:55:01.948 MQT: stat/switch1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
11:55:02.440 MQT: tasmota/discovery/E0980610F709/config = {"ip":"192.168.18.175","dn":"Switch1","fn":["Switch 1",null,null,null,null,null,null,null],"hn":"switch1-5897","mac":"E0980610F709","md":"Tuya MCU","ty":1,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.3.1.2","t":"switch1","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
11:55:02.448 MQT: tasmota/discovery/E0980610F709/sensors = {"sn":{"Time":"2022-12-30T11:55:02"},"ver":1} (retained)
```

```lua
  Configuration output here:

N/A

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

12:05:46.741 CMD: Status 0
12:05:46.743 SRC: WebConsole from 192.168.18.49
12:05:46.746 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
12:05:46.752 MQT: stat/switch1/STATUS = {"Status":{"Module":54,"DeviceName":"Switch1","FriendlyName":["Switch 1"],"Topic":"switch1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:05:46.776 MQT: stat/switch1/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"OTA update can brick your device","RestartReason":"External System","Uptime":"0T00:11:00","StartupUTC":"2022-12-30T10:54:46","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2022-12-29T20:30:47","SaveCount":15,"SaveAddress":"F8000"}}
12:05:46.796 MQT: stat/switch1/STATUS2 = {"StatusFWR":{"Version":"12.3.1.2(battery)","BuildDateTime":"2022-12-30T11:27:05","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"350/699"}}
12:05:46.815 MQT: stat/switch1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SeriouslyPro!","Avatel_91"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C80001000600003C5A00192800000000","00018000","00006000","04004000","00000000"]}}
12:05:46.839 MQT: stat/switch1/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":29,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","08100582","04008001","00000003","00000000","00000800","00000000","00000000","54000000","00000080"],"Drivers":"1,2,4,7,10,12,16,29,62","Sensors":"1,2"}}
12:05:46.857 MQT: stat/switch1/STATUS5 = {"StatusNET":{"Hostname":"switch1-5897","IPAddress":"192.168.18.175","Gateway":"192.168.18.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.18.1","DNSServer2":"0.0.0.0","Mac":"E0:98:06:10:F7:09","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:05:46.876 MQT: stat/switch1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.18.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_10F709","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:05:46.893 MQT: stat/switch1/STATUS7 = {"StatusTIM":{"UTC":"2022-12-30T11:05:46","Local":"2022-12-30T12:05:46","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00"}}
12:05:46.912 MQT: stat/switch1/STATUS10 = {"StatusSNS":{"Time":"2022-12-30T12:05:46"}}
12:05:46.929 MQT: stat/switch1/STATUS11 = {"StatusSTS":{"Time":"2022-12-30T12:05:46","Uptime":"0T00:11:00","UptimeSec":660,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SeriouslyPro!","BSSId":"D8:3B:BF:FB:56:0D","Channel":5,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.002 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F9, Count 14
00:00:00.055 SER: Set to 8N1 9600 bit/s
00:00:00.057 CFG: CR 350/699, Busy 0
00:00:00.062 SNS: Hardware Serial
[... 129 lines cut ...]

[image]

### ADDITIONAL CONTEXT

I reverse engineered the schematic, attached:

[schematic.pdf](https://github.com/arendst/Tasmota/files/10324301/schematic.pdf)

I dumped the original firmware, attached:

[20221228-2007-contact_sensor_1_1M.zip](https://github.com/arendst/Tasmota/files/10324320/20221228-2007-contact_sensor_1_1M.zip)

I took trace logs of 3v3, Reed Switch, Wifi RX/TX from the original firmware using Logic, I exported the data as CSV and also created Excel comparisons and analyzed the data a little to find the details of which command the original firmware is using the get the status of the reed switch, logs attached:

[20221230-1201-original-firmware-logic-trace-logs-and-csv-exports-and-comparisons.zip](https://github.com/arendst/Tasmota/files/10324323/20221230-1201-original-firmware-logic-trace-logs-and-csv-exports-and-comparisons.zip)

I also compiled the same firmware but re-enabled the V1 code, logs attached:

[log - v1 - reed closed.txt](https://github.com/arendst/Tasmota/files/10324336/log.-.v1.-.reed.closed.txt)
[log - v1 - reed open.txt](https://github.com/arendst/Tasmota/files/10324337/log.-.v1.-.reed.open.txt)
[log - v2 - reed closed.txt](https://github.com/arendst/Tasmota/files/10324338/log.-.v2.-.reed.closed.txt)
[log - v2 - reed open.txt](https://github.com/arendst/Tasmota/files/10324339/log.-.v2.-.reed.open.txt)

**(Please, remember to close the issue when the problem has been addressed)**

hydra, 2022-12-30:
Datasheets for the key components on the WIFI-101 PCB:

Charge Controller:
[405442xf.pdf](https://github.com/arendst/Tasmota/files/10324346/405442xf.pdf)

MCU:
[HT66F0175_185v161.pdf](https://github.com/arendst/Tasmota/files/10324347/HT66F0175_185v161.pdf)

Transistors/FETs:
[transistor_sot23_S8050_J3Y.pdf](https://github.com/arendst/Tasmota/files/10324349/transistor_sot23_S8050_J3Y.pdf)
[transistor_sot23_AO3401_A19T.pdf](https://github.com/arendst/Tasmota/files/10324350/transistor_sot23_AO3401_A19T.pdf)

Wifi Module:
[Users-Manual-3525098.pdf](https://github.com/arendst/Tasmota/files/10324351/Users-Manual-3525098.pdf)

Images of the PCB with WiFi module removed to enable reverse engineering of the schematic:

[image]
[image]

Jason2866, 2022-12-30:
Probably related to #17190

hydra, 2022-12-30:
I'm fairly adept at coding, if there's anything specific in the codebase, or even just hints, let me know and I'll take a look.

arendst, 2022-12-31:
Due to lack of support Tuya V2 in development has been reverted to V1. Sorry.

hydra, 2023-01-02:
@arendst What do you mean by 'lack of support'?  What's the issue?

I also tried this particular device with V1 code by using the `build_special` branch and revering the #ifdef as per `master` branch but the device doesn't ever show the door contact switch status.  See the logs attached.

The 'https://github.com/sillyfrog/Tasmota-Tuya-Helper' bookmarklet tool doesn't work with either V1 or V2 code and always shows a backlog command.

Any ideas on how to get either the device working with either V1 or the V2 code?

arendst, 2023-01-02:
See #17190 and other tuya related issues not being addressed by either original maintainers (as they moved on) or users willing and able to dive into the issues.

The V2 version seems to be more error prone than the V1 version so that's why V2 never made it into production and was replaced by V1 in development.

I don't own any tuya related hardware so cannot help here.

hydra, 2023-01-02:
@arendst ok, understood, thanks for the clarification.

Today I did a new build on the latest `build_special` branch which now has the V1 code as the default and used SerialSend to get the status of the contact switch.

For anyone else following along, here's the full set commands to get the Wifi-101 contact sensor working, once you've flashed `tasmota-battery` (2023/01/02):

```
Module 54

Backlog SetOption1 1; SetOption65 1; SetOption66 1

TuyaMCU 51,51

Rule1 ON TuyaReceived#Data=55AA00050005030400010213 DO publish2 stat/%topic%/BATTERY high ENDON ON TuyaReceived#Data=55AA00050005030400010112 DO publish2 stat/%topic%/BATTERY medium ENDON ON TuyaReceived#Data=55AA00050005030400010011 DO publish2 stat/%topic%/BATTERY low ENDON

Rule1 1

Rule2 ON TuyaReceived#Data=55AA0002000001 DO SerialSend5 55aa000500010005 ENDON

Rule2 1
```

Here's two logs showing the output:

Closed-To-Open:
```
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 59
00:00:00.054 SER: Set to 8N1 9600 bit/s
00:00:00.056 CFG: CR 350/699, Busy 0
00:00:00.061 SNS: Hardware Serial
00:00:00.062 TYA: Request MCU configuration at 9600 bps
00:00:00.062 SRC: Restart
00:00:00.075 TYA: Send "55aa0006000501010001000d"
00:00:00.108 Project tasmota - Switch1 Version 12.3.1.3(battery)-2_7_4_9(2023-01-02T12:30:25)
00:00:00.125 TYA: Set WiFi LED 2 (-1)
00:00:00.134 TYA: Send "55aa000200010204"
00:00:00.152 RSL: RESULT = {"TuyaReceived":{"Data":"55AA0002000001","Cmnd":2}}
00:00:00.173 RUL: TUYARECEIVED#DATA=55AA0002000001 performs "SerialSend5 55aa000500010005"
00:00:00.173 SRC: Rule
00:00:00.175 CMD: Grp 0, Cmd 'SERIALSEND', Idx 5, Len 16, Pld 55, Data '55aa000500010005'
00:00:00.177 RSL: RESULT = {"SerialSend":"Done"}
00:00:00.189 WIF: Checking connection...
00:00:00.190 WIF: Attempting connection...
00:00:00.528 WIF: Connecting to AP1 SeriouslyPro! Channel 5 BSSId D8:3B:BF:FB:56:0D in mode 11n as switch1-5897...
00:00:01.752 WIF: Checking connection...
00:00:01.753 WIF: Attempting connection...
00:00:02.752 WIF: Checking connection...
00:00:02.753 WIF: Attempting connection...
00:00:03.753 WIF: Checking connection...
00:00:03.754 WIF: Connected
00:00:04.007 HTP: Web server active on switch1-5897 with IP address 192.168.18.175
00:00:04.412 HTP: Console
00:00:04.516 WIF: Sending Gratuitous ARP
00:00:04.517 NTP: Sync time...
00:00:04.530 WIF: DNS resolved '2.pool.ntp.org' (178.215.228.24) in 13 ms
00:00:04.805 RTC: UTC 2023-01-02T12:18:48, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
13:18:48.000 RTC: Synced by NTP
13:18:48.001 TYA: Set WiFi LED 3 (0)
13:18:48.010 TYA: Send "55aa000200010305"
13:18:48.046 RSL: RESULT = {"TuyaReceived":{"Data":"55AA0002000001","Cmnd":2}}
13:18:48.064 RUL: TUYARECEIVED#DATA=55AA0002000001 performs "SerialSend5 55aa000500010005"
13:18:48.065 SRC: Rule
13:18:48.067 CMD: Grp 0, Cmd 'SERIALSEND', Idx 5, Len 16, Pld 55, Data '55aa000500010005'
13:18:48.070 RSL: RESULT = {"SerialSend":"Done"}
13:18:48.039 MQT: Attempting connection...
13:18:48.041 WIF: DNS resolved '192.168.18.5' (192.168.18.5) in 1 ms
13:18:48.052 MQT: Connected
13:18:48.055 MQT: tele/switch1/LWT = Online (retained)
13:18:48.059 MQT: cmnd/switch1/POWER = 
[... 91 lines cut ...]
13:16:51.296 SRC: Rule
13:16:51.299 CMD: Grp 0, Cmd 'PUBLISH', Idx 2, Len 27, Pld -99, Data 'stat/switch1/BATTERY medium'
13:16:51.303 MQT: stat/switch1/BATTERY = medium (retained)
13:16:51.315 TYA: fnId=0 is set for dpId=3
13:16:51.319 MQT: tele/switch1/RESULT = {"TuyaReceived":{"Data":"55AA0005000501010001000C","Cmnd":5,"CmndData":"0101000100"}}
13:16:51.341 TYA: fnId=11 is set for dpId=1
13:16:51.342 TYA: RX Relay-1 --> MCU State: Off Current State:On
13:16:51.344 SRC: Switch
13:16:51.349 MQT: stat/switch1/RESULT = {"POWER":"OFF"}
13:16:51.353 MQT: stat/switch1/POWER = OFF
13:16:51.374 HTP: Console
13:16:51.478 TYA: Send "55aa000500010106"
13:16:51.680 CFG: Saved to flash at F4, Count 55, Bytes 4096
```

This is the firmware I used:
[tasmota-battery.zip](https://github.com/arendst/Tasmota/files/10331528/tasmota-battery.zip)

Source:
https://github.com/Jason2866/Tasmota-build/tree/42327e20498839574ecc080bd7708952ed6bd7d1

Commit:
`42327e20498839574ecc080bd7708952ed6bd7d1`

Note that the MCU seems to become unresponsive after a few seconds after power-up.

hydra, 2023-01-02:
Instructions for flashing and configuration.

1) Carefully desolder battery wires.
2) Connect USB UART:
Wifi Module TX -> RX
Wifi Module RX -> TX
GND -> GND.
3) Connect wire between Q1 Gate (next to resistor with 104 marking) to GND.
4) Connect Wifi Module GPIO0 to GND via a momentary or toggle switch.
5) Connect bench PSU to battery pads, 4.2V current limit 500mA.
6) Close GPIO0/BOOT toggle switch.
7) Power on bench PSU, release toggle switch. (~30mA draw)
8) Backup firmware:
```
esptool -b 115200 --port COM7 read_flash 0x000000 0x100000 contact_sensor_1M.bin
```
Log:
```
esptool.py v4.4
Serial port COM7
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 24:62:ab:1f:04:e4
Uploading stub...
Running stub...
Stub running...
1048576 (100 %)
1048576 (100 %)
Read 1048576 bytes at 0x00000000 in 94.3 seconds (88.9 kbit/s)...
Hard resetting via RTS pin...
```
9) Install tasmota, either using PIO or pre-built.
a)Build and flash tasmota using PlatformIO:
```
pio run -e tasmota-battery --target upload --upload-port COM7
```
Log:
```
Processing tasmota-battery (platform: https://github.com/tasmota/platform-espressif8266/releases/download/2022.12.0/platform-espressif8266.zip; framework: arduino; board: esp8266_1M)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8266_1M.html
PLATFORM: Espressif 8266 (2022.12.0) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS
HARDWARE: ESP8266 80MHz, 80KB RAM, 972.00KB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 2.7.4+9
 - tool-esptoolpy @ 1.40401.0 (4.4.1)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
Converting tasmota.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- headers
|   |-- EEPROM @ 1.0
|   |-- ESP8266WebServer @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- EEPROM 24C128_256_521
|   |-- Wire @ 1.0
|-- UdpListener @ 1.0
|-- Unishox Compressor Decompressor highly customized and optimized for ESP8266 and Tasmota @ 1.0
|-- base64 @ 1.1.1
[... 140 lines cut ...]
Rule2 1
```
20) Configure other settings as required.
21) Backup config to a file using the WebUI.
22) Power off Bench PSU.
23) Disconnect Q1 GATE from GND.
24) Power on Bench PSU.
Note that the device will power on, transmit it's status via MQTT and then power off, this only takes a few seconds and you can observe the current draw on the bench PSU to know when the Wifi module is powered off by the MCU.

you should see an MQTT message `stat/<topic>/POWER` = `OFF` sent to your MQTT broker.

25) hold the magnet next to the reed switch.

you should see an MQTT message `stat/<topic>/POWER` = `ON` sent to your MQTT broker.

26) Power off bench PSU and reverse steps 5-1 above, then re-assemble the case.

To restore firmware:
```
esptool -b 115200 --port COM7 write_flash --erase-all 0x000000 contact_sensor_1M.bin
```

Reconfiguring on battery:

If you press-and-hold the 'Reset' switch for 5 seconds and release, the device will flash the red LED and stay powered on for about 2 minutes during which time the Wifi module is powered on and you can connect to it over the network.

hydra, 2023-02-01:
@arendst Is it worth noting somewhere on the site that a maintainer for the Tuya related code is needed?  Is there a list of *current* maintainers for certain portions of the codebase?

Jason2866, 2023-02-01:
Listed here https://github.com/arendst/Tasmota/blob/development/CODE_OWNERS.md
but this does not implicate that the owner still does maintain. For Tuya code there is currently no maintainer

barbudor, 2023-02-01:
Myself and a few others have recently made some maintenance on Tuya V1

Regarding Tuya V2 it was a hit and run and it is now disabled by default.
I don't know what justified this work (Tuya V2) and why the contributor walked away but that cannot work
V2 is now dead code

If you have some comments on V1, let us know and I may be in a position to help

hydra, 2023-02-07:
Hi @barbudor.

Firstly, thanks for doing some maintenance!

My observations with the setup above is that time-from contact opening, to messages being received by my MQTT broker, is quite long, and sometimes they do not appear at all.  I can see the contact switch light coming on, so I know it's powering on ok.   For example, if I open or close a door sometimes I do not get any MQTT messages.  If I then wait 10 seconds, and then close or open the same door the message may come though.

Can the the process be sped up somehow?  The original Tuya firmware on the wifi module was faster for sure, as my logic analyzer traces show.

barbudor, 2023-02-07:
Have you retested since we disabled V2 and reverted to V1 ?
I won't invest a minute into V2

hydra, 2023-02-14:
> Have you retested since we disabled V2 and reverted to V1 ? I won't invest a minute into V2

yes, the my previous comment was relating to the V1 code, as detailed here: https://github.com/arendst/Tasmota/issues/17542#issuecomment-1368906746 (above)

closed 2023-03-16 by github-actions[bot]

btsimonh, 2023-08-17:
@barbudor - you only had to ask? (about V2).

benjaminnestler, 2024-02-03:
Could it be the case that the original issue of @hydra is that his device seems to be using the Tuya low power wifi protocol.

Take a look at the responses messages of the Tuya mcu. The header including the version byte 3 is not equal to the Tuya standard wifi protocol. Take a look at #20608.

The issue with either tuya v1 or tuya v2 seems to be the misinterpretation of the protocol commands of tuya mcu due to communicate with the wrong tuya protocol.