arendst/Tasmota #17568: Sometimes current accumulated Energy kWh value ist lower then the value before

joka222, 2023-01-01:
### PROBLEM DESCRIPTION
Accumulated Energy kWh is sometimes set to a lower value then the value before. What could be the reason?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Gosund SP111 v1.1
- [ ] Tasmota binary firmware version number used: 12.3.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:10:38.551 MQT: stat/Gosund-Eingang-Garten/RESULT = {"NAME":"SP111 v1.1","GPIO":[320,0,576,0,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
18:10:38.761 MQT: stat/Gosund-Eingang-Garten/RESULT = {"Module":{"0":"SP111 v1.1"}}
18:10:39.017 MQT: stat/Gosund-Eingang-Garten/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:12:47.911 MQT: stat/Gosund-Eingang-Garten/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.69.19:32782/data/firmwares/tasmota.bin.gz","RestartReason":"Exception","Uptime":"3T08:00:54","StartupUTC":"2022-12-29T09:11:53","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2022-06-03T19:53:24","SaveCount":61,"SaveAddress":"FA000"}}
18:12:47.915 MQT: stat/Gosund-Eingang-Garten/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022.12.16 10:33:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"431/699"}}
18:12:47.920 MQT: stat/Gosund-Eingang-Garten/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TRUSTNOONE","TRUSTNOONE"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000280","00006000","00004000","00000000"]}}
18:12:47.932 MQT: stat/Gosund-Eingang-Garten/STATUS4 = {"StatusMEM":{"ProgramSize":630,"Free":372,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
18:12:47.937 MQT: stat/Gosund-Eingang-Garten/STATUS5 = {"StatusNET":{"Hostname":"Gosund-Eingang-Garten","IPAddress":"192.168.69.92","Gateway":"192.168.69.99","Subnetmask":"255.255.255.0","DNSServer1":"192.168.69.99","DNSServer2":"0.0.0.0","Mac":"F4:CF:A2:01:CC:66","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:12:47.941 MQT: stat/Gosund-Eingang-Garten/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.69.19","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_01CC66","MqttUser":"sonoff","MqttCount":6,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:12:47.947 MQT: stat/Gosund-Eingang-Garten/STATUS7 = {"StatusTIM":{"UTC":"2023-01-01T17:12:47","Local":"2023-01-01T18:12:47","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:03"}}
18:12:47.951 MQT: stat/Gosund-Eingang-Garten/STATUS9 = {"StatusPTH":{"PowerDelta":[80,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:12:48.004 MQT: stat/Gosund-Eingang-Garten/STATUS10 = {"StatusSNS":{"Time":"2023-01-01T18:12:47","ENERGY":{"TotalStartTime":"2022-06-03T19:53:24","Total":3.667,"Yesterday":0.054,"Today":0.056,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
18:12:48.011 MQT: stat/Gosund-Eingang-Garten/STATUS11 = {"StatusSTS":{"Time":"2023-01-01T18:12:48","Uptime":"3T08:00:55","UptimeSec":288055,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":6,"POWER":"OFF","Wifi":{"AP":2,"SSId":"TRUSTNOONE","BSSId":"3C:37:12:B4:AE:44","Channel":3,"Mode":"11n","RSSI":34,"Signal":-83,"LinkCount":4,"Downtime":"0T00:03:40"}}}
18:12:48.023 MQT: stat/Gosund-Eingang-Garten/STATUS12 = {"StatusSTK":{"Exception":2,"Reason":"Exception","EPC":["3ffed4c0","00000000","00000000"],"EXCVADDR":"3ffed4c0","DEPC":"00000000","CallChain":["401020c1","4024ed08","40000f49","40000f49","40000e19","40105acd","40105ad3","40105915","4026fbed","401054fa","40100aa0","40102b45","40104e49","40102714","401059af","401053fd","401056d5","4010399e","40100230","40101421","4000050c","40103670","40101410","40254f58","402555ba","401048bf","401048bc","4000050c","4010381f","402555ba","4000050c"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The current value of Energy kWh cannot be lower then the value before because it is a accumulated value.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-01-01:
Do not open an issue to ask a question, better do that in Discussions instead.

If you cut input power to an energy monitoring device, it will lose what's accumulated since last save.

joka222, 2023-01-02:
Maybe it is an issue.
Input power was not cut! And it loses kWh over hours until the device accumulates again in the positive direction.
Do you know what the energy save frequency of a device is or were I can adjust it?

sfromis, 2023-01-02:
Save is midnight local time, or when save for other reasons, by default things like relay status change.

If you think that it was not power loss, you need to document what happens. Your screenshot documents nothing. You posted nothing in the important section of log when you have the issue.