arendst/Tasmota #17570: Can't upgrade Sonoff S26 Plug from 12.2.0 to 12.3.1

hipitihop, 2023-01-02:
### PROBLEM DESCRIPTION
Can't upgrade Sonoff S26 Plug from 12.2.0 to 12.3.1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: Sonoff S26
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:54:57.319 MQT: plug-2/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
10:54:57.549 MQT: plug-2/stat/RESULT = {"Module":{"8":"Sonoff S2X"}}
10:54:57.804 MQT: plug-2/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade from 12.2.0 to 12.3.1. Error "Insufficient Space" Even when 12.3.1. minimal used first.

### EXPECTED BEHAVIOR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-01-02:
Can not reproduce. Tried with two Sonoff S20

sfromis, 2023-01-02:
You omitted nearly all details important for seeing what happened. For upgrades you should *always* use the `.gz` variant, without that, expect it to not work. Recommended to upgrade directly from the URL to the OTA http web site, without downloading first.

hipitihop, 2023-01-03:
@Jason2866 & @sfromis I appreciate you looking into this and apologies for missing the extra info.

I retried my steps and found that the OTA upgrade left me with minimal 12.3.1 and further OTA did not work. This is the state that prompted me originally to try the manual route. This included downgrading back to 12.2.0, but as you mentioned, I incorrectly used the .bin instead of the .bin.gz While the downgrade was fine with the .bin, it failed again with the space issue for 12.3.1 So this time I used the .bin.gz and viola it upgraded. My other matching Sonoff S26 was fine with the OTA upgrade.

In the interest on not wasting more of peoples time, I'm closing this. Thanks everyone on your awesome work on this firmware.

closed 2023-01-03 by hipitihop