arendst/Tasmota #17573: Sonoff ZBBrigePro inoperable after upgrade to Tasmota 12.3.1

meyergru, 2023-01-02:
### PROBLEM DESCRIPTION
After trying to flash 12.3.1 release, device is inoperable.

After reboot, the screen only showed two lines, HTTP transfer was incomplete. It seems like accessing the web UI caused a reboot.

Resetting via power-off did not help.

Factory reset via Button 1 for 40 seconds brought up Tasmota WiFi, device could be configured to another WiFi.

Bringing up the web UI shows "Safe Boot (version 12.1.0)", flashing any version does not work. Instead, the device seems to reboot every time and after that, reacts to ping but port 80 is closed. A power-off brings up the web interface again, but I cannot get it configured or even out of safeboot mode.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff ZBBridgePro
- [X] Tasmota binary firmware version number used: 12.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

Can't do

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Upgrade device from 12.2.x

### EXPECTED BEHAVIOUR
Device should work

### SCREENSHOTS
none

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-01-02:
Can you connect to serial and look at serial logs?

meyergru, 2023-01-02:
Alas, no. It looks as if the fs layout has been corrupted.

meyergru, 2023-01-02:
I just got it back by updating to 12.2.0 release version. I will leave it at that until someone else confirms a newer version is working.

nikito7, 2023-01-02:
I dont know if is related. But >E in script is not working

s-hadinger, 2023-01-02:
> I dont know if is related. But >E in script is not working

Scripting is not enabled in standard build

Jason2866, 2023-01-03:
Strange issue, just got the feedback from a user installing v12.3.1 via web installer does work fine.

meyergru, 2023-01-03:
As I said, it is probably a matter of fs layout, I do not understand the details of it, but I know that I once followed these instructions: https://notenoughtech.com/home-automation/tasmota-on-sonoff-zb-bridge-pro/ to use a 1152kbyte partition.

However, now the size seems to be 1344kbytes, as in the fs manager,, I now see: Size 1.344 MB - Free 0.908 MB

meyergru, 2023-01-03:
Yup. Size may be the explanation: 12.3.1.3 of today with 1427kbyte works, 12.3.1 with 1480kbyte did not (for me).

For reference: the URL for 12.3.1 is: https://ota.tasmota.com/tasmota32/release/tasmota32-zbbrdgpro.bin, for the daily version it is: https://ota.tasmota.com/tasmota32/tasmota32-zbbrdgpro.bin

meyergru, 2023-04-19:
And again. 12.5.0 does the same again.

sfromis, 2023-04-19:
If size of the binary is an issue to you, you should upgrade to the safeboot partition layout, where the OTA partition size is much larger, and can easily accommodate a modest sized binary like 1480 KB. With Sonoff ZbBridge "pro" 12.5 I have:
[image]

meyergru, 2023-04-19:
How can I check and how can I re-partition?

meyergru, 2023-04-19:
FWIW, on 2nd try, now the flashing worked. I am at a loss why...