arendst/Tasmota #17577: Tamota devices are missing decimals in Domoticz

HB787, 2023-01-03:
Hi Team,

I have about 10 Tasmota devices in my network.
I used the Domoticz settings for the virtual devices within the Tasmota WebUI and the MQTT client plugin within Domoticz.

So far so good, it works. But one thing the energy metering don't have decimals and it looks like only integers get through.
Another problem is that the MQTT Auto Discovery plugin does not work at all with my devices.
I tried a German plugin that overspooled my Domoticz and has issues with multichannel energy meters, like de Sonoff Dual R3.
The Autodiscovery flag is also gone so it does not help.

It is a pitty I can't have precise measering in Domoticz.
Is there any solution for this problem.

arendst, 2023-01-03:
Provide the requested info first so we can have an idea where to start to help you.

sfromis, 2023-01-03:
Don't know about Domoticz, but for other energy monitoring, I'm upping the various resolution settings available for more decimals than the very limited default setup. AmpRes, EnergyRes, FreqRes, VoltRes and WattRes.

arendst, 2023-01-03:
Again, provide the basic information as requested. Output from command ``status 0`` and in your case a console log with ``weblog 4`` showing the comms between tasmota and domoticz.

HB787, 2023-01-03:
status 0:

15:27:16.413 MQT: stat/Tasmota_tuinpaal/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota Sonoff Dual R3 V2 Tuinpaal","FriendlyName":["Tuinpaal Camera","Ledbeamers"],"Topic":"Tasmota_tuinpaal","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":8,"LedMask":"FFFF","SaveData":5,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:27:16.441 MQT: stat/Tasmota_tuinpaal/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T19:22:42","StartupUTC":"2023-01-02T19:04:34","Sleep":1,"CfgHolder":4617,"BootCount":102,"BCResetTime":"2022-08-27T10:07:36","SaveCount":2416}}
15:27:16.464 MQT: stat/Tasmota_tuinpaal/STATUS2 = {"StatusFWR":{"Version":"12.3.1.2(tasmota)","BuildDateTime":"2022-12-29T20:59:17","Core":"2_0_6","SDK":"v4.4.3.20221227","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"457/699"}}
15:27:16.486 MQT: stat/Tasmota_tuinpaal/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["berg-media",""],"TelePeriod":30,"Resolution":"561180C0","SetOption":["00208009","2805C80001000600003C5A0A192800000000","000000C1","00006000","00004000","00000001"]}}
15:27:16.517 MQT: stat/Tasmota_tuinpaal/STATUS4 = {"StatusMEM":{"ProgramSize":1497,"Free":1856,"Heap":146,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D43C482D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
15:27:16.540 MQT: stat/Tasmota_tuinpaal/STATUS5 = {"StatusNET":{"Hostname":"Tasmota-Dual-R3-V2-Tuinpaal","IPAddress":"192.168.1.62","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"8.8.8.8","Mac":"C4:DD:57:DB:68:C8","IP6Global":"2a02:a45b:be40:1:c6dd:57ff:fedb:68c8","IP6Local":"fe80::c6dd:57ff:fedb:68c8","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:27:16.562 MQT: stat/Tasmota_tuinpaal/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.220","MqttPort":1883,"MqttClientMask":"tasmota_%06X","MqttClient":"tasmota_DB68C8","MqttUser":"berg114","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:27:16.584 MQT: stat/Tasmota_tuinpaal/STATUS7 = {"StatusTIM":{"UTC":"2023-01-03T14:27:16","Local":"2023-01-03T15:27:16","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:46","Sunset":"16:36"}}
15:27:16.605 MQT: stat/Tasmota_tuinpaal/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:27:16.642 MQT: stat/Tasmota_tuinpaal/STATUS10 = {"StatusSNS":{"Time":"2023-01-03T15:27:16","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-27T10:07:35","Total":1.9627,"Yesterday":0.0898,"Today":0.0582,"Power":[3.99,0.00],"ApparentPower":[4301.24,0.00],"ReactivePower":[4301.00,0.00],"Factor":[0.00,0.00],"Voltage":234,"Current":[18.393,0.000],"BL09XX":{"Temperature":28.0}},"ESP32":{"Temperature":50.0,"HallEffect":29},"TempUnit":"C"}}
15:27:16.665 MQT: stat/Tasmota_tuinpaal/STATUS11 = {"StatusSTS":{"Time":"2023-01-03T15:27:16","Uptime":"0T19:22:42","UptimeSec":69762,"Heap":142,"SleepMode":"Dynamic","Sleep":1,"LoadAvg":999,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"berg-media","BSSId":"74:42:7F:E1:DE:EA","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}

Weblog 4

15:30:17.850 CMD: weblog 4
15:30:17.852 SRC: WebConsole from 192.168.1.131
15:30:17.854 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 4, Data '4'
15:30:17.862 MQT: stat/Tasmota_tuinpaal/RESULT = {"WebLog":4}
15:30:18.486 BL9: Rx 55380D00891D0000000001163D480300F807000000003F0100000000C20100FE0300C8
15:30:19.489 BL9: Rx 55900C00171D00000000CF1D3D6503003908000000003F0100000000C20100FE0300AF
15:30:20.489 BL9: Rx 55D30D00701D0000000072203DB20300F307000000003F0100000000C40100FE030064
15:30:21.495 BL9: Rx 55CD0C006E1D0000000095223D6D03001F08000000003F0100000000C40100FE030060
15:30:22.489 BL9: Rx 55490C00881D00000000AB1E3D400400E607000000003F0100000000C30100FE03001F
15:30:23.489 BL9: Rx 55710C009B1D00000000C01A3D0303002408000000003F0100000000C10100FE0300D4
15:30:24.489 BL9: Rx 55560C00B21D00000000F61A3D1803000208000000003F0100000000C30100FE0300AD
15:30:25.485 BL9: Rx 55080E00461D00000000EE153D880300E907000000003F0100000000C30100FE03001C
15:30:26.489 BL9: Rx 553F0B00B51D000000002A183D4D03001E08000000003F0100000000C30100FE03003F
15:30:27.490 BL9: Rx 55DC0D001A1D000000005D213D4E0300DD07000000003F0100000000C10100FE030042
15:30:28.489 BL9: Rx 55FA0C00A51D0000000095153D5603000308000000003F0100000000C20100FE03003E
15:30:28.829 WIF: Checking connection...
15:30:29.491 BL9: Rx 55260C00351D000000005D163D5B0300CB07000000003F0100000000C20100FE0300ED
15:30:30.493 BL9: Rx 55C20B00C11D0000000011193D7C03001308000000003F0100000000C20100FE0300A5
15:30:31.485 BL9: Rx 55E40D005E1D000000001E153D550300D407000000003F0100000000C20100FE030042
15:30:32.487 BL9: Rx 55370C00961D00000000D7153D200300F007000000003F0100000000C10100FE030019
15:30:33.485 BL9: Rx 55CA0D00A61D00000000C1193D3E0300C70700000000400100000000C10100FE030091

arendst, 2023-01-03:
Great! Weblog 4 contains no info regarding domoticz.

Let's try another way. Enable ``Weblog 3`` and provide the console logging for 10 minutes.

sfromis, 2023-01-03:
Your STATUS3 indicates that you have `WattRes 0` in effect. Maybe you want to have decimlas there?

HB787, 2023-01-03:
15:50:24.524 CMD: Weblog 3
15:50:24.526 SRC: WebConsole from 192.168.1.131
15:50:24.528 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 3, Data '3'
15:50:24.536 MQT: stat/Tasmota_tuinpaal/RESULT = {"WebLog":3}
15:50:25.731 CFG: Saved, Count 2422, Bytes 4096
15:50:27.804 SRC: MQTT
15:50:27.806 CMD: Grp 0, Cmd 'POWER', Idx 2, Len 312, Pld 1, Data '1'
15:50:27.818 MQT: stat/Tasmota_tuinpaal/RESULT = {"POWER2":"ON"}
15:50:27.821 MQT: stat/Tasmota_tuinpaal/POWER2 = ON
15:50:27.833 RUL: POWER2#STATE=1 performs "RuleTimer1 3600"
15:50:27.834 SRC: Rule
15:50:27.836 CMD: Grp 0, Cmd 'RULETIMER', Idx 1, Len 4, Pld 3600, Data '3600'
15:50:27.845 MQT: stat/Tasmota_tuinpaal/RESULT = {"T1":3599,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
15:50:28.354 SRC: MQTT
15:50:28.356 CMD: Grp 0, Cmd 'POWER', Idx 2, Len 312, Pld 0, Data '0'
15:50:28.366 MQT: stat/Tasmota_tuinpaal/RESULT = {"POWER2":"OFF"}
15:50:28.370 MQT: stat/Tasmota_tuinpaal/POWER2 = OFF
15:50:34.408 MQT: tele/Tasmota_tuinpaal/STATE = {"Time":"2023-01-03T15:50:34","Uptime":"0T19:46:00","UptimeSec":71160,"Heap":151,"SleepMode":"Dynamic","Sleep":1,"LoadAvg":999,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"berg-media","BSSId":"74:42:7F:E1:DE:EA","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:50:34.428 MQT: domoticz/in = {"idx":8378,"nvalue":0,"svalue":"4;1964.3","Battery":100,"RSSI":7}
15:50:34.445 MQT: tele/Tasmota_tuinpaal/SENSOR = {"Time":"2023-01-03T15:50:34","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-27T10:07:35","Total":1.9643,"Yesterday":0.0898,"Today":0.0598,"Period":[0.03,0.00],"Power":[4.12,0.00],"ApparentPower":[4328.27,0.00],"ReactivePower":[4328.00,0.00],"Factor":[0.00,0.00],"Voltage":233,"Current":[18.546,0.000],"BL09XX":{"Temperature":27.7}},"ESP32":{"Temperature":48.9,"HallEffect":30},"TempUnit":"C"}
15:51:00.411 RUL: TIME#MINUTE|%VAR1% performs "backlog WebQuery http://192.168.1.170/ GET"
15:51:00.413 SRC: Rule
15:51:00.416 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 34, Pld -99, Data 'WebQuery http://192.168.1.170/ GET'
15:51:00.430 SRC: Backlog
15:51:00.432 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 25, Pld -99, Data 'http://192.168.1.170/ GET'
15:51:01.331 MQT: stat/Tasmota_tuinpaal/RESULT = {"WebQuery":"Done"}
15:51:01.351 RUL: WEBQUERY#DATA=DONE performs "Var1 3"
15:51:01.353 SRC: Rule
15:51:01.355 CMD: Grp 0, Cmd 'VAR', Idx 1, Len 1, Pld 3, Data '3'
15:51:01.363 MQT: stat/Tasmota_tuinpaal/RESULT = {"Var1":"3"}
15:51:04.400 MQT: tele/Tasmota_tuinpaal/STATE = {"Time":"2023-01-03T15:51:04","Uptime":"0T19:46:30","UptimeSec":71190,"Heap":146,"SleepMode":"Dynamic","Sleep":1,"LoadAvg":999,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"berg-media","BSSId":"74:42:7F:E1:DE:EA","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:51:04.419 MQT: domoticz/in = {"idx":8378,"nvalue":0,"svalue":"3;1964.3","Battery":100,"RSSI":7}
15:51:04.441 MQT: tele/Tasmota_tuinpaal/SENSOR = {"Time":"2023-01-03T15:51:04","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-27T10:07:35","Total":1.9643,"Yesterday":0.0898,"Today":0.0599,"Period":[0.03,0.00],"Power":[3.99,0.00],"ApparentPower":[4281.67,0.00],"ReactivePower":[4281.00,0.00],"Factor":[0.00,0.00],"Voltage":233,"Current":[18.383,0.000],"BL09XX":{"Temperature":27.7}},"ESP32":{"Temperature":49.4,"HallEffect":29},"TempUnit":"C"}
15:51:15.781 SRC: MQTT
15:51:15.783 CMD: Grp 0, Cmd 'POWER', Idx 2, Len 312, Pld 1, Data '1'
15:51:15.796 MQT: stat/Tasmota_tuinpaal/RESULT = {"POWER2":"ON"}
15:51:15.800 MQT: stat/Tasmota_tuinpaal/POWER2 = ON
15:51:15.814 RUL: POWER2#STATE=1 performs "RuleTimer1 3600"
15:51:15.815 SRC: Rule
15:51:15.817 CMD: Grp 0, Cmd 'RULETIMER', Idx 1, Len 4, Pld 3600, Data '3600'
15:51:15.826 MQT: stat/Tasmota_tuinpaal/RESULT = {"T1":3600,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
15:51:16.594 CFG: Saved, Count 2423, Bytes 4096
15:51:18.272 SRC: MQTT
15:51:18.274 CMD: Grp 0, Cmd 'POWER', Idx 2, Len 312, Pld 0, Data '0'
15:51:18.285 MQT: stat/Tasmota_tuinpaal/RESULT = {"POWER2":"OFF"}
15:51:18.288 MQT: stat/Tasmota_tuinpaal/POWER2 = OFF
15:51:21.603 CFG: Saved, Count 2424, Bytes 4096
15:51:34.401 MQT: tele/Tasmota_tuinpaal/STATE = {"Time":"2023-01-03T15:51:34","Uptime":"0T19:47:00","UptimeSec":71220,"Heap":150,"SleepMode":"Dynamic","Sleep":1,"LoadAvg":999,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"berg-media","BSSId":"74:42:7F:E1:DE:EA","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:51:34.417 MQT: domoticz/in = {"idx":8378,"nvalue":0,"svalue":"3;1964.4","Battery":100,"RSSI":7}
15:51:34.435 MQT: tele/Tasmota_tuinpaal/SENSOR = {"Time":"2023-01-03T15:51:34","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-27T10:07:35","Total":1.9644,"Yesterday":0.0898,"Today":0.0599,"Period":[0.04,0.02],"Power":[3.93,0.00],"ApparentPower":[4277.14,0.00],"ReactivePower":[4277.00,0.00],"Factor":[0.00,0.00],"Voltage":233,"Current":[18.352,0.000],"BL09XX":{"Temperature":27.9}},"ESP32":{"Temperature":49.4,"HallEffect":28},"TempUnit":"C"}
15:51:37.677 SRC: MQTT
15:51:37.679 CMD: Grp 0, Cmd 'POWER', Idx 2, Len 312, Pld 1, Data '1'
15:51:37.693 MQT: stat/Tasmota_tuinpaal/RESULT = {"POWER2":"ON"}
15:51:37.695 MQT: stat/Tasmota_tuinpaal/POWER2 = ON
[... cut ...]
16:02:34.434 MQT: tele/Tasmota_tuinpaal/SENSOR = {"Time":"2023-01-03T16:02:34","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-27T10:07:35","Total":1.9651,"Yesterday":0.0898,"Today":0.0607,"Period":[0.03,0.00],"Power":[3.91,0.00],"ApparentPower":[4267.29,0.00],"ReactivePower":[4267.00,0.00],"Factor":[0.00,0.00],"Voltage":234,"Current":[18.214,0.000],"BL09XX":{"Temperature":27.7}},"ESP32":{"Temperature":49.4,"HallEffect":29},"TempUnit":"C"}
16:03:00.408 RUL: TIME#MINUTE|%VAR1% performs "backlog WebQuery http://192.168.1.170/ GET"

HB787, 2023-01-03:
Strange WattRes comes with 2 in the console, changed it to 3 but it has no effect within Domoticz

arendst, 2023-01-03:
I see no energy data being send to Domoticz from Tasmota other then
```
16:02:34.417 MQT: domoticz/in = {"idx":8378,"nvalue":0,"svalue":"3;1965.1","Battery":100,"RSSI":7}
```
being the total energy (in kWh) with one decimal. This is the programmed Tasmota to Domoticz decimal size. If you expect more data than your Domoticz plugin is probably in error. The plugin is not written and maintained by Tasmota.

What I do is define a P1Smartmeter in Domoticz and add it's index to Tasmota at the P1SmartMeter sensor index. It will then send a message like:
```
16:31:04.896 MQT: domoticz/in = {"idx":384,"nvalue":0,"svalue":"0.0;228220.8;0.0;0.0;38;0","Battery":100,"RSSI":9}
16:31:04.900 MQT: tele/shellyplugs/SENSOR = {"Time":"2023-01-03T16:31:04","ANALOG":{"Temperature":33.5},"ENERGY":{"TotalStartTime":"2021-04-06T10:25:17","Total":223.47,"Yesterday":0.40,"Today":0.27,"Period":3.77,"Power":38.30,"ApparentPower":56.45,"ReactivePower":41.50,"Factor":0.68,"Voltage":225.8,"Current":0.250},"TempUnit":"C"}
```
In addition you could add a voltage and current device to Domoticz and send data to their indexes too.

In short, I don't know the domoticz tasmota plugin and cannot help you there.

HB787, 2023-01-04:
Found a solution in this rule, it is publishing the decimals in Domoticz this way.
Rule 1 
ON tele-energy#power DO var1 %value% ENDON
ON tele-energy#today DO publish domoticz/in {"idx":8378,"svalue":"%var1%;%value%"} ENDON

It works, however need to figure out how to pick up the meters with two channel devices like the Sonoff Dual R3

HB787, 2023-01-04:
I tried your plugin, it created more than 100 devices with German texts. It was a little too much, now the rule solution works well and I can have just want I need so consumption in watt and no other stuff like voltage etc. I would give you the tip, to not set the devices active if you create them.

closed 2023-01-04 by HB787

joba-1, 2023-01-04:
> not set the devices active if you create them

thanks, but I like to see all devices/sensors I have as soon as they appear on my broker and then deactivate those I don't need/want on the domoticz devices page.
Other way round is more difficult: decide what you want to see while not knowing what you have (I dont regularly check the devices page)...

And sorry about the german sensor names ... still searching for a good way to detect the target language.

sfromis, 2023-01-04:
> still searching for a good way to detect the target language.

@joba-1  If you issue the command `Status 4`, the first field of the `Features` part of the JSON is the value of the `#define LANGUAGE_LCID` from each of the language files, like when German has:
`#define LANGUAGE_LCID 1031`
and the JSON has the hex value of that number in:
...,"Features":["00000407", ...
This should be a very safe language detection, as long as you pick those codes from the [translation files](https://github.com/arendst/Tasmota/tree/development/tasmota/language).

English has `#define LANGUAGE_LCID 2057`, and so on... Looks like the LCID codes are from Microsoft "Language Code IDentifer", here's a straight page with codes: https://wiki.freepascal.org/Language_Codes

sfromis, 2023-01-04:
Did a scan....
```
af_AF    1078   af_AF.h - localization for Afrikaans - Afrikaans for Tasmota
bg_BG    1026   bg-BG.h - localization for Bulgaria - Bulgarian for Tasmota
ca_AD    1027   ca-AD.h - localization for Catalan - Països Catalans for Tasmota
cs_CZ    1029   cs-CZ.h - localization for Czech with diacritics - Czech for Tasmota
de_DE    1031   de-DE.h - localization for German - Germany for Tasmota
el_GR    1032   el-GR.h - localization for Greek - Greece for Tasmota
en_GB    2057   en-GB.h - localization for English - United Kingdom for Tasmota
es_ES    1034   es-ES.h - Tasmota localization for Spanish (International Spanish valid for all latinamerica)
fr_FR    1036   fr-FR.h - localization for French - France for Tasmota
fy_NL    1122   fy-NL.h - localization for Frysk - Nederland for Tasmota
he_HE    1037   he-HE.h - localization for Hebrew - Israel for Tasmota
hu_HU    1038   hu-HU.h - localization for Hungarian in Hungary for Tasmota
it_IT    1040   it-IT.h - localization for Italian - Italy for Tasmota
ko_KO    1042   ko-KO.h - localization for Korean - Korean for Tasmota
nl_NL    1043   nl-NL.h - localization for Dutch - Nederland for Tasmota
pl_PL    1045   pl-PL-d.h - localization for Polish with diacritics - Poland for Tasmota
pt_BR    1046   pt-BR.h - localization for Portuguese - Brazil for Tasmota
pt_PT    2070   pt-PT.h - localization for Portuguese - Portugal for Tasmota
ro_RO    2072   ro-RO.h - localization for Romanian - Romania and Moldova for Tasmota
ru_RU    1049   ru-RU.h - localization for Russian - Russia for Tasmota
sk_SK    1029   sk-SK.h - localization for Slovak with diacritics - Slovak for Tasmota
sv_SE    1053   sv-SE.h - localization for Swedish - Svenska for Tasmota
tr_TR    1055   tr-TR.h - localization for Turkish - Turkey for Tasmota
uk_UA    1058   uk-UA.h - localization for Ukrainian - Ukraine for Tasmota
vi_VN    1066   vi-VN.h - localization for Vietnam for Tasmota
zh_CN    2052   zh-CN.h - localization for Chinese (Simplified) - China for Tasmota
zh_TW    1028   zh-TW.h - localization for Chinese (Traditional) - Taiwan for Tasmota
```