arendst/Tasmota #17578: SMI Counter: Errors in pulse rate calculation (using interrupt debounce mode)

sensor-freak, 2023-01-03:
### PROBLEM DESCRIPTION
There are some errors in pulse rate calculation of the SMI (Smart Meter Interface) Counter code.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Generic/NodeMCU
- [x] Tasmota binary firmware version number used: 12.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PlatformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
10:39:39.389 CMD: Backlog Template; Module; GPIO 255
10:39:39.456 MQT: stat/sonoff/039/RESULT = {"NAME":"ESP8266-counter","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
10:39:39.667 MQT: stat/sonoff/039/RESULT = {"Module":{"18":"Generic"}}
10:39:39.923 MQT: stat/sonoff/039/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [ ] Provide the output of this command: `Status 0`:
```lua
10:41:06.920 CMD: status 0
10:41:06.928 MQT: stat/sonoff/039/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota ESP8266 Counter","FriendlyName":["TasmotaCounter"],"Topic":"sonoff/039","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:41:06.934 MQT: stat/sonoff/039/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"4T20:39:34","StartupUTC":"2022-12-29T13:01:32","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2022-12-16T18:40:36","SaveCount":96,"SaveAddress":"F4000"}}
10:41:06.938 MQT: stat/sonoff/039/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T18:33:47","Boot":26,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"387/699"}}
10:41:06.942 MQT: stat/sonoff/039/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
10:41:06.954 MQT: stat/sonoff/039/STATUS4 = {"StatusMEM":{"ProgramSize":663,"Free":340,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","879AC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6,53"}}
10:41:06.962 MQT: stat/sonoff/039/STATUS5 = {"StatusNET":{"Hostname":"sonoff-039-1337","IPAddress":"192.168.xx.xx","Gateway":"192.168.xx.xx","Subnetmask":"255.255.255.0","DNSServer1":"192.168.xx.xx","DNSServer2":"0.0.0.0","Mac":"A0:20:A6:xx:xx:xx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:41:06.967 MQT: stat/sonoff/039/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xx.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_158539","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:41:06.973 MQT: stat/sonoff/039/STATUS7 = {"StatusTIM":{"UTC":"2023-01-03T09:41:06","Local":"2023-01-03T10:41:06","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
10:41:06.980 MQT: stat/sonoff/039/STATUS10 = {"StatusSNS":{"Time":"2023-01-03T10:41:06","Water1":{"Total_in":0.00000,"Current_in":0.00000},"Water2":{"Total_in":4.94952,"Current_in":0.30000}}}
10:41:06.986 MQT: stat/sonoff/039/STATUS11 = {"StatusSTS":{"Time":"2023-01-03T10:41:06","Uptime":"4T20:39:34","UptimeSec":419974,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"5C:49:79:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
```

### TO REPRODUCE
- Configure a smart meter device for counter mode with negative debounce time (so it will count using interrupts)
- Provide pulses on the selected counter port (pulse rate and pulse count will be displayed in main screen)
- Stop pulses: pulse rate will show the last value, and never return to 0

### EXPECTED BEHAVIOUR
- The pulse rate should return to 0 when there input signal does not show any pulses

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
In file `xsns_53_sml.ino:3472` (inside method `SML_Counter_Poll`) the method `InjektCounterValue` is used to generate new values. However, this call is guarded by `if (sml_counters[cindex].sml_cnt_updated)`, which indicates that it gets executed when the input signal changes (to be exact: only when at least one falling edge has been detected since the last execution). This behavior is the reason that the pulse rate value will never return to 0.

Since the method `SML_Counter_Poll` seems to be called cyclically to update the values transmitted e.g. by MQTT, the obvious solution would be to simply provide the fixed value `0` for the pulse rate in the else branch. Unfortunately I was not able to test this change yet, and I'm not sure if there might be any side effects that I do not foresee at moment.

Furthermore I'm not happy with the first argument to `InjektCounterValue`. This is supposed to be a meter number, but actually the argument given is `sml_counters[cindex].sml_cnt_old_state`. I do not see a reason why the existing variable `meters` should not be used at this point, like it is used in all other calls to that method.

**(Please, remember to close the issue when the problem has been addressed)**

sensor-freak, 2023-01-03:
There is another possibly problematic line in the code of method `InjektCounterValue`:
```
  snprintf((char*)&smltbuf[meter][0], SML_BSIZ, "1-0:1.7.0*255(%d.%d)", dec, frac);
```

That code looks like it should create fixed point representation from a float value. The float value value has been split in decimal part and fractional part (base 1/1000) (as integers). The resulting "fixed point" number would not be correct, when the fractional part is less than 100, since there will be no '0's prepended in the result:

Value 12.03456 -> dec = 12, frac = 34 -> results in "...12.34" instead of "...12.034"

gemu2015, 2023-01-04:
please check this adapted version.
there is now a timeout to reset the value when the counter pulses stop.
(currently set to 5 seconds)
also channel math (delta mode) is now working with counters

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/10342404/xsns_53_sml.ino.zip)

sensor-freak, 2023-01-04:
The changes do have a positive effect on the results, but there are still some issues. 

Especially with fast changing counters the resolution of the rate values are not as i expected them to be. If for example you have connected a 200Hz signal, the pulse length is 5ms, resulting in a rate value of 12000 1/min. One has to reduce the frequency below 166Hz in order to get a different rate value (10000 1/min). Rate values in between such two results are impossible, due to the implementation. 

I'm currently thinking about a different implementation, using the pulse count in an interval (1s?) instead of the last pulse length. But first I have to check for the effects with slow rates, e.g. with signal frequencies below 1Hz.

gemu2015, 2023-01-05:
this is trivial math, with milliseconds timer and low values the resolution must be bad.
i do not have such a meter, what are the ranges of pulse frequency in normal use ?
anyhow using pulses will never give satisfying results i guess.
i will pr these changes soon.
i propose if you find a much better solution you may do your own pull request.

Jason2866, 2023-01-10:
Implemented with #17628

closed 2023-01-10 by Jason2866