arendst/Tasmota #17579: EnergyTotal 0 not working

janjanson84, 2023-01-03:
Hello,

when trying to reset the Total Energy with command "EnergyTotal 0", the Counter is not set to zero but to 436 kWh.

Tasmota Version is 13.3.1

11:54:18.611 CMD: EnergyTotal 0
11:54:18.620 MQT: stat/meter1/RESULT = {"EnergyTotal":{"Total":[2.092,243.196,191.497],"Yesterday":[5.161,0.937,0.388],"Today":[2.092,0.181,0.126]}}
11:54:27.416 MQT: tele/meter1/STATE = {"Time":"2023-01-03T11:54:27","Uptime":"0T00:04:49","UptimeSec":289,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"X","BSSId":"XXXXXXXXXXX","Channel":6,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:54:27.430 MQT: tele/meter1/SENSOR = {"Time":"2023-01-03T11:54:27","ENERGY":{"TotalStartTime":"2023-01-03T00:00:00","Total":436.786,"Yesterday":6.486,"Today":2.400,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[0.000,0.001,0.302],"Period":[1,0,0],"Power":[298,7,0],"ApparentPower":[332,9,8],"ReactivePower":[-144,-6,-6],"Factor":[0.90,0.77,0.61],"Frequency":50,"Voltage":[234,235,235],"Current":[1.415,0.039,0.035]}}

arendst, 2023-01-03:
Because with ``EnergyTotal 0`` you only reset phase 1. phase 2 and 3 still contain their values summing up to 436kWh.

You'll need to reset phase 2 and 3 too with commands ``EnergyTotal2 0`` and ``EnergyTotal3 0``.

janjanson84, 2023-01-03:
> Because with `EnergyTotal 0` you only reset phase 1. phase 2 and 3 still contain their values summing up to 436kWh.
> 
> You'll need to reset phase 2 and 3 too with commands `EnergyTotal2 0` and `EnergyTotal3 0`.

Aaaah, that worked fine, thank you! The Documentation wasn't that clear for me, because I think in version < 10.x only one command was neccessary.

closed 2023-01-03 by Jason2866