arendst/Tasmota #17599: Serial bridge debugging for customized build

heligan, 2023-01-04:
Hi,

I am trying to debug my customized build (based on development branch @ 12.3.*) via serial bridge, because some weird issue observed. I have two types of esp8266 dev boards:

* wemos d1 mini
* nodemcu v3 (not so sure but seems to be like this https://a.aliexpress.com/_m0eGSwI)

It seems the build works fine with the wemos d1 mini boards. I had successful OTA updates (from a customized build based on 9.2 firmwares) and they work in the same way as before. However, after the OTA update for nodemcu boards, I could not find them on my wifi network any more (no AP either). I tried the official release from 12.2. It seems they can be found as AP but as soon as I add them into my wifi, the same thing will happen (can not be found in router's client list any more and AP mode remains off). When I say AP mode off, basically there is no wifi endpoint with the name tasmota_<mac>_....

At first I thought perhaps it's just one nodemcu board but the same thing happens to a second board of very similar setup. So it's reproducible. But since I have no way to connect to them via network (no access to web UI or mqtt), now the best I can do is to get something from the serial bridge.

I am using minicom with

* hardware flow control off (`pu rtscts           No`)
* software flow control off (`pu xonxoff          No`)
* not sure whether this is useful but I enabled `pu addcarreturn     Yes`
* also turned on local echo: `pu localecho        Yes`

After connecting to one of the nodemcu board (via USB port), I can see something like this

```
00:00:00.002 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F9, Count 29
00:00:00.056 QPC: Count 1
00:00:00.063 Project tasmota - Tasmota Version 12.3.1.2(...)
00:00:00.217 WIF: WifiManager active for 3 minutes
00:00:00.725 HTP: Web server active on tasmota-5D37BF-6079 with IP address
00:00:06.917 QPC: Reset
```

So it seems this call ended up with no valid ip address:

https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_01_9_webserver.ino#L613-L617

I tried to type some commands like `Status` or `IpAddress` but it seems there is nothing returned from the minicom (unlike webui). Not sure it's related to my minicom's setup or my customized build but so far I am stuck.

Any suggestions?

sfromis, 2023-01-04:
One risk factor is that the power source lacks the juice to facilitate Wifi. Another is that your customized build is doing something "bad enough" to cause complete failure. That would usually result in serial output, but that's no guarantee.... I'd suggest starting with a build without customization, and if that works, add your stuff in small steps.

Anyway, this should probably have been posted in discussions, not opened as an "issue" against Tasmota (reasons to suspect a bug).

heligan, 2023-01-04:
Sure, I will move this to discussion if I found out something new.

heligan, 2023-01-05:
This is still quite suspicious as a bug.

I rolled back to this commit 065396ff73dc1529e5242602c54dc11d17886a0d (12.2.0.1) and the wifi connection issue is gone. I tried to see whether the current development head (commit 7104d10d61a52e2e2f55f90ed4c95a1c556c9ae9) addressed this but unfortunately no.

Jason2866, 2023-01-05:
Closing since issue is not concrete enough. For unclear scenarios please use discussions.

heligan, 2023-01-05:
I updated the original post with the commit number bbde894628832583a52648ebbe3f95bd86438b09 that I found out that might be related to the issue. See whether you can reopen this issue?

arendst, 2023-01-06:
Re-openend as I can reproduce this on my ESP32-C3.

After a ``reset 1`` on latest dev release it indeed doesn't find the AP's IP address and reports
```
00:00:00.001 HDW: ESP32-C3 
00:00:00.016 UFS: FlashFS mounted with 304 kB free
00:00:00.028 CFG: Loaded from File, Count 2
00:00:00.033 QPC: Count 1
00:00:00.222 BRY: Berry initialized, RAM used=5191 bytes
00:00:00.236 Project tasmota - Tasmota Version 12.3.1.3(tasmota)-2_0_6(2023-01-06T10:40:16)
00:00:00.486 WIF: WifiManager active for 3 minutes
00:00:01.130 HTP: Web server active on tasmota-C24D0C-3340 with IP address 
00:00:06.898 QPC: Reset
```
as OP noticed.

Major changes have been done regarding IPv4/IPv6 and getting the IP address is now done by function ``IPGetListeningAddress()`` which seems to fail in this case.

Investigating...

closed 2023-01-06 by arendst

heligan, 2023-01-06:
Thank you both for the quick investigation and fix, @arendst and @s-hadinger!

I synced to the head of the development branch today and the customized builds can connect to wifi again!

heligan, 2023-01-06:
I am trying to understand how the fix works. Looking at https://github.com/arendst/Tasmota/commit/33d6dbed3dccaf729a8587c7d3b959962de4c461, the added `WiFi.softAPIP` to the `USE_IPV6` compilation condition. However, I don't think mine is using this. 

After a follow-up commit, https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support_wifi.ino#L758-L776

```
bool WifiGetIP(IPAddress *ip) {
  if ((uint32_t)WiFi.localIP() != 0) {
    if (ip != nullptr) { *ip = WiFi.localIP(); }
    return true;
  }
  if ((uint32_t)WiFi.softAPIP() != 0) {
    if (ip != nullptr) { *ip = WiFi.softAPIP(); }
    return true;
  }
#ifdef USE_IPV6
  IPAddress lip;
  if (WifiGetIPv6(&lip)) {
    if (ip != nullptr) { *ip = lip; }
    return true;
  }
  if (ip != nullptr) { *ip = IPAddress(); }
#endif // USE_IPV6
  return false;
}
```

seems to be a little bit cleaner to me but also noticed this function, https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support_wifi.ino#L508-L513

```
bool WifiGetIPv4(IPAddress *ip)
{
  uint32_t wifi_uint = (uint32_t) WiFi.localIP();
  if (ip != nullptr) { *ip = wifi_uint; }
  return wifi_uint != 0;
}
```

which doesn't check `WiFi.softAPIP`. So still seems to be quite hard to tell there is still some potential bug.

arendst, 2023-01-06:
You're right. Was noticed too and already fixed in another merged PR.

PR #17619 (2023-01-06): No IP address shown when in AP mode (#17599)
## Description:

Now IPv4 is correctly displayed when in AP mode.

**Related issue (if applicable):** fixes #17599

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blo [... cut]