arendst/Tasmota #17615: Script: output from WifiConfig don't get into Vars

nikito7, 2023-01-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

```js
>D 48

ver=10230
PF="Factor"
AP="Potência"
TE="Energia"
date=""
time=""
clk=""
old=""
wfc=""
wfp=0
cnt=0
wtd=0
hh=0
mm=0
ss=0
m:ipwrm=0 50
m:epwrm=0 50
ipwr=0
epwr=0
strm="cnt0"
fheap=0
;
ws=""
node=""
key=""
p:ikw=0
p:ekw=0
fr=0
res=0
;
lp1y=0
lp1m=0
lp1d=0
lp1hh=0
lp1mm=0
lp3i=0
lp6e=0
lps=""
lpf=""
;
p:ikwo=0
p:ekwo=0
tmp=0
chs=""
chf=""
imp=0
exp=0
;
pv1w=0
p:pv1k=0
p:pv1ko=0
sol=0

>BS

tper=20
smlj=0

=>SerialLog 0

>E

wfc=WifiConfig#?
wfp=WifiPower

>S

time=st(tstamp T 2)
date=st(tstamp T 1)
fheap=heap/1024

if cnt==25
then
=>WifiConfig
=>WifiPower
smlj=1
tper=15
endif

if cnt==40
then
=>UfsRun discovery2.txt
endif

if cnt<99
then
cnt+=1
print cnt=%0cnt%
endif

>W

@<b>NTP </b> %date% %time% <b> Heap </b> %1fheap%
@<b>Vars </b> cnt=%0cnt% tper=%0tper% smlj=%0smlj% ver=%0ver%
@<b>Vars </b> wtd=%0wtd% clk=%0clk% old=%0old%
@<b>Wifi </b> %wfc% <b> Power </b> %0wfp% <b> Topic </b> %topic%
@<br>
<br>
Solar{m}%1pv1w% W
Solar{m}%1pv1k% kWh
<br>
<a href="/ufs/%lpf%">%lpf%</a>{m}<a href="/ufs/charts.html">Charts</a>
<br>
; eof
```

Log:
```
17:47:24.437 cnt=23
17:47:25.440 cnt=24
17:47:26.399 cnt=25
17:47:26.797 WIF: Checking connection...
17:47:27.402 Script: performs "WifiConfig"
17:47:27.403 SRC: Rule
17:47:27.406 CMD: Grp 0, Cmd 'WIFICONFIG', Idx 1, Len 0, Pld -99, Data ''
17:47:27.415 MQT: stat/edpbox3/RESULT = {"WifiConfig":{"4":"Retry"}}
17:47:27.418 Script: performs "WifiPower"
17:47:27.420 SRC: Rule
17:47:27.423 CMD: Grp 0, Cmd 'WIFIPOWER', Idx 1, Len 0, Pld -99, Data ''
17:47:27.429 MQT: stat/edpbox3/RESULT = {"WifiPower":"17.0"}
17:47:27.433 cnt=26
17:47:28.413 MQT: tele/edpbox3/STATE = {"Time":"2023-01-05T17:47:28","Uptime":"0T00:00:27","UptimeSec":27,"Vcc":3.557,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"***","BSSId":"(*(","Channel":1,"Mode":"11n","RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:05"}}
17:47:28.423 cnt=27
17:47:28.721 CFG: Saved to flash at FC, Count 10415, Bytes 4096
17:47:29.528 cnt=28
17:47:30.441 cnt=29
```

12.2.0
[image]

Dev
[image]

But it works if I run WifiConfig manually

**(Please, remember to close the issue when the problem has been addressed)**

nikito7, 2023-01-05:
Minimal script 

```
>D 48

wfc=""
wfp=0
cnt=0

>BS

tper=20
smlj=0

=>SerialLog 0

>E

wfc=WifiConfig#?
wfp=WifiPower

>S

if cnt==25
then
=>WifiConfig
=>WifiPower
endif

if cnt<99
then
cnt+=1
print cnt=%0cnt%
endif

>W

@<b>Wifi </b> %wfc% <b> Power </b> %0wfp% <b> Topic </b> %topic%
@<br>
<br>

; eof
```

barbudor, 2023-01-05:
@gemu2015 has recently made some changes in the processing of JSON messages which may have impacted this use-case

gemu2015, 2023-01-06:
these are recursive calls, calls that may call scripting by generating an event.
these calls are blocked by default. due to a regression this blocking was not in place in a recent version. this is now fixed.

if you want to enable such a call simply write

+>WifiConfig

closed 2023-01-06 by nikito7