arendst/Tasmota #17624: MQTT Full Topic cannot be set to %topic%/%prefix%/

fritz47110815, 2023-01-06:
### PROBLEM DESCRIPTION
MQTT Full Topic cannot be set to %topic%/%prefix%/ 
After the restart Full Topic will be set to NaN

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Open Configuration MQTT and file %topic%/%prefix%/ under Full Topic. Allow restart

### EXPECTED BEHAVIOUR
Full Topic will be set to %topic%/%prefix%/

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-01-07:
All my devices have `%topic%/%prefix%/` and work with that.

You should always fill in the documentation parts of the template to show what happens.

fritz47110815, 2023-01-07:
I am sorry - it was a misconfiguration of the MQTT broker.

fritz47110815, 2023-01-07:
Misconfiguration - no issue

closed 2023-01-07 by fritz47110815