arendst/Tasmota #17642: SetOption47 causing misleading information

AleXSR700, 2023-01-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hello,
first of: thank you for the SetOption47! Excellent addition to tasmota!

But there is a bug or at least an unfortunate implementation.

The SetOption47 combined with PowerOnState 1 or 3 causes the device (e.g. Shelly 1PM, Shelly 2.5 or Gosund SP1/SP112 = SHP5) to show that the power output is turned on. (and the status LED also says that the outlet is turned on) However, despite the power officially being turned on, the SetOption47 causes the power to really be off.

So SetOption47 is working but causes incorrect information.

Is:
`SetOption47 120` + `PowerOnState 1`
=> WebUI says socket is turned, status LED says socket is turned on but socket does not supply power

Should be:
`SetOption47 120` + `PowerOnState 1`
=> WebUI should say socket is turned off (and LED be turned off also)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____Shelly 1PM, Shelly 2.5, Gosund SP1, Gosund SP112
- [ x] Tasmota binary firmware version number used: _____12.3.1
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: _____OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set `SetOption47 120` + `PowerOnState 1`
Unplug power socket. Plug in power socket. Plug in lamp and wait for 2 min.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

WebUI power "off" & status LED power "off"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-01-11:
You're right.

The goal of SO47 is to reduce power at AC power on by delaying power on of relays. This way the inrush current will be less than if the relays would be powered on immediatly after AC power on.

A side effect of the current implementation is that indeed all state messages are reporting the expected relay state while the relays are in fact powered off. As soon as the SO47 timer is expired the relay state will be set as requested at power on. Intermediate user changes are taking into effect too.

So once the timer is expired the relay(s) reflect the current state.

The functionality is a tool to debug devices having designed their power supply too tight. It was never meant to be power on delay tool. Use different commands for that functionality.

What you see is as designed.

sfromis, 2023-01-11:
You can setup actions delayed after boot using a boot rule creating a RuleTime to expire after the desired latency.

Jason2866, 2023-02-01:
Closing since answered. Function is as designed.