arendst/Tasmota #17650: Regression from 12.2.0 to 12.3.1 : VoltageSet missmatches PowerSet

NiklausS, 2023-01-08:
### PROBLEM DESCRIPTION
after upgrading to 12.3.1 when setting the voltage the power gets set and visa versa. I've also tried with VoltageSet1

backlog CurrentSet 2000 ; VoltageSet 230 ; PowerSet 460 -> will result in 
19:02:52.595 CMD: backlog CurrentSet 2000 ; VoltageSet 230 ; PowerSet 460

19:02:52.619 MQT: stat/T-Wemos-D1/RESULT = {"CurrentSetCal":200000}
19:02:52.828 MQT: stat/T-Wemos-D1/RESULT = {"VoltageSetCal":23000}
19:02:53.036 MQT: stat/T-Wemos-D1/RESULT = {"PowerSetCal":46000}
...
19:04:03.458 MQT: tele/T-Wemos-D1/SENSOR = {"Time":"2023-01-08T19:04:03","ENERGY":{"TotalStartTime":"2023-01-08T14:23:49","Total":0.582,"Yesterday":0.000,"Today":0.582,"Period":0,"Power":230,"ApparentPower":920,"ReactivePower":891,"Factor":0.25,"Frequency":50,"Voltage":460,"Current":2.000}}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:05:49.339 CMD: Backlog Template; Module; GPIO 255
19:05:49.362 MQT: stat/T-Wemos-D1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:05:49.565 MQT: stat/T-Wemos-D1/RESULT = {"Module":{"0":"Generic"}}
19:05:49.782 MQT: stat/T-Wemos-D1/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"480":"Buzzer"},"GPIO13":{"5729":"Option A2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"416":"PWM1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-01-08:
As you don't give much details, and your GPIO configuration doesn't seem to point to a real power measument device, I assume that you are using the dummy driver
This has alerady been fixed https://github.com/arendst/Tasmota/commit/10af47f8f41c5cd01f4ddef2329969b169afef57
Please update to latest development version

NiklausS, 2023-01-08:
Hi Barbudor,
indeed, is use "GPIO13":{"5729":"Option A2"} - the dummy device. 
Thank you very much for the quick reply.

Cheers 
 Nik

closed 2023-01-08 by NiklausS