arendst/Tasmota #17652: conflict between PWM output and AM2301 sensor reading

scimmiettarossa, 2023-01-08:
### PROBLEM DESCRIPTION
I have a Sonoff DualR2 with the last Tasmota (12.3.1); I have hacked it a bit using additional GPIOs with respect to the standard two buttons / two relays:

stat/garageswitch/RESULT = {"NAME":"DualR2_ext","GPIO":[33,416,0,1216,0,225,163,32,224,320,0,0,0,0],"FLAG":0,"BASE":39}
GPIO0: Button 2 [33] (default)
GPIO1: PWM 1 [416], added to drive a led stripe
GPIO3: AM2301 [1216], added to monitor temperature and humidity with a DHT22
GPIO5: Relay 2 [225] (default)
GPIO9: Switch 4 [163], added to monitor a door contact
GPIO10: Button 1 [32] (default]
GPIO12: Relay 1 [224] (default)
GPIO13: Led_i 1 [320] (default)

everything works fine, if the PWM output is off, while if I set it to any value >0 (via web GUI or via domoticz or via console with the dimmer command), the readings from the DHT22 goes to "null" or "0".
If I set the PWM back to 0, after a few seconds the readings are back

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff DualR2
- [x] Tasmota binary firmware version number used: 12.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
20:47:11.321 CMD: Backlog Template; Module; GPIO 255
20:47:11.405 MQT: stat/garageswitch/RESULT = {"NAME":"DualR2_ext","GPIO":[33,416,0,1216,0,225,163,32,224,320,0,0,0,0],"FLAG":0,"BASE":39}
20:47:11.623 MQT: stat/garageswitch/RESULT = {"Module":{"0":"DualR2_ext"}}
20:47:11.877 MQT: stat/garageswitch/RESULT = {"GPIO0":{"33":"Pulsante2"},"GPIO1":{"416":"PWM1"},"GPIO2":{"0":"Nessuno"},"GPIO3":{"1216":"AM2301"},"GPIO4":{"0":"Nessuno"},"GPIO5":{"225":"Relè2"},"GPIO9":{"163":"Switch4"},"GPIO10":{"32":"Pulsante1"},"GPIO12":{"224":"Relè1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"Nessuno"},"GPIO15":{"0":"Nessuno"},"GPIO16":{"0":"Nessuno"},"GPIO17":{"0":"Nessuno"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
20:48:26.671 CMD: Backlog Rule1; Rule2; Rule3
20:48:26.761 MQT: stat/garageswitch/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":73,"Free":438,"Rules":"ON switch4#state DO publish domoticz/in {\"idx\":36,\"nvalue\":%value%} ENDON"}}
20:48:26.982 MQT: stat/garageswitch/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:48:27.232 MQT: stat/garageswitch/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:49:19.451 CMD: status 0
20:49:19.458 MQT: stat/garageswitch/STATUS = {"Status":{"Module":0,"DeviceName":"garageswitch","FriendlyName":["garageswitch_1","garageswitch_2","garageswitch_3"],"Topic":"garageswitch","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:49:19.468 MQT: stat/garageswitch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-IT.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:44","StartupUTC":"2023-01-08T19:38:35","Sleep":50,"CfgHolder":4617,"BootCount":95,"BCResetTime":"2023-01-07T19:54:19","SaveCount":502,"SaveAddress":"F8000"}}
20:49:19.474 MQT: stat/garageswitch/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:34:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"398/699"}}
20:49:19.481 MQT: stat/garageswitch/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["roxale_ext",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]}}
20:49:19.497 MQT: stat/garageswitch/STATUS4 = {"StatusMEM":{"ProgramSize":630,"Free":372,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000410","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
20:49:19.505 MQT: stat/garageswitch/STATUS5 = {"StatusNET":{"Hostname":"roxoff-G1","IPAddress":"<masked>","Gateway":"<masked>","Subnetmask":"255.255.255.0","DNSServer1":"<masked>","DNSServer2":"<masked>","Mac":"<masked>","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
20:49:19.511 MQT: stat/garageswitch/STATUS6 = {"StatusMQT":{"MqttHost":"<masked>","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A4FE88","MqttUser":"<masked>","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:49:19.519 MQT: stat/garageswitch/STATUS7 = {"StatusTIM":{"UTC":"2023-01-08T19:49:19","Local":"2023-01-08T20:49:19","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:10"}}
20:49:19.527 MQT: stat/garageswitch/STATUS10 = {"StatusSNS":{"Time":"2023-01-08T20:49:19","Switch4":"OFF","AM2301":{"Temperature":12.4,"Humidity":60.3,"DewPoint":4.9},"TempUnit":"C"}}
20:49:19.549 MQT: stat/garageswitch/STATUS11 = {"StatusSTS":{"Time":"2023-01-08T20:49:19","Uptime":"0T00:10:44","UptimeSec":644,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Dimmer":38,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"roxale_ext","BSSId":"98:DA:C4:D4:12:50","Channel":2,"Mode":"11n","RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
20:57:06.262 CMD: weblog 4
20:57:06.265 SRC: WebConsole from 192.168.1.20
20:57:06.267 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 4, Data '4'
20:57:06.273 MQT: stat/garageswitch/RESULT = {"WebLog":4}
20:57:06.456 DHT: Pin3 read 025E007BDB                ---> POWER 3 OFF (PWM off) --> DHT22 readings ok
20:57:10.450 DHT: Pin3 timeout waiting for low pulse
20:57:14.452 DHT: Pin3 timeout waiting for low pulse
[... 28 lines cut ...]
20:58:14.452 DHT: Pin3 timeout waiting for low pulse
20:58:18.452 DHT: Pin3 timeout waiting for low pulse
20:58:19.001 WIF: Controllo connessione...
20:58:22.440 DHT: Pin3 read 025D007BDA
etc...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

switch PWM on and off

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

non conflict between GPIO1 (PWM and GPIO3 (sensor)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-01-08:
Does not look like a conflict between GPIO1 and GPIO3. As the one-wire communication with the DHT22 (not a good sensor choice) is very fickle about timing, and software PWM on ESP8266 pushes the MCU hard, failure is not surprising.

scimmiettarossa, 2023-01-09:
thanks, @sfromis ... do you think that with other sensors I can overcome this issue?  any suggestion? 

Until now I used a sonoff TH with Si7021, using a PWM output as well, with no issue: is it worth to try  to use the Si7021? I have also available a DHT11 (but very inaccurate, and I think it shares the same logic), or even DS18B20 (without humidity reading, but it is not a big issue): do you think any of them is more "robust" in behaviour?

sfromis, 2023-01-09:
What I'd prefer would be an I2C sensor, instead of any of those. Point being that the I2C bus communication is more robust than those one-wire protocols. But if you have good experiences with the Sonoff SI7021 wrapped as non-I2C - whatever works for you.

scimmiettarossa, 2023-01-09:
I leave the issue open until I try  with alternative sensors, then I'll let you know

thanks again