arendst/Tasmota #17658: Feautre request for PID-Control: Adjustable cycle time

Wintespe, 2023-01-09:
### Discussed in https://github.com/arendst/Tasmota/discussions/17656

<div type='discussions-op-text'>

<sup>Originally posted by **Wintespe** January  9, 2023</sup>
This feautre request refers to the PID-feature implemented in xdrv_49_pid.ino.

### Motivation

The PID (Proportional Integral Derivative)-feature implements a temperature controller into the Tasmota software.

The algorithm allows a relay to be used in a time proportioned way. For this it is using the Time Proportioned output extension, so called TimeProp.

TimeProp receive the power value - PidPower – from Pid and convert it into a PWM related signal.

[image]

During the time **Tc * PidPower** the relay is on and in the time **Tc * ( 1 - PidPower )** it is off. The signal is a “pulse width modulated signal” (PWM) with frequency 1 / Tc.

The cycle time Tc to be selected depends on the size of the room, the heating output, the objects in the room and much more. This means each room has its own optimal characteristic cycle time. Tc also depends on the type of the relay. A solid state relay is not critical but a mechanical relay is very limited.

The optimal time for each Tc must therefore be determined. Unfortunately, the TimeProp module does not have a variable parameter for this. Tc can only be changed by compiling and flashing the software. If several controllers are required in your home, a separate software version must be created for each. All these shortcomings are the reason for this for this feature request.

To circumvent the problem, a new command - PidTc - is required in the Pid Controller. The following section describes the implementation.

### Environment

Tasmota version 11.0.1 und above.
The Hardware is a NodeMcu with a DS18B20 temperatur sensor.

### Implementation

File: **xdrv_49_pid.ino**

**Define the command name**

```
#define D_PRFX_PID "Pid"
#define D_CMND_PID_SETPV "Pv"
...
#define D_CMND_PID_SETCYCLETIME "Tc"
```

**Extend the command list**

```
const char kPIDCommands[] PROGMEM = D_PRFX_PID "|" // first entry is prefix
  D_CMND_PID_SETPV "|"
  D_CMND_PID_SETSETPOINT "|"
  ...
  D_CMND_PID_SETMAX_INTERVAL "|"
#if defined(PID_USE_TIMPROP) && (PID_USE_TIMPROP > 0)
  D_CMND_PID_SETUPDATE_SECS "|"
  D_CMND_PID_SETCYCLETIME;
#else
  D_CMND_PID_SETUPDATE_SECS;
#endif
  ;

```

**and the command jump table**

```
// list of pointers to commands
void (* const PIDCommand[])(void) PROGMEM = {
  &CmndSetPv,
  &CmndSetSp,
  ...
  &CmndSetMaxInterval,
#if defined(PID_USE_TIMPROP) && (PID_USE_TIMPROP > 0)
  &CmndSetUpdateSecs,
  &CmdSetTc
#else
  &CmndSetUpdateSecs
#endif
  };
```

**Implement the command**

```
#if defined(PID_USE_TIMPROP) && (PID_USE_TIMPROP > 0)
void CmdSetTc(void){
  if (XdrvMailbox.data_len > 0) TimepropSetCycleTime(PID_USE_TIMPROP-1, CharToFloat(XdrvMailbox.data));
  ResponseCmndNumber(TimepropGetCycleTime(PID_USE_TIMPROP-1));
}
#endif // PID_USE_TIMPROP
```

**Optional show in JSON**

```
void PIDShowValues(void) {
...
// actual cycle time
#if defined(PID_USE_TIMPROP) && (PID_USE_TIMPROP > 0)
  i_buf = TimepropGetCycleTime(PID_USE_TIMPROP-1);
  ResponseAppend_P(PSTR("\"PidTc\":%d,"), i_buf);
#endif
```
File: **xdrv_48_timeprop.ino**

**Implement set and get cycle time functions in the driver**

```
/*  call this from elsewhere if required to set / get cycleTime */
/* index specifies which one, 0 up */
void TimepropSetCycleTime(int index, int cycleTime) {
  if (index >= 0  &&  index < TIMEPROP_NUM_OUTPUTS){
    Tprop.timeprops[index].setCycleTime(cycleTime);
  }
}
```

```
int TimepropGetCycleTime(int index) {
  if (index >= 0  &&  index < TIMEPROP_NUM_OUTPUTS){
    return Tprop.timeprops[index].getCycleTime();
  }
  return -1; // on Error -1
}
```

File: **timeprop.h**

**Implement set and get cycle time member function in the related class**

```
  /* set / get cycleTime */
  void setCycleTime(int cycleTime);
  int getCycleTime();
```

File: **timeprop.cpp**

```
/* set / get cycleTime */
void Timeprop::setCycleTime(int cycleTime) {
  m_cycleTime = cycleTime;
}
```

```
int Timeprop::getCycleTime() {
  return m_cycleTime;
}
```

### Results

Time / temperature diagram of a step function (setpoint 25 → 30 °C) Tc = 160  and the related time / PidPower diagram. 

[image]

It is easy to see that the system cools down too much in the off time because Tc was chosen too large. It becomes unstable.

Time / temperature diagram of a step function (setpoint 25 → 30 °C) Tc = 40 and the related time / PidPower diagram.

[image]

 Iterative corrections brings up a stable system.

</div>

Wintespe, 2023-01-09:
Moved from discussion as requested by Jason2866

sfromis, 2023-01-09:
What Jasn2866 suggested was not to move the discussion, but to submit a PR with your solution, meaning that you do the changes in the source code, make sure that it is well-tested, and then submit the changes for merge.

Wintespe, 2023-01-09:
Of course the code is well tested. Where else should the results come from? I have attached the pictures. Since I don't maintain any code on Git, I can't make a merge either.

Jason2866, 2023-01-10:
Changes which are tested and working can be provided via a PR (Pull Request) there
https://github.com/arendst/Tasmota/pulls
That's the only way to Contribute code changes in Github.

Jason2866, 2023-01-14:
Closing since it is wrong here.

closed 2023-01-14 by Jason2866

Wintespe, 2023-01-14:
Due to lack of interest I close the request. What a pity.

Jason2866, 2023-01-14:
Please provide a Pull Request for as mentioned before. There is interest, and probably it will be integrated.
The way it is done is via Pull Requests.