arendst/Tasmota #17659: Hardware counter (SetOption72) doesn't work for Generic Modbus energy meter

SirkoVZ, 2023-01-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

when using the Generic Modbus energy meter 
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xnrg_energy/xnrg_29_modbus.ino
the energy monitor hardware counter (SetOption72) is not working (although the code points to SetOption72(?)):
https://github.com/arendst/Tasmota/blob/de900e17bb1dd20bf9d4df9e0ccc065b1d301f34/tasmota/tasmota_xnrg_energy/xnrg_29_modbus.ino#L655
means, the hardware counter of the meter is not used, even when SetOption72 "on" is set.

In this example the hardware counter is 511.3500 kWh (please refer to the weblog 4 output below), but the "Energy Total" is showing 4905 kWh (screenshot).
Energy Today shows 1015 kWh (about two x EnergyTotal).
It seems that sometimes the Modbus returns 0 kWh and the next reading shows 511 kWh and so the firmware counter assumes there was an increase of 511 kWh, which leads to the wrong Energy Total, Energy today and Energy yesterday.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 12.3.1.3
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VS Code
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:27:41.954 MQT: stat/LutzSun2000Adapter/RESULT = {"NAME":"LutzSun2000Adapter","GPIO":[1,8864,1,8896,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:27:42.153 MQT: stat/LutzSun2000Adapter/RESULT = {"Module":{"0":"LutzSun2000Adapter"}}
15:27:42.362 MQT: stat/LutzSun2000Adapter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"8864":"NrgMbs Tx"},"GPIO2":{"0":"None"},"GPIO3":{"8896":"NrgMbs Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:29:56.906 RUL: Rule3 on file#modbus do {"Name":"Sun2000","Baud":9600,"Config":"8N1","Address":1,"Function":3,"Voltage":{"R":32069,"T":3,"M":10},"Power":{"R":32080,"T":2},"Total":{"R":32106,"T":4,"M":100},"User":[{"R":32016,"T":1,"M":10,"J":"PV1Voltage","G":"String 1 Spannung","U":"V","D":21},{"R":32017,"T":1,"M":100,"J":"PV1Current","G":"String 1 Strom","U":"A","D":22},{"R":32018,"T":1,"M":10,"J":"PV2Voltage","G":"String 2 Spannung","U":"V","D":21},{"R":32019,"T":1,"M":100,"J":"PV2Current","G":"String 2
15:29:56.908 RUL: Rule3 +Strom","U":"A","D":22},{"R":32064,"T":2,"J":"Solarleistung","G":"Solarleistung","U":"W","D":23},{"R":37113,"T":2,"J":"Einspeiseleistung","G":"Einspeisung","U":"W","D":23},{"R":37119,"T":2,"M":100,"J":"Eingespeist","G":"Eingespeist","U":"kWh","D":24},{"R":37121,"T":2,"M":100,"J":"Bezogen","G":"Bezogen","U":"kWh","D":24}]} endon
15:29:56.915 MQT: stat/LutzSun2000Adapter/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":816,"Free":83,"Rules":"on file#modbus do {\"Name\":\"Sun2000\",\"Baud\":9600,\"Config\":\"8N1\",\"Address\":1,\"Function\":3,\"Voltage\":{\"R\":32069,\"T\":3,\"M\":10},\"Power\":{\"R\":32080,\"T\":2},\"Total\":{\"R\":32106,\"T\":4,\"M\":100},\"User\":[{\"R\":32016,\"T\":1,\"M\":10,\"J\":\"PV1Voltage\",\"G\":\"String 1 Spannung\",\"U\":\"V\",\"D\":21},{\"R\":32017,\"T\":1,\"M\":100,\"J\":\"PV1Current\",\"G\":\"String 1 Strom\",\"U\":\"A\",\"D\":22},{\"R\":32018,\"T\":1,\"M\":10,\"J\":\"PV2Voltage\",\"G\":\"String 2 Spannung\",\"U\":\"V\",\"D\":21},{\"R\":32019,\"T\":1,\"M\":100,\"J\":\"PV2Current\",\"G\":\"String 2 Strom\",\"U\":\"A\",\"D\":22},{\"R\":32064,\"T\":2,\"J\":\"Solarleistung\",\"G\":\"Solarleistung\",\"U\":\"W\",\"D\":23},{\"R\":37113,\"T\":2,\"J\":\"Einspeiseleistung\",\"G\":\"Einspeisung\",\"U\":\"W\",\"D\":23},{\"R\":37119,\"T\":2,\"M\":100,\"J\":\"Eingespeist\",\"G\":\"Eingespeist\",\"U\":\"kWh\",\"D\":24},{\"R\":37121,\"T\":2,\"M\":100,\"J\":\"Bezogen\",\"G\":\"Bezogen\",\"U\":\"kWh\",\"D\":24}]} endon"}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:31:34.682 CMD: Status 0
15:31:34.690 MQT: stat/LutzSun2000Adapter/STATUS = {"Status":{"Module":0,"DeviceName":"Lutz Sun2000 Adapter","FriendlyName":["LutzSun2000Adapter"],"Topic":"LutzSun2000Adapter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:31:34.701 MQT: stat/LutzSun2000Adapter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T00:00:31","StartupUTC":"2023-01-09T14:31:03","Sleep":250,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2022-10-24T16:23:41","SaveCount":709,"SaveAddress":"F5000"}}
15:31:34.710 MQT: stat/LutzSun2000Adapter/STATUS2 = {"StatusFWR":{"Version":"12.3.1.3(tasmota)","BuildDateTime":"2023.01.02 11:09:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"498/699"}}
15:31:34.722 MQT: stat/LutzSun2000Adapter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":1,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["APSF","FRITZ!Box Lutz"],"TelePeriod":20,"Resolution":"559B88C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00400080","00006000","00004000","00000000"]}}
15:31:34.736 MQT: stat/LutzSun2000Adapter/STATUS4 = {"StatusMEM":{"ProgramSize":549,"Free":452,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F800787","04100001","000000CF","010013C0","00000800","00004000","00000000","54000020","00000090"],"Drivers":"1,2,3,4,8,9,10,12,20,21,29,62","Sensors":"1,2,3,5,6"}}
15:31:34.746 MQT: stat/LutzSun2000Adapter/STATUS5 = {"StatusNET":{"Hostname":"LutzSun2000Adapter","IPAddress":"192.168.178.98","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"E0:98:06:85:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:31:34.754 MQT: stat/LutzSun2000Adapter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.91","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_85XXXX","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:31:34.763 MQT: stat/LutzSun2000Adapter/STATUS7 = {"StatusTIM":{"UTC":"2023-01-09T14:31:34","Local":"2023-01-09T15:31:34","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:11","Sunset":"16:24"}}
15:31:34.771 MQT: stat/LutzSun2000Adapter/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:31:34.783 MQT: stat/LutzSun2000Adapter/STATUS10 = {"StatusSNS":{"Time":"2023-01-09T15:31:34","ENERGY":{"TotalStartTime":"2022-12-29T09:05:04","Total":4395.809,"Yesterday":501.260,"Today":506.120,"Power":177.000,"Voltage":228.0,"PV1Voltage":413.6,"PV1Current":0.170,"PV2Voltage":407.6,"PV2Current":0.160,"Solarleistung":198.000,"Einspeiseleistung":-695.000,"Eingespeist":6.860,"Bezogen":900.340}}}
15:31:34.793 MQT: stat/LutzSun2000Adapter/STATUS11 = {"StatusSTS":{"Time":"2023-01-09T15:31:34","Uptime":"0T00:00:31","UptimeSec":31,"Heap":24,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"MqttCount":1,"Wifi":{"AP":2,"SSId":"FRITZ!Box Lutz","BSSId":"E8:DF:70:XXXXXX","Channel":13,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
[... cut ...]
When SetOption72 is set to "on" then the hardware counter of the meter should be used.
(As there is a reference to SetOption72 in the code)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
attached

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
In this example the generic modbus energy meter is used to read a Huawei Sun2000 10 KTL M1 inverter, which works fine

**(Please, remember to close the issue when the problem has been addressed)**
[image]

arendst, 2023-01-09:
What does it show after executing command ``energytotal 0``

SirkoVZ, 2023-01-09:
After 
energytotal 0
the Total is reset to the Today's value (but not to the hardware counter)

```
17:59:30.091 CMD: Energytotal 0
17:59:30.099 MQT: stat/LutzSun2000Adapter/RESULT = {"EnergyTotal":{"Total":1015.490,"Yesterday":501.260,"Today":1015.490}}
```

Modbus values:

```
18:00:00.999 NRG: Modbus send Device 1, Function 3, Register 7D45 (0/0), Size 1
18:00:01.194 NRG: Modbus register 0, phase 0, rcvd 0103020000B844, T 3, F -1, value 0.0000
18:00:01.195 NRG: Modbus send Device 1, Function 3, Register 7D50 (2/0), Size 2
18:00:01.258 WIF: Prüfe Verbindung...
18:00:01.394 NRG: Modbus register 2, phase 0, rcvd 01030400000000FA33, T 2, F 0, value 0.0000
18:00:01.396 NRG: Modbus send Device 1, Function 3, Register 7D6A (7/0), Size 2
18:00:01.594 NRG: Modbus register 7, phase 0, rcvd 0103040000C7C8A995, T 4, F -2, value 511.4400
18:00:01.596 NRG: Modbus send Device 1, Function 3, Register 7D10 (9/0), Size 1
18:00:01.794 NRG: Modbus register 9, phase 0, rcvd 0103020000B844, T 1, F -1, value 0.0000
18:00:01.796 NRG: Modbus send Device 1, Function 3, Register 7D11 (10/0), Size 1
18:00:01.994 NRG: Modbus register 10, phase 0, rcvd 0103020000B844, T 1, F -2, value 0.0000
18:00:01.996 NRG: Modbus send Device 1, Function 3, Register 7D12 (11/0), Size 1
18:00:02.195 NRG: Modbus register 11, phase 0, rcvd 0103020000B844, T 1, F -1, value 0.0000
18:00:02.197 NRG: Modbus send Device 1, Function 3, Register 7D13 (12/0), Size 1
18:00:02.395 NRG: Modbus register 12, phase 0, rcvd 0103020000B844, T 1, F -2, value 0.0000
18:00:02.397 NRG: Modbus send Device 1, Function 3, Register 7D40 (13/0), Size 2
18:00:02.595 NRG: Modbus register 13, phase 0, rcvd 01030400000000FA33, T 2, F 0, value 0.0000
18:00:02.597 NRG: Modbus send Device 1, Function 3, Register 90F9 (14/0), Size 2
18:00:02.795 NRG: Modbus register 14, phase 0, rcvd 010304FFFFFE48BA41, T 2, F 0, value -440.0000
18:00:02.797 NRG: Modbus send Device 1, Function 3, Register 90FF (15/0), Size 2
18:00:02.995 NRG: Modbus register 15, phase 0, rcvd 010304000002AE7AEF, T 2, F -2, value 6.8600
18:00:02.996 NRG: Modbus send Device 1, Function 3, Register 9101 (16/0), Size 2
18:00:03.196 NRG: Modbus register 16, phase 0, rcvd 0103040001602BC3EC, T 2, F -2, value 901.5500
18:00:03.197 NRG: EnergyTotal[0] 511.4400 kWh
```

arendst, 2023-01-11:
> It seems that sometimes the Modbus returns 0 kWh and the next reading shows 511 kWh and so the firmware counter assumes there was an increase of 511 kWh, which leads to the wrong Energy Total, Energy today and Energy yesterday.

There is your issue. Investigate why your device reports alternating values. Tasmota cannot know what is a valid entry or what isn't.

If ``SO72 1`` then Tasmota expects an incrementing or decrementing energy value at regular intervals. 

To "fix" your zero total energy supply you might want to update file ``xdrv_03_energy.ino`` adding line 317 resulting in the below function:
```
void EnergyUpdateTotal(void) {
  // Provide total import active energy as float Energy.import_active[phase] in kWh: 98Wh = 0.098kWh

  for (uint32_t i = 0; i < Energy.phase_count; i++) {
    AddLog(LOG_LEVEL_DEBUG_MORE, PSTR("NRG: EnergyTotal[%d] %4_f kWh"), i, &Energy.import_active[i]);

    if (0 == Energy.import_active[i]) { continue; }                // No valid energy value received
    if (0 == Energy.start_energy[i] || (Energy.import_active[i] < Energy.start_energy[i])) {
      Energy.start_energy[i] = Energy.import_active[i];            // Init after restart and handle roll-over if any
    }
    else if (Energy.import_active[i] != Energy.start_energy[i]) {
      Energy.kWhtoday[i] = (int32_t)((Energy.import_active[i] - Energy.start_energy[i]) * 100000);
    }

    if ((Energy.total[i] < (Energy.import_active[i] - 0.01f)) &&   // We subtract a little offset of 10Wh to avoid continuous updates
        Settings->flag3.hardware_energy_total) {                   // SetOption72 - Enable hardware energy total counter as reference (#6561)
      // The following calculation allows total usage (Energy.import_active[i]) up to +/-2147483.647 kWh
      RtcSettings.energy_kWhtotal_ph[i] = (int32_t)((Energy.import_active[i] * 1000) - ((Energy.kWhtoday_offset[i] + Energy.kWhtoday[i]) / 100));
      Settings->energy_kWhtotal_ph[i] = RtcSettings.energy_kWhtotal_ph[i];
      Energy.total[i] = Energy.import_active[i];
      Settings->energy_kWhtotal_time = (!Energy.kWhtoday_offset[i]) ? LocalTime() : Midnight();
  //    AddLog(LOG_LEVEL_DEBUG, PSTR("NRG: Energy Total updated with hardware value"));
    }
  }

  EnergyUpdateToday();
}
```

arendst, 2023-01-13:
Alas, you're wrong. If it would work as you expect it would write to flash at every major change as happens in your case. It would wear out flash soon making your hardware useless.

Tasmota tries not to write to flash if possible. That's why it reads the absolute value once and stores it in flash. it then keeps score of the differences locally and stores it in RTC memory which survives a restart but loses it's values on power drop. At midnight it updates flash again with the today value.

So as long as your input differs it won't work relyably unless you tried my suggestion of adding a line of code skipping any zero input.

So pls try the suggested change, compile and see if it solves your issue.

SirkoVZ, 2023-01-13:
Ok, thanks, I understand.
I patched and uploaded the firmware. Let's wait for the next hours/days.

However, probably, ignoring 0 values (as the patch would do) is worth to have another SetOption to enable/disble this feature.

SirkoVZ, 2023-01-14:
Checking for 0 was not enough, because the values where not 0. My current workaround is ignoring all below 500 kWh. I found that sometimes the Modbus reading fails and wrong values are reported, causing the issue.
Probably there should be an option to ignore the complete Modbus readings for the current TelePeriod in case of errors. (Maybe similar to SetOption39). But maybe you have a better idea.
Here is one example for error reading from Modbus, register 7 is (one of) the wrong one:
```

2023-01-14T08:15:20.871734+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D45 (0/0), Size 1
2023-01-14T08:15:20.873034+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 0, phase 0, rcvd 01030208D97E1E, T 3, F -1, value 226.5000
2023-01-14T08:15:20.874757+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D50 (2/0), Size 2
2023-01-14T08:15:20.876582+00:00 LutzSun2000Adapter ESP-NRG: Modbus retry state 2 phase 0
2023-01-14T08:15:20.878171+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D6A (7/0), Size 2
2023-01-14T08:15:20.880192+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 7, phase 0, rcvd 0103040000058D3906, T 4, F -2, value 14.2100
2023-01-14T08:15:20.881915+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D10 (9/0), Size 1
2023-01-14T08:15:21.869056+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 9, phase 0, rcvd 0103040000D274
2023-01-14T08:15:21.870918+00:00 LutzSun2000Adapter ESP-NRG: Modbus error 9
2023-01-14T08:15:21.872230+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D11 (10/0), Size 1
2023-01-14T08:15:21.873943+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 10, phase 0, rcvd 01030205D0BA88, T 1, F -2, value 14.8800
2023-01-14T08:15:21.876044+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D12 (11/0), Size 1
2023-01-14T08:15:21.877389+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 11, phase 0, rcvd 01030200C57817, T 1, F -1, value 19.7000
2023-01-14T08:15:21.879608+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D13 (12/0), Size 1
2023-01-14T08:15:21.882905+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 12, phase 0, rcvd 01030213123579, T 1, F -2, value 48.8200
2023-01-14T08:15:21.884434+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 7D40 (13/0), Size 2
2023-01-14T08:15:21.886391+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 13, phase 0, rcvd 01030200E979CA0103
2023-01-14T08:15:21.888116+00:00 LutzSun2000Adapter ESP-NRG: Modbus error 9
2023-01-14T08:15:21.889955+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 90F9 (14/0), Size 2
2023-01-14T08:15:22.868565+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 14, phase 0, rcvd 01030400000468F91D, T 2, F 0, value 1128.0000
2023-01-14T08:15:22.869718+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 90FF (15/0), Size 2
2023-01-14T08:15:22.871463+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 15, phase 0, rcvd 010304000005F7B8E5, T 2, F -2, value 15.2700
2023-01-14T08:15:22.873264+00:00 LutzSun2000Adapter ESP-NRG: Modbus send Device 1, Function 3, Register 9101 (16/0), Size 2
2023-01-14T08:15:22.874789+00:00 LutzSun2000Adapter ESP-NRG: Modbus register 16, phase 0, rcvd 01030400016E54866C, T 2, F -2, value 937.8000
2023-01-14T08:15:22.876747+00:00 LutzSun2000Adapter ESP-NRG: EnergyTotal[0] 14.2100 kWh
```
The next reading shows correctly EnergyTotal[0] 538.7600 kWh

arendst, 2023-01-14:
Apparently does Modbus send valid data regarding checksum. You might need to lower the poll rate of default 200ms as your hardware cannot handle too many requests at that rate. Perhaps you can find this info in your hardwares documentation.

It's not so obvious to discard lower than last time values as the totals might decrement during power production.

You also seem to receive much larger numbers so this should also be filtered out.

arendst, 2023-01-14:
Latest development allows the use of SO39 for selecting a valid bandwidth in kWh between 1 and 127.

I suggest you do ``SetOption39 10`` to make a window of +/- 10kWh for validation.

SirkoVZ, 2023-01-16:
Thank you very much,
I applied the bandwidth feature already.
I also increased the poll interval to one second, resulting in a stable communication.
I will watch the logs for some further time.
Thanks also for the syslog feature, very helpful :-)

Jason2866, 2023-01-21:
Closing since issue is solved

closed 2023-01-21 by Jason2866