arendst/Tasmota #17660: SetOption15 causes reset on latest production release (12.3.1) during Servo PWM setup on Wemos D1 mini

ggodart, 2023-01-09:
### PROBLEM DESCRIPTION
Since version 12.3, SetOption15 0 causes a reset on a WEMOS D1 Mini when setting up a Servo to use PWM. All worked fine with 12.2.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
D3 GPIO0 is set to Relay 1
D4 GPIO2 is set to PWM 1
D5 GPIO14 is set to Relay 2

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:25:08.293 CMD: status 0
16:25:08.299 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_E3E829","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:25:08.327 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:59","StartupUTC":"2023-01-09T15:19:09","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2023-01-09T15:34:15","SaveCount":29,"SaveAddress":"F9000"}}
16:25:08.360 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"383/699"}}
16:25:08.379 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Seamans","Seamans-ext"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
16:25:08.412 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
16:25:08.443 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E3E829-2089","IPAddress":"192.168.100.199","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.100.1","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:E3:E8:29","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:25:08.468 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E3E829","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:25:08.490 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-01-09T15:25:08","Local":"2023-01-09T16:25:08","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:12"}}
16:25:08.508 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-01-09T16:25:08"}}
16:25:08.515 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-01-09T16:25:08","Uptime":"0T00:05:59","UptimeSec":359,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","PWM":{"PWM1":0},"Wifi":{"AP":2,"SSId":"Seamans-ext","BSSId":"CC:2D:21:D9:AC:41","Channel":3,"Mode":"11n","RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:05"}}}
16:25:08.546 RSL: STATUS13 = {"StatusSHT":{}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Issue occurs outside of web interface, but this is what appears on a Serial console;
backlog  SetOption80 1 ; Shuttermode 5 ; PWMfrequency 200 ;
16:40:01.105 RSL: RESULT = {"SetOption80":"ON"}
16:40:01.335 RSL: RESULT = {"ShutterMode":5}
16:40:01.584 RSL: RESULT = {"PWMFrequency":200}
16:41:08.326 CMD: SetOption15 0 ;
16:41:08.330 RSL: RESULT = {"SetOption15":"OFF"}
16:41:09.530 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F5, Count 15
00:00:00.055 QPC: Count 1
00:00:00.066 Project tasmota - Tasmota Version 12.3.1(tasmota)-2_7_4_9(2022-12-16T10:28:17)
00:00:00.506 WIF: Connecting to AP2 Seamans-ext Channel 3 BSSId CC:2D:21:D9:AC:41 in mode 11n as tasmota-E3E829-2089...
00:00:03.758 WIF: Connected
00:00:04.011 HTP: Web server active on tasmota-E3E829-2089 with IP address 192.168.100.199
16:41:16.025 RSL: INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"12.3.1(tasmota)","FallbackTopic":"cmnd/DVES_E3E829_fb/","GroupTopic":"cmnd/tasmotas/"}}
16:41:16.030 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-E3E829-2089","IPAddress":"192.168.100.199"}}
16:41:16.041 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":8}}
16:41:16.049 RSL: RESULT = {"POWER":"ON"}
16:41:16.052 RSL: POWER = ON
16:41:17.470 QPC: Reset
16:41:20.446 RSL: STATE = {"Time":"2023-01-09T16:41:20","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":2,"SSId":"Seamans-ext","BSSId":"CC:2D:21:D9:AC:41","Channel":3,"Mode":"11n","RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:05"}}
16:42:03.149 CMD: SetOption80
16:42:03.153 RSL: RESULT = {"SetOption80":"ON"}
16:42:22.859 CMD: Shuttermode
16:42:22.864 RSL: RESULT = {"ShutterMode":0}

```

### TO REPRODUCE
Flash with latest release, then issue;
SetOption80 1 
Shuttermode 5 
PWMfrequency 200 
SetOption15 0 
restart 1
check Shuttermode 

### EXPECTED BEHAVIOUR
In order to contol a servo and continue the configuration, ShutterMode needs to be 5 after a restart. It all worked fine with release 12.2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Just in case I compiled my own version with a few items disabled and 
#ifdef SHUTTER_SUPPORT
#undef SHUTTER_SUPPORT
#endif
#define SHUTTER_SUPPORT true 
#define USE_SHUTTER 
But get the same error

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-01-10:
@stefanbode can you have a look?

ggodart, 2023-01-10:
Here is the output as requested;
```
savedata
13:49:02.222 CMD: savedata
13:49:02.234 RSL: RESULT = {"SaveData":"ON"}
syslog 4
13:49:32.433 CMD: syslog 4
13:49:32.440 RSL: RESULT = {"SysLog":{"4":{"Active":"4"}}}
backlog SetOption80 1 ; Shuttermode 5 ; PWMfrequency 200 ; 
13:49:53.894 CMD: backlog SetOption80 1 ; Shuttermode 5 ; PWMfrequency 200 ; 
13:49:53.940 RSL: RESULT = {"SetOption80":"ON"}
13:49:54.191 RSL: RESULT = {"ShutterMode":5}
13:49:54.446 RSL: RESULT = {"PWMFrequency":200}
restart 1
13:50:26.770 CMD: restart 1
13:50:26.778 RSL: RESULT = {"Restart":"Restarting"}
13:50:28.526 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 10
00:00:00.055 QPC: Count 1
00:00:00.066 Project tasmota - Tasmota Version 12.3.1(tasmota)-2_7_4_9(2022-12-16T10:28:17)
00:00:00.506 WIF: Connecting to AP2 Seamans-ext Channel 3 BSSId CC:2D:21:D9:AC:41 in mode 11n as tasmota-E3E829-2089...
00:00:03.765 WIF: Connected
00:00:04.017 HTP: Web server active on tasmota-E3E829-2089 with IP address 192.168.100.199
13:50:34.019 RSL: INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"12.3.1(tasmota)","FallbackTopic":"cmnd/DVES_E3E829_fb/","GroupTopic":"cmnd/tasmotas/"}}
13:50:34.024 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-E3E829-2089","IPAddress":"192.168.100.199"}}
13:50:34.035 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":4}}
13:50:34.043 RSL: RESULT = {"POWER":"OFF"}
13:50:34.046 RSL: POWER = OFF
13:50:35.454 QPC: Reset
13:50:38.437 RSL: STATE = {"Time":"2023-01-10T13:50:38","Uptime":"0T00:00:10","UptimeSec":10,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Seamans-ext","BSSId":"CC:2D:21:D9:AC:41","Channel":3,"Mode":"11n","RSSI":28,"Signal":-86,"LinkCount":1,"Downtime":"0T00:00:05"}}
shuttermode
13:50:41.985 CMD: shuttermode
13:50:41.992 RSL: RESULT = {"ShutterMode":0}
```

Not sure if this is relevant, but as soon as I do a ```SetOption15 0``` it restarts on it s own, I do not need to issue a restart command.

ggodart, 2023-01-10:
Please dont close this just yet as I tried again and it failed, Will do a little more digging to find out what changed

closed 2023-01-10 by ggodart

Jason2866, 2023-01-10:
Yes it is mandatory to set `shutterrelay`.