arendst/Tasmota #17671: Generic Modbus energy meter support, only summarized Export Active

w2533, 2023-01-10:
### PROBLEM DESCRIPTION
When using Generic Modbus energy meter support along with SDM630 Meter, Export Active in GUI only appears as sum of the 3 phases. In MQTT, the single phases are transferred.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Olimex ESP32-PoE
- [ ] Tasmota binary firmware version number used: 12.3.1.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Firmware Upload
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:00:44.174 MQT: stat/Zaehler/RESULT = {"NAME":"Olimex ESP32-PoE","GPIO":[1,1,1,1,1,1,0,0,5536,1,8832,8800,0,0,5600,1,1,1,1,5568,1,1,1,1,0,0,0,0,1,1,32,1,1,1,1,1],"FLAG":0,"BASE":1}
13:00:44.397 MQT: stat/Zaehler/RESULT = {"Module":{"0":"Olimex ESP32-PoE"}}
13:00:44.654 MQT: stat/Zaehler/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"9504":"NrgMbs Tx"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"5536":"ETH POWER"},"GPIO13":{"0":"None"},"GPIO14":{"8832":"SDIO CLK"},"GPIO15":{"8800":"SDIO CMD"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"32":"Button1"},"GPIO35":{"0":"None"},"GPIO36":{"9536":"NrgMbs Rx"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:01:43.079 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"Zaehler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:01:43.085 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software reset CPU","Uptime":"0T00:44:12","StartupUTC":"2023-01-10T11:17:31","Sleep":50,"CfgHolder":4617,"BootCount":171,"BCResetTime":"2020-06-27T14:58:26","SaveCount":1164}}
13:01:43.089 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1.3(tasmota)","BuildDateTime":"2023-01-09T18:00:27","Core":"2_0_6","SDK":"v4.4.3.20221227","CpuFrequency":160,"Hardware":"ESP32-D0WD rev.1","CR":"469/699"}}
13:01:43.092 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["zeitler",""],"TelePeriod":10,"Resolution":"559F98C0","SetOption":["08008009","2805C80001000600001E5A0A002800000000","00400080","00006000","00004000","00000000"]}}
13:01:43.106 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1471,"Free":1984,"Heap":139,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D43C482D","000000B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
13:01:43.112 RSL: STATUS5 = {"StatusNET":{"Hostname":"Zaehler-7132","IPAddress":"192.168.6.147","Gateway":"192.168.6.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.6.115","DNSServer2":"0.0.0.0","Mac":"30:AE:A4:9E:FB:DC","IP6Global":"","IP6Local":"","Ethernet":{"Hostname":"Zaehler-7132-eth","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.6.115","DNSServer2":"0.0.0.0","Mac":"30:AE:A4:9E:FB:DF","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:01:43.116 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.6.34","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9EFBDC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:01:43.122 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-01-10T12:01:43","Local":"2023-01-10T13:01:43","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:41","Sunset":"17:13"}}
13:01:43.125 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:01:43.139 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-01-10T13:01:43","ENERGY":{"TotalStartTime":"2023-01-10T00:00:00","Total":16647.941,"Yesterday":26.425,"Today":7.974,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[1508.365,1929.488,1696.624],"Power":[-188.189,-15.609,-6.492],"ApparentPower":[201.264,14.340,11.569],"ReactivePower":[-66.412,-8.804,11.629],"Factor":[-0.95,-0.65,-0.35],"Frequency":49.893,"Voltage":[231.441,229.800,231.405],"Current":[0.875,0.064,0.059],"ImportActive":[4392.588,2795.954,4301.033]}}}
13:01:43.150 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-01-10T13:01:43","Uptime":"0T00:44:12","UptimeSec":2652,"Heap":141,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":46},"Wifi":{"AP":1,"SSId":"zeitler","BSSId":"CC:CE:1E:F7:86:1A","Channel":6,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:04"}}}
13:01:51.005 RSL: STATE = {"Time":"2023-01-10T13:01:50","Uptime":"0T00:44:20","UptimeSec":2660,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":46},"Wifi":{"AP":1,"SSId":"zeitler","BSSId":"CC:CE:1E:F7:86:1A","Channel":6,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:04"}}
13:01:51.020 RSL: SENSOR = {"Time":"2023-01-10T13:01:51","ENERGY":{"TotalStartTime":"2023-01-10T00:00:00","Total":16647.943,"Yesterday":26.425,"Today":7.976,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[1508.366,1929.488,1696.624],"Period":[1.950,0.000,0.000],"Power":[-190.770,-16.095,-17.277],"ApparentPower":[203.508,21.712,16.980],"ReactivePower":[-67.262,-11.162,11.533],"Factor":[-0.94,-0.84,0.25],"Frequency":49.939,"Voltage":[231.200,229.861,231.525],"Current":[0.872,0.074,0.096],"ImportActive":[4392.588,2795.954,4301.033]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:03:11.031 MQT: tele/Zaehler/STATE = {"Time":"2023-01-10T13:03:11","Uptime":"0T00:45:40","UptimeSec":2740,"Heap":139,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":46},"Wifi":{"AP":1,"SSId":"zeitler","BSSId":"CC:CE:1E:F7:86:1A","Channel":6,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:04"}}
13:03:11.047 MQT: tele/Zaehler/SENSOR = {"Time":"2023-01-10T13:03:11","ENERGY":{"TotalStartTime":"2023-01-10T00:00:00","Total":16647.945,"Yesterday":26.425,"Today":7.978,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[1508.369,1929.488,1696.624],"Period":[0.000,0.000,0.000],"Power":[-136.495,0.000,-78.478],"ApparentPower":[153.005,10.484,80.817],"ReactivePower":[-68.949,-9.750,-27.549],"Factor":[-0.89,-0.24,-0.93],"Frequency":49.900,"Voltage":[230.750,230.456,230.327],"Current":[0.675,0.046,0.366],"ImportActive":[4392.588,2795.954,4301.033]}}
[... cut ...]
{"Name":"SDM630","Baud":9600,"Config":"8N1","Address":1,"Function":4,"Voltage":[0,2,4],"Current":[6,8,10],"Power":[12,14,16],"ApparentPower":[18,20,22],"ReactivePower":[24,26,28],"Factor":[30,32,34],"Frequency":70,"Total":342,"ExportActive":[352,354,356],"User":{"R":[346,348,350],"J":"ImportActive","G":"Import Active","U":"kWh","D":24}}

### EXPECTED BEHAVIOUR
Display Export Active in GUI as 3 values for each phase.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-01-10:
SetOption129 1

w2533, 2023-01-10:
Hi Theo,
Thank you, that was too easy. Sorry for not reading the doc completely.

Problem fixed!
Wolfgang

closed 2023-01-10 by w2533