arendst/Tasmota #17677: Tasmota 12.1.0 brakes IRSend repeat for Panasonic TVs

Saur0o0n, 2023-01-10:
### PROBLEM DESCRIPTION
After upgrade to Tasmota v12.3.1 Percy, I've noticed I cannot turn on the TV with my IR sender/blaster. Getting back, one by one upgrade, last working version is 12.0.2. It looks like perhaps pull " Support for multiple IRsend GPIOs #16138 " brakes something in protocol.

The issued command (that is working on 12.0.2) is:
  irsend {"Protocol":"PANASONIC","Bits":48,"Data":"0x40040100BCBD","DataLSB":"0x22080003DBD","Repeat":5}

The TV is turning on with the same ircode as turning off (or channel number) but repeated quickly 4 times.

In all cases I've used pre-compiled, official binary tasmota-ir.bin.gz.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): custom
- [x] Tasmota binary firmware version number used: 12.3.1; 12.2.0; 12.1.1; 12.1.0; 12.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:56:18.255 CMD: Backlog Template; Module; GPIO 255
21:56:18.278 MQT: stat/tasmota_65B81C/RESULT = {"NAME":"TasmotaIRBridge","GPIO":[0,0,544,0,288,1088,0,0,0,32,1056,0,0,0],"FLAG":0,"BASE":62}
21:56:18.505 MQT: stat/tasmota_65B81C/RESULT = {"Module":{"0":"TasmotaIRBridge"}}
21:56:18.756 MQT: stat/tasmota_65B81C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"544":"LedLink"},"GPIO3":{"0":"None"},"GPIO4":{"288":"Led1"},"GPIO5":{"1088":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"32":"Button1"},"GPIO14":{"1056":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
21:57:10.702 CMD: Backlog Rule1; Rule2; Rule3
21:57:10.721 MQT: stat/tasmota_65B81C/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:57:10.951 MQT: stat/tasmota_65B81C/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:57:11.201 MQT: stat/tasmota_65B81C/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:47:05.265 CMD: status0
21:47:05.292 MQT: stat/tasmota_65B81C/STATUS0 = {"Status":{"Module":0,"DeviceName":"TasmotaIRBridge","FriendlyName":["Tasmota_IRBridge"],"Topic":"tasmota_65B81C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-min.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:44","StartupUTC":"2023-01-10T20:34:21","Sleep":50,"CfgHolder":4617,"BootCount":5715,"BCResetTime":"2022-08-04T14:47:00","SaveCount":7912,"SaveAddress":"FA000"},"StatusFWR":{"Version":"12.0.2(ir)","BuildDateTime":"2022-06-20T12:43:45","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SauronIoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000180","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":623,"Free":380,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12,62","Sensors":"1"},"StatusNET":{"Hostname":"tasmota-65B81C-6172","IPAddress":"192.168.100.12","Gateway":"192.168.100.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.100.254","DNSServer2":"1.1.1.1","Mac":"5C:CF:7F:65:B8:1C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.100.254","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_65B81C","MqttUser":"saurons_iot","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-01-10T20:47:05","Local":"2023-01-10T21:47:05","StartDST":"2023-03-26T03:00:00","EndDST":"2023-10-22T03:00:00","Timezone":99,"Sunrise":"08:41","Sunset":"17:13"},"StatusSNS":{"Time":"2023-01-10T21:47:05"},"StatusSTS":{"Time":"2023-01-10T21:47:05","Uptime":"0T00:12:44","UptimeSec":764,"Vcc":3.502,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SauronIoT","BSSId":"26:5A:4C:1F:06:3E","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Send "irsend {"Protocol":"PANASONIC","Bits":48,"Data":"0x40040100BCBD","DataLSB":"0x22080003DBD","Repeat":5}" on Tasmota 12.0 - this will turn on the Panasonic TH TV. Upgrade to anything newer (12.1+) - this won't work. Downgrade again, it will work again.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS

So I thought for a while, that the repeating time has changed (it's too slow) - because it was enough to send one additional remote button push, while Tasmota was sending proper irsend, and TV was turning on. To check this I've recorded video (slowmo) of the original remote, and Tasmota ir sender - but there is no difference in the pause between sending repeats with 12.0 and 12.1. Although original remote has shorter gap window then Tasmota.
https://youtu.be/nDQawmt1tDo
On the left - original remote, on the right Tasmota 12.1 on bottom Tasmota 12.0 (working one).

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Saur0o0n, 2023-02-10:
What a relief.. it's fixed by being closed.

barbudor, 2023-02-10:
If you suspect the change "Support for multiple IRsend GPIOs" to be the cause, could you please test:
- the last development build before that change (29 july)
https://github.com/tasmota/install/tree/389b95b47edb6c5568bb7941c57675f9536adab6/firmware/development

- the first development build after that change (3rd august)
https://github.com/tasmota/install/tree/d5c620cd48b1176ee3033fc714c4b0a5e3cc3143/firmware/development

Saur0o0n, 2023-02-11:
The binary from 3rd august is working properly (and reporting version as 12.0.2.4).
How can I see the tree of the development binaries to find the first one that's not working?

barbudor, 2023-02-11:
Hi @Saur0o0n 

So the change for multiple GPIOs is not in cause

The development builds are archived in that repo : https://github.com/tasmota/install/tree/main/firmware/development

Other PR I see regarding IR are:
- https://github.com/arendst/Tasmota/pull/6311 - merged on Aug 29th
- https://github.com/arendst/Tasmota/pull/16566 - merged on Sep 18th

The next change impacting IRSend is upgrade of the IR library on
https://github.com/arendst/Tasmota/commit/914859ad485157a6d22cefeb39905f520df5a16d (16/09)
and
https://github.com/arendst/Tasmota/commit/4cf0e0ce9f8e5c8b2009d2b99a85c78bf3e2ad3c (18/09)

To find a version go there ("History" from the above link) : https://github.com/tasmota/install/commits/main/firmware/development
ANd use the buttons at the bottom to navigate
Yes, it's painfull
Once you have a date, choose one of the line (to be sure you are before or after the PR, I suggest to pick the day before and the day after) and use the one before last button on the right (View at this point in History)
There you click on your firmware and choose the Download button on the right

Let me know if you have any question

Saur0o0n, 2023-02-20:
Hi
So this are my testings (i was going back by month untill it was working) by date:
31.12.2022 - nok
30.11.2022 - nok
31.10.2022 - nok
30.09.2022 - nok
31.08.2022 - nok
15.08.2022 - nok
9.08.2022 - nok
8.08.2022 - nok
7.08.2022 - ok
6.08.2022 - ok
29.07.2022 - ok

From this it seams like this IR fix for esp32 (https://github.com/arendst/Tasmota/commit/5915ffdc48de3da82a879b5043a1261b64a1cab8) brakes it for my Panasonic TV (probably also others :) )

barbudor, 2023-02-20:
Awesome analysis @Saur0o0n 

@s-hadinger It looks like this PR of yours has broken some protocols on ESP8266

My understanding is that what you changed is impacting the behavior during transmission
Initially the lib was disabling the interrupts but now they are no more disabled so the interrupt handler is always called and the `recvstate` is taken care there.

Could this disturb the real-time enough to break the protocols ?

Wouldn't it be safer to keep the `detachInterrupt()` ?

barbudor, 2023-02-20:
I'm trying something

@Saur0o0n what binary would you need for testing ? tasmota-ir ?

barbudor, 2023-02-20:
@Saur0o0n If you could try this, please let me know, thanks

[tasmota-ir_test1.zip](https://github.com/arendst/Tasmota/files/10787947/tasmota-ir_test1.zip)

Saur0o0n, 2023-02-21:
Yep, the one from zip works fine :)

barbudor, 2023-02-21:
Thanks I'll PR the change

PR #18013 (2023-02-21): Fix ir panasonic esp8266
## Description:

**Related issue (if applicable):** fixes #17677 

Partially revert changes from @s-hadinger 's #16179 as it breaks some protocols on ESP8266 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x ] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [x ] I accept the [CLA](https://git [... cut]
closed 2023-02-21 by arendst