arendst/Tasmota #17679: AWS IOT not working

philhar88, 2023-01-10:
### PROBLEM DESCRIPTION
this guide no longer works:
https://tasmota.github.io/docs/AWS-IoT/

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): CloudFree P2
- [ ] Tasmota binary firmware version number used: 
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: gitpod
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
00:32:44.710 RSL: RESULT = {"NAME":"CloudFree P2","GPIO":[320,0,576,0,224,2720,0,0,2624,32,2656,0,0,0],"FLAG":0,"BASE":45}
00:32:44.918 RSL: RESULT = {"Module":{"0":"CloudFree P2"}}
00:32:45.124 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
00:33:52.007 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_25D043","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:33:52.034 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:51","StartupUTC":"2023-01-10T23:30:01","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2023-01-10T18:11:43","SaveCount":33,"SaveAddress":"F8000"}}
00:33:52.066 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"463/699"}}
00:33:52.086 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CenturyLink0085",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00206000","00004000","00000000"]}}
00:33:52.118 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
00:33:52.149 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-25D043-4163","IPAddress":"192.168.4.60","Gateway":"192.168.4.1","Subnetmask":"255.255.252.0","DNSServer1":"205.171.3.65","DNSServer2":"205.171.2.65","Mac":"2C:F4:32:25:D0:43","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:33:52.176 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"a1in7z6d62eyx-ats.iot.us-west-2.amazonaws.com","MqttPort":443,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25D043","MqttUser":"tasmota?x-amz-customauthorizer-name=TasmotaAuth","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:33:52.203 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-01-10T23:33:52","Local":"2023-01-11T00:33:52","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:14"}}
00:33:52.220 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:33:52.238 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-01-11T00:33:52","ENERGY":{"TotalStartTime":"2023-01-10T18:11:43","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":125,"Current":0.000}}}
00:33:52.258 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-01-11T00:33:52","Uptime":"0T00:03:51","UptimeSec":231,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"CenturyLink0085","BSSId":"AC:EC:85:CF:D9:14","Channel":7,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:34:08.218 MQT: Attempting connection...
00:34:12.250 MQT: Connect failed to a1in7z6d62eyx-ats.iot.us-west-2.amazonaws.com:443, rc -4. Retry in 70 sec

```

### TO REPRODUCE
follow guide. deploy CloudFormation template in us-west-2

Use gitpod to modify the `user_config_override.h` file

```
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

#ifndef USE_MQTT_TLS
#define USE_MQTT_TLS
#define USE_MQTT_TLS_CA_CERT // Optional but highly recommended
#endif
#ifndef USE_MQTT_AWS_IOT_LIGHT
#define USE_MQTT_AWS_IOT_LIGHT
#endif
#ifdef USE_DISCOVERY
#undef USE_DISCOVERY
#endif

#endif  // _USER_CONFIG_OVERRIDE_H_
```

upload firmware (minimal first, then new firmware)
[firmware.bin.gz](https://github.com/arendst/Tasmota/files/10387560/firmware.bin.gz)

### EXPECTED BEHAVIOUR
MQTT packets sent to ios IOT

### ADDITIONAL CONTEXT
Never see any logs in the custom authorizer lambda function

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-01-11:
It looks like TLS is not enabled. Can you force it with `SetOption103 1` and also set log level to `Weblog 3` or `Seriallog 3`

philhar88, 2023-01-11:
sorry, I see the `TLS connection error: 0` entry in the logs so believe error is at MQTT layer not TLS. Also, the `rc -4` leads me to believed its an AUTH issue (i.e. layer 7)

sfromis, 2023-01-11:
Yes, the posted status command output does indicate SetOption103 being on. Now try the suggestion of setting log level for better output.

philhar88, 2023-01-12:
`Weblog 3` doesn't add any more verbosity:

I tried the entire process from scratch again... same result. 

```
06:28:26.545 CMD: Weblog 3
06:28:26.549 RSL: RESULT = {"WebLog":3}
06:28:26.866 CFG: Saved to flash at F7, Count 30, Bytes 4096
06:28:47.266 MQT: Attempting connection...
06:28:52.547 MQT: TLS connection error: 0
06:28:52.549 MQT: Connect failed to a1in7z6d62eyx-ats.iot.us-west-2.amazonaws.com:443, rc -4. Retry in 30 sec
```

There are no invocations of the lambda functions.

geri-m, 2023-02-09:
[Linking](https://github.com/arendst/Tasmota/discussions/17632#discussioncomment-4876685) a related discussion as the issues seams to be the same.

geri-m, 2023-02-13:
@s-hadinger provided fix https://github.com/arendst/Tasmota/pull/17936 which solved the problem on my end.

closed 2023-02-15 by Jason2866