arendst/Tasmota #17694: Sonoff  TH Elite THR320D display issue when exactly 2 DS18b20 sensors are used

msjfb, 2023-01-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
One sensor: displays temp correctly on the first line. Three sensors: also display temp correctly on the first line (rotates with the rotation number on the second line. Two sensors: Line 1 and 2 display sensor2 temp. So there is a problem with line 1, as it should display temp from sensor1. I have found a solution that works by modifying one line of code (see below).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff TH320D_____
- [x ] Tasmota binary firmware version number used: 12.3.1_____
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [x ] Flashing tools used: gitpod_____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:06:00.419 MQT: stat/Tasmota620040/RESULT = {"NAME":"Sonoff THR320D","GPIO":[32,0,0,0,226,9280,0,0,0,321,0,576,320,9184,9216,9312,0,0,9313,9248,0,1312,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
13:06:00.659 MQT: stat/Tasmota620040/RESULT = {"Module":{"0":"Sonoff THR320D"}}
13:06:00.869 MQT: stat/Tasmota620040/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"9280":"TM1621 DAT"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"0":"None"},"GPIO15":{"576":"LedLink_i"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"9184":"TM1621 CS"},"GPIO18":{"9216":"TM1621 WR"},"GPIO19":{"9312":"Relay_b1"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"9313":"Relay_b2"},"GPIO23":{"9248":"TM1621 RD"},"GPIO24":{"0":"None"},"GPIO25":{"1312":"DS18x201"},"GPIO26":{"0":"None"},"GPIO27":{"3840":"Output Hi"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:07:44.866 MQT: stat/Tasmota620040/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:07:45.099 MQT: stat/Tasmota620040/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":59,"Free":452,"Rules":"ON DS18B20-2#Temperature>0.5 DO TempTargetSet %value% ENDON"}}
13:07:45.334 MQT: stat/Tasmota620040/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":108,"Free":403,"Rules":"ON system#boot DO Backlog cmnd/Tasmota620040/ThermostatModeSet 1; cmnd/Tasmota620040/SensorInputSet 1 ENDON "}}

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:09:08.053 CMD: Status 0
13:09:08.061 MQT: stat/Tasmota620040/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota620040","FriendlyName":["Tasmota620040","Tasmota2"],"Topic":"Tasmota620040","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:09:08.111 MQT: stat/Tasmota620040/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:04:01","StartupUTC":"2023-01-13T18:05:07","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2023-01-01T22:00:29","SaveCount":454}}
13:09:08.158 MQT: stat/Tasmota620040/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2023-01-13T02:23:55","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"427/699"}}
13:09:08.181 MQT: stat/Tasmota620040/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BOULANGER",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
13:09:08.222 MQT: stat/Tasmota620040/STATUS4 = {"StatusMEM":{"ProgramSize":1459,"Free":1856,"Heap":143,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400844D2","20200000","D43C482D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,39,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
13:09:08.295 MQT: stat/Tasmota620040/STATUS5 = {"StatusNET":{"Hostname":"Tasmota620040-0064","IPAddress":"192.168.5.101","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.1","DNSServer2":"0.0.0.0","Mac":"40:22:D8:62:00:40","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:09:08.343 MQT: stat/Tasmota620040/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.236","MqttPort":1883,"MqttClientMask":"Tasmota620040","MqttClient":"Tasmota620040","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:09:08.379 MQT: stat/Tasmota620040/STATUS7 = {"StatusTIM":{"UTC":"2023-01-13T18:09:08","Local":"2023-01-13T13:09:08","StartDST":"2023-03-12T02:00:00","EndDST":"2023-11-05T02:00:00","Timezone":99,"Sunrise":"02:39","Sunset":"11:17"}}
13:09:08.432 MQT: stat/Tasmota620040/STATUS10 = {"StatusSNS":{"Time":"2023-01-13T13:09:08","DS18B20-1":{"Id":"03173041E2FF","Temperature":22.6},"DS18B20-2":{"Id":"051730232EFF","Temperature":23.7},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":23.40,"CtrDutyCycleRead":23},"TempUnit":"C"}}
13:09:08.468 MQT: stat/Tasmota620040/STATUS11 = {"StatusSTS":{"Time":"2023-01-13T13:09:08","Uptime":"0T00:04:01","UptimeSec":241,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BOULANGER","BSSId":"18:31:BF:AC:3D:28","Channel":9,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:00:17.148 MQT: tele/Tasmota620040/STATE = {"Time":"2023-01-13T14:00:17","Uptime":"0T00:55:10","UptimeSec":3310,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BOULANGER","BSSId":"18:31:BF:AC:3D:28","Channel":9,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:06"}}
14:00:17.222 MQT: tele/Tasmota620040/SENSOR = {"Time":"2023-01-13T14:00:17","DS18B20-1":{"Id":"03173041E2FF","Temperature":22.2},"DS18B20-2":{"Id":"051730232EFF","Temperature":27.1},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":26.90,"CtrDutyCycleRead":66},"TempUnit":"C"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect 2 DS18B20 sensors to the unit

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
With 2 sensors connected, line1 should display temp from sensor1 and line2 thetemp from sensor2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Changing one line of code solves the problem. All situations display correctly: 1,2 or 3 sensors are OK.
 But is this the proper way ???
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-01-14:
In case of two sensors chk ``GlobalTemp`` if it's set to 1. This should solve the duplicates.

msjfb, 2023-01-15:
recompiled with the commit, and all is fine whether Globaltemp2 is =0 or =1 (with 2 sensors). thanks

closed 2023-01-15 by msjfb

copitz, 2023-01-22:
Hi @msjfb - sorry to ask this on this issue but how exactly can I connect two sensors to the device? Simply by using a splitte [like this](https://m.media-amazon.com/images/I/51Ur4FVN1eL._AC_SL1200_.jpg) and connecting both to it?

(I do not have this device yet as I could find anywhere if or how it could work with two THS01 sensors)

barbudor, 2023-01-22:
DS18x20 can be connected in parrallel on the same wire so you can add almos as many as you want

msjfb, 2023-01-23:
The DS18B20 is a 1-wire device and by definition you can connect a number of devices in parallel (same color wires connected together). The Tasmota Thermostat software will automatically detect all the different probes (I tested with up to 3 DS18B20 probes).
If the THS01 is a 1-wire device (I have not checked) then you should be able to connect them in parallel.

barbudor, 2023-01-23:
There are "1-wire (C)" and "1-wire like"
The real "1-wire" protocol is from Dallas and a complex bus where devices have addresses (ID) and know how to cooperate

The Sonoff 1-wire protocol is nothing related to that 
It's a copy of the DHT protocol
and AFAIK doesn't support multi-devices

copitz, 2023-01-23:
Thank you very much for the information - I'll try to connect two THS01 and post my findings 🤞

sfromis, 2023-01-23:
I'd take the easy way out, connect sensors to a cheap board like a NodeMCU, where you have full flexibility for many sensors and how to connect them. Those Sonoff devices are not great as a building block to add to.

barbudor, 2023-01-23:
and with Tasmota on both (the Sonoff as a relay control and a NodeMCU as sensor), you can make them communicate to the Sonoff could still react on data provided by the NodeMCU

copitz, 2023-01-23:
Thank you guys! Yes, the NodeMCU way will be the one I'll go - probably with a relay board directly connected. This sonoff thing just looked very interesting as it appeared to have everything for my project (which btw. is controlling a vent in a humid basement by automatic dew point comparison)