arendst/Tasmota #17709: ESP32 : Manage File system : Download function seems to be broken

Noschvie, 2023-01-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The file download at the "Manage File system" web interface seems to lead to a broken filename.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 C3mini
- [ ] Tasmota binary firmware version number used: 12.3.1.3
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Web Installer, OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
14:51:27.867 MQT: stat/tasmota_1D926C/RESULT = {"NAME":"01Space_ESP32-C3-RGB","GPIO":[640,608,1,1,1,1,1,1376,0,32,544,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
14:51:28.067 MQT: stat/tasmota_1D926C/RESULT = {"Module":{"0":"01Space_ESP32-C3-RGB"}}
14:51:28.275 MQT: stat/tasmota_1D926C/RESULT = {"GPIO0":{"640":"I2C SDA1"},"GPIO1":{"608":"I2C SCL1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"1376":"WS28121"},"GPIO8":{"0":"None"},"GPIO9":{"32":"Button1"},"GPIO10":{"544":"LedLink"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"6016":"AS608 Rx"},"GPIO21":{"5984":"AS608 Tx"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
14:52:57.271 MQT: stat/tasmota_1D926C/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1D926C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:52:57.275 MQT: stat/tasmota_1D926C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T01:08:02","StartupUTC":"2023-01-15T12:44:55","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2023-01-13T19:47:34","SaveCount":158}}
14:52:57.279 MQT: stat/tasmota_1D926C/STATUS2 = {"StatusFWR":{"Version":"12.3.1.3(tasmota)","BuildDateTime":"2023-01-15T12:38:59","Core":"2_0_6","SDK":"v4.4.3.20221227","CpuFrequency":160,"Hardware":"ESP32-C3 rev.3","CR":"451/699"}}
14:52:57.284 MQT: stat/tasmota_1D926C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["neunzehn.iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:52:57.293 MQT: stat/tasmota_1D926C/STATUS4 = {"StatusMEM":{"ProgramSize":1554,"Free":1325,"Heap":166,"StackLowMark":5,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","979AD7CF","001DA001","B7F7BFCF","05DA9BC0","E0360DCF","080840D2","20200800","D43C482D","000000B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,53,55,56,58,59,64,66,67,74,79,85,92,95,98,127"}}
14:52:57.298 MQT: stat/tasmota_1D926C/STATUS5 = {"StatusNET":{"Hostname":"tasmota-1D926C-4716","IPAddress":"192.168.7.159","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.7.1","DNSServer2":"0.0.0.0","Mac":"A0:76:4E:1D:92:6C","IP6Global":"","IP6Local":"fe80::a276:4eff:fe1d:926c","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:52:57.301 MQT: stat/tasmota_1D926C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.43","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D926C","MqttUser":"myUser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:52:57.306 MQT: stat/tasmota_1D926C/STATUS7 = {"StatusTIM":{"UTC":"2023-01-15T13:52:57","Local":"2023-01-15T14:52:57","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:20"}}
14:52:57.312 MQT: stat/tasmota_1D926C/STATUS10 = {"StatusSNS":{"Time":"2023-01-15T14:52:57"}}
14:52:57.319 MQT: stat/tasmota_1D926C/STATUS11 = {"StatusSTS":{"Time":"2023-01-15T14:52:57","Uptime":"0T01:08:02","UptimeSec":4082,"Heap":167,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":41},"POWER":"OFF","Dimmer":13,"Color":"000622","HSBColor":"231,98,13","Channel":[0,2,13],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxx.iot","BSSId":"A8:BD:27:64:7E:40","Channel":11,"Mode":"11n","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Got to "Manage File system" and try to download a file from the list, the filename will be fragmented.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
The file script.txt is the selected file to be download.

[image]

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Tested with Firefox and Chrome

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-01-15:
I rarely use file download but I already saw similar behavior. To be investigated

barbudor, 2023-01-15:
Looking with Chrome dev-tools, trying to download a file named "test_simple.stl" I see:
[image]

currently adding more logs to track

barbudor, 2023-01-15:
definitively some bad memory management in the `donload_task'

barbudor, 2023-01-15:
fixed
PR on the way

PR #17710 (2023-01-15): UFS download bad filename
## Description:

**Related issue (if applicable):** fixes #17709 

ESP32 only
in download_task, filename was freed too early

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [X] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/develop [... cut]
closed 2023-01-15 by arendst