arendst/Tasmota #17716: I2S Problem

baccaglini-fabio, 2023-01-15:
Hi, I'm trying to compile I2S on esp32 with the lines in your site:

#ifndef USE_I2S_AUDIO
#define USE_I2S_AUDIO   // Add support for I2S audio output
#endif

#ifndef USE_I2S_MIC
#define USE_I2S_MIC       // in case you want to use a microphone
#endif

// in case you want to use a microphone with mp3 encoding, this also requires PSRAM
#ifndef USE_SHINE
#define USE_SHINE       // use mp3 encoding     
#endif

Every time I compile it, I get the error!

I noticed that if I leave the USE_I2S_MIC alone and USE_SHINE it works but as I insert the USE_I2S_AUDIO it breaks

Jason2866, 2023-01-16:
> Every time I compile it, I get the error!

What error?

baccaglini-fabio, 2023-01-16:
....
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino:557:70: error: macro "mul" passed 5 arguments, but takes just 2
       if (BR_EC_P256_IMPL.mul(pk, pk_len, sk, sk_len, BR_EC_secp256r1)) {
                                                                      ^
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino:675:76: error: macro "mul" passed 5 arguments, but takes just 2
       uint32_t ret = BR_EC_P256_IMPL.mul(res, 65, x, x_len, BR_EC_secp256r1);
                                                                            ^
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino:713:92: error: macro "muladd" passed 8 arguments, but takes just 4
       uint32_t ret = BR_EC_P256_IMPL.muladd(res, B, 65, x, x_len, y, y_len, BR_EC_secp256r1);
                                                                                            ^
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino:776:73: error: macro "mul" passed 5 arguments, but takes just 2
         if (BR_EC25519_IMPL.mul(pk, pk_len, sk, sk_len, BR_EC_curve25519)) {
                                                                         ^
Retrieved `.pio\build\tasmota32-IT\liba69\Adafruit_SGP30-1.2.0\Adafruit_SGP30.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\liba69\libAdafruit_SGP30-1.2.0.a' from cache
Retrieved `.pio\build\tasmota32-IT\lib457\Adafruit_BusIO\Adafruit_BusIO_Register.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib457\Adafruit_BusIO\Adafruit_I2CDevice.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib457\Adafruit_BusIO\Adafruit_SPIDevice.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib9d9\Adafruit_SGP40-1.1.0\Adafruit_SGP40.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib9d9\Adafruit_SGP40-1.1.0\sensirion_voc_algorithm.c.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib457\libAdafruit_BusIO.a' from cache
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino: In function 'int32_t m_ec_p256_mul(bvm*)':
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino:675:38: error: invalid conversion from 'uint32_t (*)(unsigned char*, size_t, const unsigned char*, size_t, int)' {aka 'unsigned int (*)(unsigned char*, unsigned int, const unsigned char*, unsigned int, int)'} to 'uint32_t' {aka 'unsigned int'} [-fpermissive]
       uint32_t ret = BR_EC_P256_IMPL.mul(res, 65, x, x_len, BR_EC_secp256r1);
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino: In function 'int32_t m_ec_p256_muladd(bvm*)':
C:/Users/Fabio/Desktop/VSC/Tasmota-development/tasmota/tasmota_xdrv_driver/xdrv_52_3_berry_crypto.ino:713:38: error: invalid conversion from 'uint32_t (*)(unsigned char*, const unsigned char*, size_t, const unsigned char*, size_t, const unsigned char*, size_t, int)' {aka 'unsigned int (*)(unsigned char*, const unsigned char*, unsigned int, const unsigned char*, unsigned int, const unsigned char*, unsigned int, int)'} to 'uint32_t' {aka 'unsigned int'} [-fpermissive]
       uint32_t ret = BR_EC_P256_IMPL.muladd(res, B, 65, x, x_len, y, y_len, BR_EC_secp256r1);
Retrieved `.pio\build\tasmota32-IT\libcab\Adafruit_VEML7700\Adafruit_VEML7700.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\libcab\libAdafruit_VEML7700.a' from cache
Retrieved `.pio\build\tasmota32-IT\lib9d9\libAdafruit_SGP40-1.1.0.a' from cache
Retrieved `.pio\build\tasmota32-IT\lib4ed\FS\FS.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib4ed\FS\vfs_api.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib734\WiFi\WiFi.cpp.o' from cache
Retrieved `.pio\build\tasmota32-IT\lib734\WiFi\WiFiAP.cpp.o' from cache
*** [.pio\build\tasmota32-IT\src\tasmota.ino.cpp.o] Error 1
======================================================== [FAILED] Took 107.25 seconds ========================================================
Environment    Status    Duration
-------------  --------  ------------
tasmota32-IT   FAILED    00:01:47.253
=================================================== 1 failed, 0 succeeded in 00:01:47.253 ===================================================

Jason2866, 2023-01-16:
@gemu2015 @s-hadinger 
Can reproduce the error.

s-hadinger, 2023-01-16:
`mp3_shine_esp32` defines a macro called `#define mul(x,y)`. This is nonsense and an overly large pollution of the global namespace. There is nothing I can do. 

On top of that, this macro is used to insert assembly language which makes the code non-portable.

@gemu2015 Maybe you can limit the blast radius of those macros.

PR #17725 (2023-01-17): fix shine macros
## Description:

**Related issue (if applicable):** fixes #17716

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The  [... cut]
closed 2023-01-17 by arendst