arendst/Tasmota #17737: SDM230 - no data

joedupa, 2023-01-18:
Hello, I have been using the SDM.h library (https://github.com/reaper7/SDM_Energy_Meter) to monitor the energy using the SDM230 meter. The boards I use are ESP2866 based (NodeMCU, Wemos) that are connected to SDM230 via "RS485 to TTL" converter. That has been working very well for me, gathering data without any problem.
For simplicity, and my admiration to Tasmota project, I decided to flash my boards with the latest tasmota_sensors binary. I used Generic 18 for module type, and SDMx20Tx for GPIO1 and SDMx20Rx as a mode parameters.

[image]

However, after making an identical connections as I used with SDM.h library and corresponding sketches, Tasmota page displays no data, all values are 0.

[image]

I can visually confirm that some data is flowing by seeing a short blip of a TX LED followed by another short blip of RX LED on the RS485 to TTL converter.
While using the SDM.h library mentioned earlier, I had to change the baud rate in SDM230 (and the sketch configuration file) to 2400 as it was the only rate the ESP boards and the meter could use for successful communication. I keep the same rate in the meter using Tasmota firmware, and still see the expected sequence of LED blips on the converter.

https://user-images.githubusercontent.com/12486073/213056437-0b3d5a29-5faa-4587-97bd-d29b4a7aa95f.MOV

One more information, my version of the SDM230 monitor is 110V, not 220V as used in Europe, would that be a critical issue with Tasmota firmware?
Please advise, thank you.

arendst, 2023-01-18:
Tasmota default SDM230 baudrate is 9600 bps.

You can change it by compiling the code with ``#define SDM230_SPEED 2400``

arendst, 2023-01-18:
Log level 4

joedupa, 2023-01-18:
Right, the output in log level 4 is this:

20:51:11.425 RSL: SENSOR = {"Time":"2023-01-18T20:51:11","ENERGY":{"TotalStartTime":"2023-01-18T19:14:18","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"Reactive
Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
20:51:24.411 WIF: Sending Gratuitous ARP
20:51:25.497 WIF: Checking connection...
20:51:45.502 WIF: Checking connection...
20:52:05.501 WIF: Checking connection...
20:52:24.410 WIF: Sending Gratuitous ARP
20:52:25.501 WIF: Checking connection...
20:52:45.502 WIF: Checking connection...
20:53:05.502 WIF: Checking connection...
20:53:24.410 WIF: Sending Gratuitous ARP
20:53:25.502 WIF: Checking connection...

I also have a bunch of Sonoff S31 monitoring plugins and they continually dump info in this format, which I honestly don't understand, but it is there:

20:58:56.070 DMP: 55 5A 03 09 58 00 06 93 00 42 09 00 2B 7E 53 7F 00 00 79 D0 F1 22 06 25
20:58:56.122 WIF: Checking connection...
20:58:56.175 DMP: 55 5A 03 09 58 00 06 93 00 42 09 00 2B AD 53 7F 00 00 79 D0 F1 22 09 57
20:58:56.280 DMP: 55 5A 03 09 58 00 06 93 00 42 09 00 2B AD 53 7F 00 00 79 D0 F1 22 0D 5B

joedupa, 2023-01-18:
IT WORKS !!!!!!!!!!!!!!!
I don't understand why, but the GPIOs 1 and 3 did not work on my esp2866 boards, I relocated the wires to GPIO 12 (Tx) and GPIO 14(Rx) and configured Tasmota accordingly. And it works on default baud rate of 9600. All is well, what would explain the GPIO 1 and 3 behaviour?

sfromis, 2023-01-18:
With gpio1/3 being connected to the USB/serial adapter chip on the dev board, that may interfere with connecting more to the same pins, especially if you power it via USB from the computer, meaning data connected too.

joedupa, 2023-01-19:
I consider this issue solved. I flashed another board (ESP32) and connected and configured GPIO1 and GPIO3 as Tx and Rx. This time the board is powered from a dedicated power supply and not via laptop USB port as per sfromis advice. Works great.
Unrelated - what would be the way to remove the "Export Active" field from the main menu?

closed 2023-01-21 by Jason2866

Quest24, 2025-03-15:
I had the same problem with communication. The problem was the RX input on the NodeMCU. I changed RX to GPIO 0, and it all worked. It looks like the RX input port gives to much load on the RS422 converter.  I have SDM230RX on GPIO and SMD230Tx on TX GPIO1. I use the HW-519 as RS422 converter