arendst/Tasmota #17739: Clear DevGroupName

lucize, 2023-01-18:
### PROBLEM DESCRIPTION
DevGroupName 0 links devices togheter

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [Sonoff T1 3CH]
- [x] Tasmota binary firmware version number used: [Tasmota 12.3.1.4 by Theo Arends]
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: webflash
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

18:53:19.866 CMD: Backlog Template; Module; GPIO 255
18:53:19.989 MQT: stat/tasmota_E95AF0/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:53:20.163 MQT: stat/tasmota_E95AF0/RESULT = {"Module":{"30":"Sonoff T1 3CH"}}
18:53:20.415 MQT: stat/tasmota_E95AF0/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

18:54:13.475 CMD: Status 0
18:54:13.540 MQT: stat/tasmota_E95AF0/STATUS = {"Status":{"Module":30,"DeviceName":"Children_3CH","FriendlyName":["Children_3CH","Children_3CH_2","Children_3CH_3"],"Topic":"tasmota_E95AF0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:54:13.591 MQT: stat/tasmota_E95AF0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":" ","RestartReason":"Software/System restart","Uptime":"0T00:03:49","StartupUTC":"2023-01-18T17:50:24","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2023-01-01T19:13:40","SaveCount":190,"SaveAddress":"F6000"}}
18:54:13.629 MQT: stat/tasmota_E95AF0/STATUS2 = {"StatusFWR":{"Version":"12.3.1.3(tasmota)","BuildDateTime":"2023-01-01T14:45:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"343/699"}}
18:54:13.656 MQT: stat/tasmota_E95AF0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HA",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00206048","00004000","00000000"]}}
18:54:13.698 MQT: stat/tasmota_E95AF0/STATUS4 = {"StatusMEM":{"ProgramSize":686,"Free":316,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC7C7","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
18:54:13.744 MQT: stat/tasmota_E95AF0/STATUS5 = {"StatusNET":{"Hostname":"tasmota-E95AF0-6896","IPAddress":"10.160.20.149","Gateway":"10.160.20.1","Subnetmask":"255.255.255.0","DNSServer1":"10.160.20.1","DNSServer2":"10.160.20.10","Mac":"C4:4F:33:E9:5A:F0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:54:13.777 MQT: stat/tasmota_E95AF0/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxx","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E95AF0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:54:13.814 MQT: stat/tasmota_E95AF0/STATUS7 = {"StatusTIM":{"UTC":"2023-01-18T17:54:13","Local":"2023-01-18T18:54:13","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"17:24"}}
18:54:13.840 MQT: stat/tasmota_E95AF0/STATUS10 = {"StatusSNS":{"Time":"2023-01-18T18:54:13"}}
18:54:13.859 MQT: stat/tasmota_E95AF0/STATUS11 = {"StatusSTS":{"Time":"2023-01-18T18:54:13","Uptime":"0T00:03:49","UptimeSec":229,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"xx","BSSId":"2x:5x:Bx:5x:9x:5x","Channel":6,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
setup any `DevGroupName` then do `DevGroupName 0` on multiple devices, all of them will respond

### EXPECTED BEHAVIOUR
`DevGroupName 0` should get the device out of any group, not add all to 0 group

pcdiem, 2023-01-18:
Clearing the device group names returns to the default of using the MQTT group topic as the device group name,

lucize, 2023-01-18:
I don't understand what is the exact command, I issued `DevGroupName 0` or `DevGroupName1 0`

lucize, 2023-01-18:
I don't want to disable groups as I use other groups on the device, for me is clear this is a bug, as any DevGroupName cleared, as per docs, will get linked, so all will fall in the same "cleared" group

barbudor, 2023-01-18:
An empty group name falls back to the group topic.
I had such situation in the past where I used a dummy group name, different on each device so it doesn't sync.

pcdiem, 2023-01-18:
I'm not sure what you mean by "seems that DevGroupName1 is cleared, but it will work with other DevGroupNameX where I issued the DevGroupNameX 0."

Per the docs "If a device group name is not set for a group, the MQTT group topic (GroupTopic) is used (with the device group number appended for device group numbers > 1).

lucize, 2023-01-18:
what I don't understand here is that until (by mistake) I added the name to DevGroupName it did not react, now after clear is reacting whit the other that I cleared also, so are you saying this is expected ? where other  variables changed there ? Option 85 was enabled earlier

pcdiem, 2023-01-18:
If SO85 was on, then device group one was active. It was using whatever was set by DevGroupName (or DevGroupName1). If the name had not been set, it was using the MQTT group topic.

lucize, 2023-01-18:
SW1
SO85 on
SO88 on
devg1 not touched
devg2 defined

SW2
SO85 on
devg1 defined

SW3
SO85 on
SO88 on
devg1 defined
devg2 cleared

after defining and clearing SW1 devg1: devg1 SW1 and devg2 SW3 get linked

pcdiem, 2023-01-18:
The only way I can see that happening is if the MQTT group topic on SW1 was the same at the devg1 name on SW3. You can enable debugging with WebLog 4 and watch the console to see what device group messages are being sent and received.

Jason2866, 2023-02-01:
Closing since no feedback. Feel free to reopen if the issue is not solved.

closed 2023-02-01 by Jason2866