arendst/Tasmota #17741: Shelly 2.5 Phase 2 `Power Factor` not adjusting correctly

AleXSR700, 2023-01-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello,

I was calibrating all my Shelly devices (2.5 and 1PM) and found that on one of my Shelly 2.5 the power factor of phase 2 is not being corrected.

As a result, the power is shown correctly but the calculation is performed incorrectly. This primarily causes an issue with lower power device where the power factor then falsely drops to 0.0x.

I calibrated using an old school 100 W light bulb.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

[17:02:01] cmnd/tasmota_228C3D/MqttLog 5 [17:02:02] stat/tasmota_228C3D/RESULT {"MqttLog":0} [17:02:08] cmnd/tasmota_228C3D/MqttLog 4 [17:02:08] stat/tasmota_228C3D/RESULT {"MqttLog":4} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.537 MQT: stat/tasmota_228C3D/RESULT = {"GPIOs2":{"6656":"TM1637 DIO","6688":"DLP Tx","6720":"DLP Rx","6752":"SSD1351 DC","6784":"XPT2046 CS","6816":"CSE7761 Tx","6848":"CSE7761 Rx","6880":"VL53LXX XSHUT","6912":"MAX7219 CLK","6944":"MAX7219 DIN","6976":"MAX7219 CS","7008":"TFmini+ TX","7040":"TFmini+ RX","7072":"ZC Pulse","7104":"Input","7136":"I2S DOUT","7168":"I2S BCLK","7200":"I2S WS","7232":"I2S DIN","7264":"I2S WS IN","7296":"I2S BCLK IN","7328":"Interrupt","7360":"MCP2515 CS","7392":"HRG15 Tx","7424":"HRG15 Rx","7456":"VINDRIKTNING","7488":"BL0939 Rx","7520":"BL0942 Rx","7552":"HM330X SET","7584":"Heartbeat","7616":"Heartbeat_i","7648":"74x595 SRCLK","7680":"74x595 RCLK","7712":"74x595 OE","7744":"74x595 SER","7776":"SolaxX1 RTS","7808":"Option E","7840":"SDM230 Tx","7872":"SDM230 Rx","7904":"ADC MQ","7936":"CM110x TX","7968":"CM110x RX","8000":"BL6523 Tx","8032":"BL6523 Rx","8064":"ADE7880 IRQ","8096":"Reset","8128":"MS01","8160":"SDIO CMD","8192":"SDIO CLK","8224":"SDIO D0","8256":"SDIO D1","8288":"SDIO D2","8320":"SDIO D3","8352":"Flowrate","8384":"BP5758D Clk","8416":"BP5758D Dat","8448":"SM2335 Clk","8480":"SM2335 Dat","8512":"MP3 Busy","8544":"TM1621 CS","8576":"TM1621 WR","8608":"TM1621 RD","8640":"TM1621 DAT","8672":"Relay_b","8704":"Relay_bi","8736":"I2S MCLK","8768":"ModBr Tx","8800":"ModBr Rx","8832":"ADE7953 RST","8864":"NrgMbs Tx","8896":"NrgMbs Rx","8928":"ADE7953 CS","8960":"Dali RX","8992":"Dali TX","9024":"BP1658CJ Clk","9056":"BP1658CJ Dat","9088":"Dingtian CLK","9120":"Dingtian SDI","9152":"Dingtian Q7","9184":"Dingtian PL","9216":"Dingtian RCK","9248":"LD2410 Tx","9280":"LD2410 Rx","9312":"ModBr Tx Ena","9344":"NrgMbs Tx Ena","9376":"ME007 Tri","9408":"ME007 Rx"}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.557 MQT: stat/tasmota_228C3D/RESULT = {"SwitchDebounce":50} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.656 MQT: stat/tasmota_228C3D/RESULT = {"BlinkTime":10} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.708 MQT: stat/tasmota_228C3D/RESULT = {"MqttLog":0} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.758 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime2":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.807 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime4":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.859 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime6":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.909 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime8":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:54.960 MQT: stat/tasmota_228C3D/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.009 MQT: stat/tasmota_228C3D/RESULT = {"ButtonDebounce":50} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.060 MQT: stat/tasmota_228C3D/RESULT = {"BlinkCount":10} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.110 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime3":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.161 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime7":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.212 MQT: stat/tasmota_228C3D/RESULT = {"GPIO":"Not supported"} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.261 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime1":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.311 MQT: stat/tasmota_228C3D/RESULT = {"PulseTime5":{"Set":0,"Remaining":0}} [17:02:08] stat/tasmota_228C3D/LOGGING 17:01:55.362 MQT: stat/tasmota_228C3D/RESULT = {"Interlock":"OFF","Groups":"1,2"} [17:02:08] stat/tasmota_228C3D/LOGGING 17:02:02.423 MQT: stat/tasmota_228C3D/RESULT = {"MqttLog":0} [17:02:08] stat/tasmota_228C3D/LOGGING 17:02:09.204 MQT: stat/tasmota_228C3D/RESULT = {"MqttLog":4} [17:02:09] stat/tasmota_228C3D/LOGGING 17:02:09.411 ADE: ACCMODE 0x2D1800, VRMS 6104611, 6104999, Period 4470, 4471, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:10] stat/tasmota_228C3D/LOGGING 17:02:09.854 CFG: Saved to flash at F6, Count 86, Bytes 4096 [17:02:10] stat/tasmota_228C3D/LOGGING 17:02:10.010 WIF: Checking connection... [17:02:10] stat/tasmota_228C3D/LOGGING 17:02:10.411 ADE: ACCMODE 0x2D1800, VRMS 6101932, 6103829, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0 [17:02:10] stat/tasmota_228C3D/LOGGING 17:02:11.411 ADE: ACCMODE 0x2D1800, VRMS 6099505, 6103070, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:12] stat/tasmota_228C3D/LOGGING 17:02:12.411 ADE: ACCMODE 0x2D1800, VRMS 6103485, 6107061, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0 [17:02:13] stat/tasmota_228C3D/LOGGING 17:02:13.411 ADE: ACCMODE 0x2D1800, VRMS 6100396, 6102236, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:14] stat/tasmota_228C3D/LOGGING 17:02:14.411 ADE: ACCMODE 0x2D1800, VRMS 6100001, 6100502, Period 4471, 4471, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0 [17:02:15] stat/tasmota_228C3D/LOGGING 17:02:15.412 ADE: ACCMODE 0x2D1800, VRMS 6102084, 6098747, Period 4471, 4471, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:16] stat/tasmota_228C3D/LOGGING 17:02:16.412 ADE: ACCMODE 0x2D1800, VRMS 6101487, 6097697, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0 [17:02:16] stat/tasmota_228C3D/LOGGING 17:02:17.411 ADE: ACCMODE 0x2D1800, VRMS 6101044, 6097500, Period 4471, 4471, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:18] stat/tasmota_228C3D/LOGGING 17:02:18.412 ADE: ACCMODE 0x2D1800, VRMS 6101924, 6100130, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0 [17:02:19] stat/tasmota_228C3D/LOGGING 17:02:19.412 ADE: ACCMODE 0x2D1800, VRMS 6097365, 6097952, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:20] stat/tasmota_228C3D/LOGGING 17:02:20.411 ADE: ACCMODE 0x2D1800, VRMS 6097979, 6099902, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0 [17:02:21] stat/tasmota_228C3D/LOGGING 17:02:21.410 ADE: ACCMODE 0x2D1800, VRMS 6096759, 6100421, Period 4470, 4470, IRMS 1773, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0 [17:02:22] cmnd/tasmota_228C3D/power1 on [17:02:22] stat/tasmota_228C3D/RESULT {"POWER1":"ON"} [17:02:22] stat/tasmota_228C3D/POWER1 ON [17:02:22] stat/tasmota_228C3D/LOGGING 17:02:22.412 ADE: ACCMODE 0x2D1800, VRMS 6096553, 6100015, Period 4470, 4470, IRMS 1773, 1773, WA [... cut ...] age":0,"Current":[0.000,0.000]}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Power factor should become 1.0 after calibration (as does for phase 1 and other devices.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-01-27:
Pls provide requested info regarding ``status 0``.

arendst, 2023-01-27:
In addition provide output of command ``VoltageSet``

Jason2866, 2023-02-01:
Closing since no feedback.

closed 2023-02-01 by Jason2866

AleXSR700, 2023-02-15:
Status 0
```
00:17:44.134 MQT: tele/tasmota_228C3D/STATE = {"Time":"2023-02-15T00:17:44","Uptime":"28T05:01:40","UptimeSec":2437300,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"06:31:92:E7:DC:7E","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
00:17:44.147 MQT: tele/tasmota_228C3D/SENSOR = {"Time":"2023-02-15T00:17:44","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":37.7},"ENERGY":{"TotalStartTime":"2023-01-17T18:35:56","Total":[0.205,0.242],"Yesterday":[0.003,0.005],"Today":[0.000,0.000],"Period":[0.0,0.0],"Power":[0.0,0.0],"ApparentPower":[0.0,0.0],"ReactivePower":[0.0,0.0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
01:17:44.159 MQT: tele/tasmota_228C3D/STATE = {"Time":"2023-02-15T01:17:44","Uptime":"28T06:01:40","UptimeSec":2440900,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"06:31:92:E7:DC:7E","Channel":1,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
01:17:44.173 MQT: tele/tasmota_228C3D/SENSOR = {"Time":"2023-02-15T01:17:44","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":37.7},"ENERGY":{"TotalStartTime":"2023-01-17T18:35:56","Total":[0.205,0.242],"Yesterday":[0.003,0.005],"Today":[0.000,0.000],"Period":[0.0,0.0],"Power":[0.0,0.0],"ApparentPower":[0.0,0.0],"ReactivePower":[0.0,0.0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
02:17:44.149 MQT: tele/tasmota_228C3D/STATE = {"Time":"2023-02-15T02:17:44","Uptime":"28T07:01:40","UptimeSec":2444500,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"06:31:92:E7:DC:7E","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
02:17:44.164 MQT: tele/tasmota_228C3D/SENSOR = {"Time":"2023-02-15T02:17:44","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":37.0},"ENERGY":{"TotalStartTime":"2023-01-17T18:35:56","Total":[0.205,0.242],"Yesterday":[0.003,0.005],"Today":[0.000,0.000],"Period":[0.0,0.0],"Power":[0.0,0.0],"ApparentPower":[0.0,0.0],"ReactivePower":[0.0,0.0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
03:17:44.148 MQT: tele/tasmota_228C3D/STATE = {"Time":"2023-02-15T03:17:44","Uptime":"28T08:01:40","UptimeSec":2448100,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"06:31:92:E7:DC:7E","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
03:17:44.161 MQT: tele/tasmota_228C3D/SENSOR = {"Time":"2023-02-15T03:17:44","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":37.1},"ENERGY":{"TotalStartTime":"2023-01-17T18:35:56","Total":[0.205,0.242],"Yesterday":[0.003,0.005],"Today":[0.000,0.000],"Period":[0.0,0.0],"Power":[0.0,0.0],"ApparentPower":[0.0,0.0],"ReactivePower":[0.0,0.0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
04:17:45.139 MQT: tele/tasmota_228C3D/STATE = {"Time":"2023-02-15T04:17:45","Uptime":"28T09:01:41","UptimeSec":2451701,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"06:31:92:E7:DC:7E","Channel":1,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
04:17:45.152 MQT: tele/tasmota_228C3D/SENSOR = {"Time":"2023-02-15T04:17:45","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":37.1},"ENERGY":{"TotalStartTime":"2023-01-17T18:35:56","Total":[0.205,0.242],"Yesterday":[0.003,0.005],"Today":[0.000,0.000],"Period":[0.0,0.0],"Power":[0.0,0.0],"ApparentPower":[0.0,0.0],"ReactivePower":[0.0,0.0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
04:44:08.738 CMD: status 0
04:44:08.746 MQT: stat/tasmota_228C3D/STATUS = {"Status":{"Module":0,"DeviceName":"LightSwitch_LR","FriendlyName":["LightSwitch_LR_Main","LightSwitch_LR_Ambient"],"Topic":"tasmota_228C3D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
04:44:08.752 MQT: stat/tasmota_228C3D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"28T09:28:04","StartupUTC":"2023-01-17T18:16:04","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2023-01-17T18:35:56","SaveCount":444,"SaveAddress":"F8000"}}
04:44:08.757 MQT: stat/tasmota_228C3D/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"446/699"}}
04:44:08.762 MQT: stat/tasmota_228C3D/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BaumlerlangenHA24",""],"TelePeriod":3600,"Resolution":"558980C0","SetOption":["4000800B","2805C80001000600003C5A0A192800780000","00000080","00006000","0000C000","0 [... cut ...]
04:44:08.815 MQT: stat/tasmota_228C3D/STATUS11 = {"StatusSTS":{"Time":"2023-02-15T04:44:08","Uptime":"28T09:28:04","UptimeSec":2453284,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"06:31:92:E7:DC:7E","Channel":1,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

VoltageSet
```
04:45:24.343 CMD: VoltageSet
04:45:24.350 MQT: stat/tasmota_228C3D/RESULT = {"VoltageSetCal":[26119,26420]}
```

arendst, 2023-02-15:
Try latest dev release and (re-)calibrate output two using command ``VoltageSet2``, ``CurrentSet2`` and ``PowerSet2``.

AleXSR700, 2023-03-01:
So, tried with latest release as it is newer than 14 days.
Unfortunately to the same result.

```
15:16:07.282 CMD: status 0
15:16:07.291 MQT: stat/tasmota_228C3D/STATUS = {"Status":{"Module":0,"DeviceName":"LightSwitch_LR","FriendlyName":["LightSwitch_LR_Main","LightSwitch_LR_Ambient"],"Topic":"tasmota_228C3D","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:16:07.296 MQT: stat/tasmota_228C3D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:10","StartupUTC":"2023-03-01T14:12:57","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2023-01-17T18:35:56","SaveCount":562,"SaveAddress":"FA000"}}
15:16:07.303 MQT: stat/tasmota_228C3D/STATUS2 = {"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023-02-24T15:32:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"446/699"}}
15:16:07.309 MQT: stat/tasmota_228C3D/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BaumlerlangenHA24",""],"TelePeriod":3600,"Resolution":"558980C0","SetOption":["4000800B","2805C80001000600003C5A0A192800780000","00000080","00006000","0000C000","00000000"]}}
15:16:07.322 MQT: stat/tasmota_228C3D/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
15:16:07.329 MQT: stat/tasmota_228C3D/STATUS5 = {"StatusNET":{"Hostname":"tasmota-228C3D-3133","IPAddress":"192.168.0.203","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"40:F5:20:22:8C:3D","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:16:07.335 MQT: stat/tasmota_228C3D/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_228C3D","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:16:07.342 MQT: stat/tasmota_228C3D/STATUS7 = {"StatusTIM":{"UTC":"2023-03-01T14:16:07","Local":"2023-03-01T15:16:07","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"07:32","Sunset":"18:32"}}
15:16:07.348 MQT: stat/tasmota_228C3D/STATUS9 = {"StatusPTH":{"PowerDelta":[102,102,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:16:07.361 MQT: stat/tasmota_228C3D/STATUS10 = {"StatusSNS":{"Time":"2023-03-01T15:16:07","Switch1":"OFF","Switch2":"ON","ANALOG":{"Temperature":50.9},"ENERGY":{"TotalStartTime":"2023-01-17T18:35:56","Total":[0.289,0.302],"Yesterday":[0.014,0.014],"Today":[0.000,0.006],"Power":[0.0,99.9],"ApparentPower":[0.0,164.5],"ReactivePower":[0.0,129.5],"Factor":[0.00,0.61],"Frequency":50,"Voltage":231,"Current":[0.000,0.437]},"TempUnit":"C"}}
15:16:07.369 MQT: stat/tasmota_228C3D/STATUS11 = {"StatusSTS":{"Time":"2023-03-01T15:16:07","Uptime":"0T00:03:10","UptimeSec":190,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"BaumlerlangenHA24","BSSId":"5C:7D:7D:D7:A8:88","Channel":1,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
PowerFactor is still far below 1.