arendst/Tasmota #17744: all settings gone

montyfert, 2023-01-19:
me again lol 
uptill yesterday everything working great had weather icon in the middle mqtt temps either side .
today had to set it up again .

_Originally posted by @montyfert in https://github.com/arendst/Tasmota/discussions/13590#discussioncomment-4728444_

montyfert, 2023-01-19:
woud it be posible to add config backup option .
thanks 
martin

montyfert, 2023-01-19:
hi there isnt a backup config in the wled web ui.
martin

sfromis, 2023-01-19:
Look harder, in the "Configuration" menu.

montyfert, 2023-01-19:
no still cant find it 
[image]

sfromis, 2023-01-19:
I was referring to what you have in Tasmota, and that screenshot is not from Tasmota.

montyfert, 2023-01-19:
am i in the wrong place

barbudor, 2023-01-19:
Yes
Here is Tasmota
Wled is here : https://github.com/Aircoookie/WLED

arendst, 2023-01-20:
Closing as not a tasmota issue.

closed 2023-01-20 by arendst