arendst/Tasmota #17761: Gosund EP2 resets due to HW-Watchdog, shows wrong relay-state while consumption works

florianwgnr, 2023-01-21:
### PROBLEM DESCRIPTION
When using Tasmota on a Gosund EP2 socket, Tasmota crashes and restarts after several days due to the "Hardware Watchdog". If the EP2 relay was on before the crash, the (hardware) relay remains on, but is shown as "OFF" within Tasmota. The blue light (state LED) on the socket is also off in this case. 

Under this condition, current or power is shown as zero within Tasmota, while the daily or total energy consumption increases and displays the correct values.

If the relay is turned on via the webinterface, no "click" sound can be heard (as the relay is already on) but tasmota starts to display the correct values for voltage, current and power.  

I tested it with 2 EP2 plugs, both were flashed after an upgrade to 1.0.6 firmware and using tuya-convert.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund EP2 
- [x] Tasmota binary firmware version number used: 12.3.1 and/or 12.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tuya-convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  # config from here: https://templates.blakadder.com/gosund_EP2.html
17:36:53.186 CMD: Backlog Template; Module; GPIO 255
17:36:53.257 MQT: SmartPlug/T9/stat/RESULT = {"NAME":"Gosund EP2","GPIO":[320,1,576,1,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
17:36:53.471 MQT: SmartPlug/T9/stat/RESULT = {"Module":{"0":"Gosund EP2"}}
17:36:53.485 MQT: SmartPlug/T9/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-21T17:36:53","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":11.756,"Yesterday":0.144,"Today":0.100,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:36:53.725 MQT: SmartPlug/T9/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:38:31.484 CMD: Status 0
17:38:31.491 MQT: SmartPlug/T9/stat/STATUS = {"Status":{"Module":0,"DeviceName":"T9","FriendlyName":["T9"],"Topic":"T9","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:38:31.499 MQT: SmartPlug/T9/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"8T11:21:41","StartupUTC":"2023-01-13T05:16:50","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2022-10-28T16:53:17","SaveCount":204,"SaveAddress":"F5000"}}
17:38:31.508 MQT: SmartPlug/T9/stat/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022.10.17 08:43:02","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"385/699"}}
17:38:31.515 MQT: SmartPlug/T9/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800A","2805C80000000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
17:38:31.528 MQT: SmartPlug/T9/stat/STATUS4 = {"StatusMEM":{"ProgramSize":627,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
17:38:31.536 MQT: SmartPlug/T9/stat/STATUS5 = {"StatusNET":{"Hostname":"T9-3301","IPAddress":"xxx","Gateway":"xxx","Subnetmask":"255.255.255.0","DNSServer1":"xxx","DNSServer2":"0.0.0.0","Mac":"xxx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:38:31.541 MQT: SmartPlug/T9/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"xxx","MqttClient":"xxx","MqttUser":"xxx","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:38:31.547 MQT: SmartPlug/T9/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-01-21T16:38:31","Local":"2023-01-21T17:38:31","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:33","Sunset":"17:29"}}
17:38:31.550 MQT: SmartPlug/T9/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:38:31.560 MQT: SmartPlug/T9/stat/STATUS10 = {"StatusSNS":{"Time":"2023-01-21T17:38:31","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":11.756,"Yesterday":0.144,"Today":0.100,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:38:31.569 MQT: SmartPlug/T9/stat/STATUS11 = {"StatusSTS":{"Time":"2023-01-21T17:38:31","Uptime":"8T11:21:41","UptimeSec":732101,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":11,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":2,"Downtime":"0T00:00:30"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

Install Tasmota on a Gosund EP2, turn the relay on and check back after several days. 

### EXPECTED BEHAVIOUR

- Tasmota should not crash
- If Tasmota crashes, the current relay state should be shown

### SCREENSHOTS
Relay is on, shown as off and the consumption increases:

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-01-21:
I have Gosund SP111 which use the exact same template, same power measument chip so same driver/code
Running v12.3.1.4 dev
- It never crashes
- I just tried ` restart -2` and ` restart -3`  to force watchdog reboot and after restart the web GUI give proper POWER state (ON)

Questions:
- Does the web UI problem also occur on a simple ` restart 1` (or restart button from Web UI)
- Does the web UI still show OFF if you do a refresh ?
- Please provide the restart logs with level 3 or 4 right after a crash (you may need to use syslog or mqttlog to record the logs)

florianwgnr, 2023-01-21:
To answer your questions:

- after `restart 1`, the web still shows "OFF" while the relay is turned on
- refreshing the page does not change anything.

- Then i turned the relay in tasmota back on manually (to get the state right), turned it off and back on (hardware relay clicks 2 times), i tried `restart -2` and `restart -3`. After each restart, the correct state is shown ("ON"), so the error cannot be reproduced with neither `restart -2` nor `restart -3`

- i will try to set the logging up, but it might take several days to install this and record a crash

barbudor, 2023-01-21:
So a `restart 1` seems to be enough to reproduce the problem ?
It is strange that you can reproduce the problem with a simple `restart 1` but  `restart -2/-3` doesn't show

If `restart 1` does reproduce the problem, just set `weblog 4` then `restart 1` and provide the logs from the beginning f boot until the 1st teleperiod message
No need for complex setup
Thanks

florianwgnr, 2023-01-21:
> So a `restart 1` seems to be enough to reproduce the problem ?

No, a `restart 1` does not trigger the problem 
- everything is working, tasmota "ON", relay "ON" -> `restart 1`  -> everything is working, tasmota ON, relay ON

and a `restart 1` does not solve the problem 
- tasmota crashes -> tasmota "OFF", relay "ON" -> `restart 1` -> error persists: tasmota "OFF", relay "ON"

Also neither `restart -2` nor `restart -3` triggers the problem.

barbudor, 2023-01-21:
OK.
So we need better logs

Jason2866, 2023-01-22:
Have you done a `reset 5` or `reset 6` after the Tuya Convert process to erase everything in flash left over from Tuya firmware? Reminents in flash (specially in Tasmota settings Area) can generate weird issues.

florianwgnr, 2023-01-22:
> OK. So we need better logs

I just set up MqttLog and NodeRed to write logs into a database. Might take some time...

> Have you done a `reset 5` or `reset 6` after the Tuya Convert process? Done to erase everything in flash left over from Tuya firmware. Reminents in flash (specially in Tasmota settings Area) can generate weird issues.

Yes, `reset 5 on both devices.

florianwgnr, 2023-01-29:
> OK. So we need better logs

One of the devices crashed today. This is the log output (starting from the bottom):

```
...
13:31:18.201 HLW: EPC 4, CFlen 249664 usec
13:31:17.201 HLW: EPC 4, CFlen 252243 usec
13:31:16.201 HLW: EPC 4, CFlen 257862 usec
13:31:15.778 WIF: Prüfe Verbindung...
13:31:15.201 HLW: EPC 4, CFlen 264711 usec
13:31:14.201 HLW: EPC 4, CFlen 256501 usec
13:31:13.177 MQT: XXX/T9/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:31:13","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":12.960,"Yesterday":0.136,"Today":0.077,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:31:13.168 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:31:13.201 HLW: EPC 4, CFlen 253616 usec
13:31:13.164 SRC: MQTT
13:31:12.201 HLW: EPC 4, CFlen 261392 usec
13:31:11.201 HLW: EPC 3, CFlen 258518 usec
13:31:10.201 HLW: EPC 4, CFlen 259825 usec
13:31:09.201 HLW: EPC 4, CFlen 260613 usec
13:31:08.202 HLW: EPC 4, CFlen 252653 usec
13:31:07.201 HLW: EPC 4, CFlen 256218 usec
13:31:06.463 MQT: tasmota/discovery/E09806DFECE5/sensors = {"sn":{"Time":"2023-01-29T13:31:06","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":12.960,"Yesterday":0.136,"Today":0.077,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}},"ver":1} (beibehalten)
13:31:06.446 MQT: tasmota/discovery/E09806DFECE5/config = {"ip":"192.168.8.213","dn":"T9","fn":["T9",null,null,null,null,null,null,null],"hn":"T9-3301","mac":"E09806DFECE5","md":"Gosund EP2","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.3.1","t":"T9","ft":"XXX/%topic%/%prefix%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (beibehalten)
13:31:06.201 HLW: EPC 4, CFlen 286912 usec
13:31:05.200 HLW: EPC 4, CFlen 250280 usec
13:31:04.201 HLW: EPC 4, CFlen 260008 usec
13:31:03.201 HLW: EPC 3, CFlen 254954 usec
13:31:03.119 MQT: XXX/T9/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:31:03","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":12.960,"Yesterday":0.136,"Today":0.077,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:31:03.109 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:31:03.106 SRC: MQTT
13:31:02.201 HLW: EPC 4, CFlen 253085 usec
13:31:01.429 MQT: XXX/T9/tele/SENSOR = {"Time":"2023-01-29T13:31:01","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":12.960,"Yesterday":0.136,"Today":0.077,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
13:31:01.398 MQT: XXX/T9/tele/STATE = {"Time":"2023-01-29T13:31:01","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"F","BSSId":"AA:BB:CC:DD:EE:FF","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:31:01.200 HLW: EPC 4, CFlen 261726 usec
13:31:00.627 CFG: in Flash gespeichert an F5, Anzahl 225, Bytes 4096
13:31:00.388 APP: Anzahl Startvorgänge 64
13:31:00.200 HLW: EPC 4, CFlen 259796 usec
13:30:59.200 HLW: EPC 4, CFlen 251904 usec
13:30:58.469 QPC: Reset
13:30:58.200 HLW: EPC 4, CFlen 256157 usec
13:30:57.358 MQT: XXX/T9/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:57","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":12.960,"Yesterday":0.136,"Today":0.077,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:30:57.240 CMD: Grp 0, Cmd 'POWER', Idx 0, Len 0, Pld -99, Data ''
13:30:57.238 SRC: MQTT
13:30:57.245 MQT: XXX/T9/stat/RESULT = {"POWER":"OFF"}
13:30:57.349 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:57.347 SRC: MQTT
13:30:57.250 MQT: XXX/T9/stat/POWER = OFF
13:30:57.200 HLW: EPC 4, CFlen 262591 usec
13:30:57.128 MQT: XXX/T9/stat/RESULT = {"POWER":"OFF"}
13:30:57.084 MQT: XXX/T9/tele/INFO3 = {"Info3":{"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["40003b55","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025964c","4025967f","4024f4ea","4027720a","4024f523","4024ed1e","40000f49","40000f49","40000e19","40105acd","40105ad3","4010000d","4026fc3c","4026fbed","40101421","40101410","40101348","4000050c","40103456","40100aa0","40101348","40102714","40100aa0","401032ab","40102714","401059c7","4010348c","40100aa0","40102714","401059c7","40100230"]},"BootCount":64}}
13:30:57.065 MQT: XXX/T9/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"T9-3301","IPAddress":"192.168.8.213"}}
13:30:57.133 MQT: XXX/T9/stat/POWER = OFF
13:30:57.045 MQT: abonniere XXX/tasmotas/cmnd/#
13:30:57.055 MQT: XXX/T9/tele/INFO1 = {"Info1":{"Module":"Gosund EP2","Version":"12.3.1(tasmota)","FallbackTopic":"cmnd/14675173_fb/","GroupTopic":"XXX/tasmotas/cmnd/"}}
13:30:57.049 MQT: abonniere cmnd/14675173_fb/#
13:30:57.034 MQT: XXX/T9/tele/LWT = Online (beibehalten)
13:30:57.040 MQT: abonniere XXX/T9/cmnd/#
13:30:57.038 MQT: XXX/T9/cmnd/POWER = 
00:00:02.200 HLW: EPC 4, CFlen 259811 usec
00:00:01.754 WIF: Verbindungsversuch...
00:00:01.753 WIF: Prüfe Verbindung...
00:00:01.201 HLW: EPC 6, CFlen 262806 usec
13:30:57.031 MQT: verbunden
13:30:57.011 MQT: Verbindungsversuch...
[... cut ...]
13:30:02.698 MQT: XXX/T9/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:02","ENERGY":{"TotalStartTime":"2022-10-28T16:53:17","Total":12.960,"Yesterday":0.136,"Today":0.076,"Power":6,"ApparentPower":15,"ReactivePower":14,"Factor":0.37,"Voltage":227,"Current":0.067}}}
13:30:02.687 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:03.201 HLW: EPC 4, CFlen 253905 usec
13:30:02.685 SRC: MQTT
13:30:02.201 HLW: EPC 4, CFlen 261853 usec
13:30:01.201 HLW: EPC 4, CFlen 255451 usec
13:30:00.201 HLW: EPC 4, CFlen 264901 usec
13:29:59.201 HLW: EPC 4, CFlen 256009 usec
....

```

florianwgnr, 2023-01-29:
Might be unrelated, but the second EP2 did not show these "HLW: EPC 5, CFlen 191089 usec" entries until today, starting at a time just around the crash: 

```
...
13:31:22.218 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:31:22","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.249,"Yesterday":0.000,"Today":0.000,"Power":12,"ApparentPower":40,"ReactivePower":38,"Factor":0.31,"Voltage":223,"Current":0.178}}}
13:31:22.208 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:31:22.205 SRC: MQTT
13:31:22.200 HLW: EPC 8, CFlen 129104 usec
13:31:21.201 HLW: EPC 7, CFlen 128460 usec
13:31:20.201 HLW: EPC 8, CFlen 132251 usec
13:31:19.201 HLW: EPC 9, CFlen 128945 usec
13:31:18.201 HLW: EPC 13, CFlen 76319 usec
13:31:17.201 HLW: EPC 13, CFlen 76465 usec
13:31:16.201 HLW: EPC 13, CFlen 76616 usec
13:31:15.201 HLW: EPC 13, CFlen 77728 usec
13:31:14.201 HLW: EPC 14, CFlen 77135 usec
13:31:13.201 HLW: EPC 13, CFlen 75060 usec
13:31:12.281 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:31:12.290 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:31:12","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.249,"Yesterday":0.000,"Today":0.000,"Power":21,"ApparentPower":47,"ReactivePower":42,"Factor":0.45,"Voltage":223,"Current":0.210}}}
13:31:12.278 SRC: MQTT
13:31:12.201 HLW: EPC 13, CFlen 75414 usec
13:31:11.201 HLW: EPC 14, CFlen 73756 usec
13:31:10.470 WIF: Prüfe Verbindung...
13:31:10.201 HLW: EPC 13, CFlen 74234 usec
13:31:09.201 HLW: EPC 14, CFlen 72840 usec
13:31:08.201 HLW: EPC 16, CFlen 76604 usec
13:31:07.201 HLW: EPC 19, CFlen 78346 usec
13:31:06.201 HLW: EPC 21, CFlen 49050 usec
13:31:05.205 HLW: EPC 17, CFlen 47905 usec
13:31:04.201 HLW: EPC 10, CFlen 74970 usec
13:31:03.201 HLW: EPC 7, CFlen 132777 usec
13:31:02.277 SRC: MQTT
13:31:02.201 HLW: EPC 8, CFlen 79138 usec
13:31:02.290 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:31:02","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":13,"ApparentPower":13,"ReactivePower":0,"Factor":1.00,"Voltage":223,"Current":0.058}}}
13:31:02.280 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:31:01.423 WIF: Sending Gratuitous ARP
13:31:01.201 HLW: EPC 2, CFlen 528268 usec
13:31:00.201 HLW: EPC 2, CFlen 539989 usec
13:30:59.201 HLW: EPC 2, CFlen 498937 usec
13:30:52.220 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:52.231 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:52","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:30:52.217 SRC: MQTT
13:30:50.461 WIF: Prüfe Verbindung...
13:30:42.312 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:42.323 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:42","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:30:42.309 SRC: MQTT
13:30:32.247 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:32.257 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:32","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:30:32.244 SRC: MQTT
13:30:30.431 WIF: Prüfe Verbindung...
13:30:22.238 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:22","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:30:22.228 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:22.225 SRC: MQTT
13:30:12.303 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:12","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:30:12.292 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:12.289 SRC: MQTT
13:30:10.462 WIF: Prüfe Verbindung...
13:30:01.373 WIF: Sending Gratuitous ARP
13:30:02.390 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:30:02","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:30:02.379 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:30:02.376 SRC: MQTT
13:29:52.215 MQT: XXX/T10/stat/STATUS8 = {"StatusSNS":{"Time":"2023-01-29T13:29:52","ENERGY":{"TotalStartTime":"2022-10-28T16:53:23","Total":48.248,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":223,"Current":0.000}}}
13:29:52.204 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
13:29:52.201 SRC: MQTT
13:29:50.445 WIF: Prüfe Verbindung...
...
```

barbudor, 2023-01-29:
The HLW: EPC logs are from the HLW8012 energy monitoring driver
Normal with a log level of 4

Unfortunately, I don't see anything in the logs prior to the crash any usefull information
I'm checking  if we can get some valuable info from the stackdump in INFO3
```
13:30:57.084 MQT: XXX/T9/tele/INFO3 = {"Info3":{"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["40003b55","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025964c","4025967f","4024f4ea","4027720a","4024f523","4024ed1e","40000f49","40000f49","40000e19","40105acd","40105ad3","4010000d","4026fc3c","4026fbed","40101421","40101410","40101348","4000050c","40103456","40100aa0","40101348","40102714","40100aa0","401032ab","40102714","401059c7","4010348c","40100aa0","40102714","401059c7","40100230"]},"BootCount":64}}
```

barbudor, 2023-01-29:
What is the value of `SetOption63` ?
If `1` can you try to set it to `0` ?

https://tasmota.github.io/docs/Commands/#setoption63

florianwgnr, 2023-01-30:
> What is the value of `SetOption63` ? If `1` can you try to set it to `0` ?
> 
> https://tasmota.github.io/docs/Commands/#setoption63

`SetOption63` is set to `0` on both devices.

martin072, 2023-02-04:
I Have a few EP2's running with Tasmota 12.0.2 with the EP2 template. Have not seen this behaviour. Have one up for >125 days, with the relay on for most of that time..

barbudor, 2023-02-04:
I really have no idea on both of the problems

Have you tried to upgrade 1 to latest dev ?
Sometime you can have some luck ...

florianwgnr, 2023-02-05:
T9 crashed again with the CallChain being identically to the last crash: 

`19:18:39.110 MQT: SmartPlug/T9/tele/INFO3 = {"Info3":{"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["40003b53","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025964c","4025967f","4024f4ea","4027720a","4024f523","4024ed1e","40000f49","40000f49","40000e19","40105acd","40105ad3","4010000d","4026fc3c","4026fbed","40101421","40101410","40101348","4000050c","40103456","40100aa0","40101348","40102714","40100aa0","401032ab","40102714","401059c7","4010348c","40100aa0","40102714","401059c7","40100230"]},"BootCount":65}}`

> Have you tried to upgrade 1 to latest dev ? Sometime you can have some luck ...
i will update this one to the latest dev and test if it crashes again.

Jason2866, 2023-02-05:
Imho it is a defect device.