arendst/Tasmota #17779: Tasmota auto configuration not applied after factory reset

darkxst, 2023-01-23:
### PROBLEM DESCRIPTION
Using a bunch of  ESP32 device with a autoconf configuration loaded and have noticed the configuration is not reinstalled after a factory reset.

since the `.autoconf` file is not removed as part of the reset, the device does not get the configuration loaded again until one manual re-applies the auto configuration profile. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ZBBridge-Pro, T-Display-S3, any other ESP32 device with Autoconf loaded
- [ ] Tasmota binary firmware version number used: v12.3.1.4
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

### TO REPRODUCE
- Load a auto-conf profile
- perform factory reset
- device will boot in the ESP-Devkit Module configuration

### EXPECTED BEHAVIOUR
 If possible, it would make sense to clean up this file as part of the reset process so the autoconf profile is reloaded on first boot after reset. This would make the first boot after reset consistent with first boot after flashing an image that is configured with auto-conf such as ZBBridge-Pro.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-01-24:
Well, "factory reset" means a full reset not retaining any configuration. If you want to have the `.autoconf` file applied again, you could create a small Berry script to do this, after detecting initial configuration state. The script could then delete, and trigger a restart.

darkxst, 2023-01-24:
I am thinking of for example `tasmota32-zbbrdgpro.bin` which includes an autoconf file in the firmware. On first boot after flashing it loads that config. However after reset it doesnt load the factory included config. But I guess it probably not that easy to determine if the autoconf file is actually the one from the firmware.

Jason2866, 2023-01-24:
That's a corner case. It is easy to to start the Autoconfig manually when needed.
It is more or less impossible to guess when the autoconfig is wanted to be applied again.
To avoid unwanted side effects such a guess work feature will not implemented.

closed 2023-01-24 by Jason2866