arendst/Tasmota #17789: Sonoff basic has trouble connecting to wifi after update.

Martin-Dolan, 2023-01-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After updating several sonoff basic modules to 12.3.1 if I use the 'lite' version, they work fine.  If I use the 'sensors' version (which was my goal as I have BME280 on some of them)  or if I use the 'standard' version, they wont connect to wifi.  After entering wifi ssid and password they try but don't connect.
I connected one to my computer via serial adapter, reflashed using tasmotizer including a full erase etc..  Also tried the updates via the module web interface and also the tasmota web installer via computer serial port.   I can connect to it via phone wifi, and enter the credentials, and when it reboots it won't connect.    Tried uploading wifi credentials via tasmotizer, also didnt connect on reboot.

What I've tried:
  * Sonoff basic modules with original memory chip
  * Sonoff basic modules with 4mb memory chip update
  * On my actual wifi SSID (which has some not normal characters)
  * On another wifi ssid just called "test" in case the strange characters were the issue
  * On above wifi network with all the fancy wifi6 stuff turned off

what works:
  * Sonoff basic modules with all of the firmware versions I used up to and including 9.3.1  lite/standard/sensors  using modules with normal memory or upgraded 4mb flash memory
  * Sonoff basic modules with version 12.3.1 LITE  using modules with normal memory or upgraded 4mb flash memory
  *  All my other devices (smart plugs, light switches etc) with tasmota 12.3.1 sensors or standard  (just letting you know this so you know my wifi has no problems with those)

what doesn't work:
  * Sonoff basic modules with version 12.3.1 sensors or standard using modules with normal memory or upgraded 4mb flash memory

* Also tried adjusting wifi transmit power lower as I read it might be a voltage /power consumption issue
* tried it using 3.3v from the computer serial adaptor
* tried it using the 240v sonoff basic internal powersupply

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-01-25:
What exact type of Sonoff Basics the are? Revision? Maybe you post Photos?

stefanbode, 2023-01-25:
Since 12.3 we have several issues with wifi connect or wifi loss afte some time and device gets disconnected. some mentioned power supply is the issue. Maybe. I also have a Zigbee Bridge showing the same behavior. For me my first guess is to change from dynamic sleep to normal (setoption60) and then define sleep 20. Cannot guarantee, but it looks like that these devices stay online.

Jason2866, 2023-01-25:
@stefanbode  Maybe the fix https://github.com/arendst/Tasmota/pull/17786 merged today does help/solve.
Anyways if it is Sonoff Basic R2 it is useless to search deeper. The reason why the Tasmota-lite Version exists is this device. It is not able to run a full Tasmota version. Hardware design is crap. Especially the power supply.

Jason2866, 2023-01-25:
> Since 12.3 we have several issues with wifi connect or wifi loss afte some time and device gets disconnected. some mentioned power supply is the issue. Maybe. I also have a Zigbee Bridge showing the [...]

None of my devices has issues. No more Sonoff Basics. The all died (with defect power supply) after a few years.

epandi, 2023-01-25:
I'm having exactly the same issue on a sonoff basic (non v2) with a bme280 sensor attached to it combines with the Tasmota Sensors build. When reverting back to 12.2.0 all works fine.

Jason2866, 2023-01-25:
@epandi Can you try latest development version? There is a fix of a regression regarding sleep function not restored to switch back to a higher sleep level. Sleep is active and working but the default higher value before was not restored.

epandi, 2023-01-25:
@Jason2866 thanks for the suggestion, I've just flashed 12.3.1.4 (sensors) dev from the Ota server and my sonoff basic device did come back onlinr after this upgrade. Also my bme280 sensor information is sent over mqtt like before. For more this issue is resolved, I hope this will also resolve the issue from @Martin-Dolan.

barbudor, 2023-01-25:
@epandi Could you please share your Sonoff Basic configuration ?
Are you using PWM outputs or Led PWM commands on that device ?

epandi, 2023-01-25:
@barbudor 

22:49:55.174 CMD: Backlog Template; Module; GPIO 255:
22:49:55.319 MQT: stat/whiskeykast/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:49:55.537 MQT: stat/whiskeykast/RESULT = {"Module":{"1":"Sonoff Basic"}}
22:49:55.743 MQT: stat/whiskeykast/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"640":"I2C SDA"},"GPIO2":{"0":"None"},"GPIO3":{"608":"I2C SCL"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

22:46:58.437 CMD: Status 0
22:46:58.446 MQT: stat/whiskeykast/STATUS = {"Status":{"Module":1,"DeviceName":"WhiskeyKast","FriendlyName":["WhiskeyKast"],"Topic":"whiskeykast","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:46:58.483 MQT: stat/whiskeykast/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:37","StartupUTC":"2023-01-25T21:39:21","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2022-12-17T11:24:41","SaveCount":37,"SaveAddress":"FA000"}}
22:46:58.518 MQT: stat/whiskeykast/STATUS2 = {"StatusFWR":{"Version":"12.3.1.4(sensors)","BuildDateTime":"2023-01-25T10:56:52","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"366/699"}}
22:46:58.538 MQT: stat/whiskeykast/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxx","xxxxxxxxxx"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["40008009","2805C80001000600003C5A0A192800000000","000000C9","00006000","00004000","00000000"]}}
22:46:58.576 MQT: stat/whiskeykast/STATUS4 = {"StatusMEM":{"ProgramSize":695,"Free":308,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","00004081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98"}}
22:46:58.622 MQT: stat/whiskeykast/STATUS5 = {"StatusNET":{"Hostname":"whiskeykast-3662","IPAddress":"192.168.1.11","Gateway":"192.168.0.1","Subnetmask":"255.255.252.0","DNSServer1":"8.8.4.4","DNSServer2":"8.8.4.4","Mac":"5C:CF:7F:72:CE:4E","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
22:46:58.646 MQT: stat/whiskeykast/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_72CE4E","MqttUser":"sonoff","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:46:58.672 MQT: stat/whiskeykast/STATUS7 = {"StatusTIM":{"UTC":"2023-01-25T21:46:58","Local":"2023-01-25T22:46:58","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:29","Sunset":"17:35"}}
22:46:58.694 MQT: stat/whiskeykast/STATUS10 = {"StatusSNS":{"Time":"2023-01-25T22:46:58","BME280":{"Temperature":20.0,"Humidity":39.1,"DewPoint":5.7,"Pressure":1025.9},"PressureUnit":"hPa","TempUnit":"C"}}
22:46:58.711 MQT: stat/whiskeykast/STATUS11 = {"StatusSTS":{"Time":"2023-01-25T22:46:58","Uptime":"0T00:07:37","UptimeSec":457,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"64:D1:54:F9:F3:E5","Channel":10,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":4,"Downtime":"0T00:00:43"}}}

epandi, 2023-01-25:
I guess In was cheering a bit too fast. Device lost connection. But at least the connection was up for a moment. I'll check the device tomorrow if it got its connection back.

stefanbode, 2023-01-26:
I might be wrong. Following findings. 
Most are sonoff devices (e.g.zigbee Bridge) but also one wemos D1 mini. 
All have a bit complicated WIFI access. Quite far away. 
Device seems not to crash. Button still work. Serial not tested for sure. Mains connected. 240V. Dangerous!! Power the device solves problem in my cases.

stefanbode, 2023-01-26:
I cannot confirm that the light version does not have the issue. Same there. I updated on of my devices to .4 version right now and change back to Setoption60 off. Let's see what's happening.

Martin-Dolan, 2023-01-26:
Sorry for the delay in getting back to you all.
Here is the photo as requested of one of the devices and its board.  (This is one with an upgraded flash, and pin header added but I have others that are "standard" with same issue.  

[image]

I will update the firmware to the dev branch version on one and see if it works..  (one that I don't have to climb ladders to get to it if it screws up haha)

Martin-Dolan, 2023-01-26:
Update.  Didn't work, can't even get phone to connect to it  :(
I've got to go out for a few hours, will try again when I get back

Jason2866, 2023-01-26:
@Martin-Dolan This is the known bad batch R2 version. The train has ended for that device with a Tasmot version that luckily works. Tasmota-lite is the smallest version we can do.

@stefanbode Replace the to weak LDO on the board of the Wemos D1 mini clone, and your problems will be gone. Original D1 mini has a more powerful one.

Jason2866, 2023-01-26:
Closing since the issue is caused from the hardware. The powers supply is too weak to power the esp8266 as needed.

closed 2023-01-26 by Jason2866

Martin-Dolan, 2023-01-26:
Thanks for the update and letting me know.  What is the best alternative these days ?

Jason2866, 2023-01-26:
My choice is a Shelly1

epandi, 2023-01-26:
> Since 12.3 we have several issues with wifi connect or wifi loss afte some time and device gets disconnected. some mentioned power supply is the issue. Maybe. I also have a Zigbee Bridge showing the [...]

For me this reply from @stefanbode did the trick in my sonoff basic R1( backlog Setoption60 1; sleep 20; ) . My device stays online and keeps responding ❤️.

Martin-Dolan, 2023-01-26:
One last question, do shelly1 devices have some spots I can solder some extra wires to add access to other IO lines for BME280 humidity sensor etc?

Jason2866, 2023-01-26:
Nope since https://tasmota.github.io/docs/devices/Shelly-1/

stefanbode, 2023-01-27:
Sonoff with 12.3.1.4 lite and dynamic sleep is online for about a day. Keep you postet RSSI strength is 54

wchulze, 2023-01-27:
I don't understand the bad power supply. Martin-Dolan tried to supply the device externally. He wrote:
**[Martin-Dolan](https://github.com/Martin-Dolan) tried it using 3.3v from the computer serial adaptor**

With a bad power supply, a computer / microcontroller typically crashes completely. That part of the hardware works and another part doesn't is strange.

lalo-uy, 2023-01-27:
The computer serial adaptor is not a very good power supplier.
At least add a 470uF in parallel with it.
Better get an independent power supply

stefanbode, 2023-01-28:
Update: device still online. 2d5h. Power supply is essential for operating. Quite strange things can happen. Therefore on any self developed board it is good advice to solder a 470mircoF capacitor on the 3.3V line. This stabilize quite a lot. Adding this capacitor on the 3.3V rail would most likely also solve the problem if power supply is the issue. At least for the firmware update the the GND and 3.3 is exposed on Pins. Should be no big deal to add a capacitor here. I will now a POW2 with tamp sensor with the .4 firmware let's see what happen here. This will not be the lite version.

wchulze, 2023-01-28:
In this case you use Low-ESR type. Switching power supplies are relatively high-frequency. The series inductance and the series resistance of the capacitor should be as low as possible.

Martin-Dolan, 2023-02-01:
I tried  a low-ESR cap (10v 1000uf) and it didn't help :(  
This was while using a usb to provide the power.

I have some of those Hi-link HLK-5M03 power supply modules, I might try as an experiment to isolate the board power supply and try the Hi-link if possible

stefanbode, 2023-02-02:
I personally have concerns that power is the reason. But @Jason2866 is 100% sure and he quite more experienced than I am. To summarize my latest findings:
Seems only happen on bad wlan connection. RSSI 40 or below. 
Setoption60 On to change the sleep from dynamic do static seems to fix it
Also last recent versions not really make a difference. Anyhow one sonoff basic with .4 version and a RSSI of 51 seems to be stable.

wchulze, 2023-02-05:
At the moment I'm observing 3 posts that are said to have problems with the power supply. It seems to me that Tasmota version 12.3.1 is being used there. I've been using Sonnoff Basic, Wemos clones and Esp-01 hardware for a long time without problems with older Tasmota versions.

BartAgterbosch, 2023-07-12:
I have a bunch of Sonoff Basic R2 (a whole batch of them actually) that have worked fine for years with older tasmota versions, but after updating them to the latest version they all fail to connect to wifi, and putting them into setup/ap mode will broadcast an ssid, but any devices that I try to connect to it do not receive an ip address.
Again, they all worked fine for the past 5 to 6 years using older tasmota versions, but with the latest it does not, meaning I highly doubt it's a weak powersupply