arendst/Tasmota #17799: DS3231 Tasmota Incorrect year

gfn256, 2023-01-27:
I checked the time/date set by Tasmota on the RTC and all is correct, except for the year being 2053 instead of 2023.
I know tasmota is very advanced and ahead of its time, but 30 years is a bit much!!

barbudor, 2023-01-27:
I will have a look tonight

sfromis, 2023-01-27:
Maybe this is somehow the upcoming [Y2K38](https://en.wikipedia.org/wiki/Year_2038_problem) trouble popping up? Year 2023 is 15 years before the Unix timestamp rollover, and 2053 is 15 years after the Epochalypse.

barbudor, 2023-01-27:
So indeed I can see that tasmota is breaking the `time_t` epoch into an almost standard `struct tm` except that the `tm.year` field in giving the number of years from 1970 instead of the standard full 4-digit year.
So what gets written in the register is that value : (this year - 1970)
As well the `century bit` is not set

There is indeed a risk Feb 28th or a Feb 29th (depending of the year) as the chip may not calculate correctly the leap year.

So probably best to set correctly the century bit and shift the year correctly.

PR #17803 (2023-01-28): non-breaking fix year in DS3231
## Description:

**Related issue (if applicable):** fixes #17799

As of today, the DS3231 driver write the Year register with number of years since 1970 while the chip expect the last 2 digit of the true year
For example, in 2024 the driver would have written 54 instead of 24

This could potentionally create some issues on leap years as 54 would not be identified as a leap year while 2024 is a leap year (and also an olympic year)

This PR fixes that in a non-breaking way for existing DS3221 user, as after this PR:
- year is always written as expected by the chip [... cut]
closed 2023-01-28 by arendst