arendst/Tasmota #17805: Sonoff NSPanel

TrustedFloppa, 2023-01-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sonoff NSPanel dosn't want to connect to MQTT even when adress is correct password also is correct

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff NSPanel
- [X] Tasmota binary firmware version number used: v12.3.1 
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmota Web Installer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`: 12:05:41.161 CMD: Backlog Template; Module; GPIO 255
12:05:41.195 RSL: RESULT = {"NAME":"NSPanel","GPIO":[0,0,0,0,3872,0,0,0,0,0,32,0,0,0,0,225,0,480,224,1,0,0,0,33,0,0,0,0,0,0,0,0,0,0,4736,0],"FLAG":0,"BASE":1}
12:05:41.415 RSL: RESULT = {"Module":{"0":"NSPanel"}}
12:05:41.671 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"3872":"Output Lo"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"32":"Button1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"225":"Relay2"},"GPIO20":{"0":"None"},"GPIO21":{"480":"Buzzer"},"GPIO22":{"224":"Relay1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"33":"Button2"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"4736":"ADC Temp1"},"GPIO39":{"0":"None"}}
12:05:45.637 RSL: RESULT = {"CustomRecv":"event,startup,48,eu"}
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: 12:07:03.918 CMD: status 0
12:07:03.922 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"Sonoff_NSPanel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:07:03.927 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T01:25:21","StartupUTC":"2023-01-28T09:41:42","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2023-01-27T19:14:36","SaveCount":59}}
12:07:03.935 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:31:03","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"448/699"}}
12:07:03.941 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["T-Mobile_Swiatlowod_2.4GHz_0E0E",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
12:07:03.954 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1479,"Free":1400,"Heap":124,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D43C482D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
12:07:03.965 RSL: STATUS5 = {"StatusNET":{"Hostname":"Sonoff-NSPanel-2612","IPAddress":"192.168.1.15","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"7C:87:CE:C8:2A:34","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:07:03.972 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C82A34","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:07:03.979 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-01-28T11:07:03","Local":"2023-01-28T12:07:03","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"17:40"}}
12:07:03.990 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-01-28T12:07:03","ANALOG":{"Temperature1":4.4},"TempUnit":"C"}}
12:07:03.999 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-01-28T12:07:03","Uptime":"0T01:25:21","UptimeSec":5121,"Heap":123,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":16,"Objects":240},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"T-Mobile_Swiatlowod_2.4GHz_0E0E","BSSId":"34:5D:9E:1F:0E:12","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
My Sonoff NSPanel would connect to MQTT, i need to connect it to mqtt because it can't be used without it

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-01-28:
[image]

No logs
No help

TrustedFloppa, 2023-01-28:
Updated