arendst/Tasmota #17810: After upgrade to 12.3.1.5 Shelly Plus 2PM shows VEML6070 device

Joylancer, 2023-01-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After upgrade to 12.3.1.5 Shelly Plus 2PM shows VEML6070 device
no voltage, 
[image]

- FIX: i2cdriver12 0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Shelly Plus 2PM V 0.1.9 singlecore
- [ x] Tasmota binary firmware version number used: 12.3.1.5 tasmota32solo1.bin
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:40:50.786 MQT: stat/Shelly2PM-2/RESULT = {"NAME":"Shelly Plus 2PM v0.1.9","GPIO":[320,0,0,0,32,193,0,0,224,225,0,0,0,0,192,0,0,0,0,0,0,608,640,3456,0,0,0,0,0,0,0,4736,7584,0,0,7585],"FLAG":0,"BASE":1}
19:40:51.831 MQT: stat/Shelly2PM-2/RESULT = {"Module":{"0":"Shelly Plus 2PM v0.1.9"}}
19:40:52.896 MQT: stat/Shelly2PM-2/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"193":"Switch_n2"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"192":"Switch_n1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"3456":"ADE7953 IRQ1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"4736":"ADC Temp1"},"GPIO36":{"7584":"HallEffect1"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"7585":"HallEffect2"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:41:48.015 MQT: stat/Shelly2PM-2/STATUS = {"Status":{"Module":0,"DeviceName":"Shelly2PM-2","FriendlyName":["Line1","Line2"],"Topic":"Shelly2PM-2","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:41:48.026 MQT: stat/Shelly2PM-2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32solo1.bin","RestartReason":"Software reset CPU","Uptime":"0T00:07:26","StartupUTC":"2023-01-28T18:34:22","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-10-31T14:53:30","SaveCount":53}}
19:41:48.033 MQT: stat/Shelly2PM-2/STATUS2 = {"StatusFWR":{"Version":"12.3.1.5(tasmota)single-core","BuildDateTime":"2023-01-28T12:51:10","Core":"2_0_6","SDK":"v4.4.3.20221227","CpuFrequency":240,"Hardware":"ESP32-U4WDH-S rev.3","CR":"450/699"}}
19:41:48.040 MQT: stat/Shelly2PM-2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["kein_Internet_",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00208009","2805C80001000600003C5A0A192800000000","00000080","00006000","0000C000","00000001"]}}
19:41:48.055 MQT: stat/Shelly2PM-2/STATUS4 = {"StatusMEM":{"ProgramSize":1496,"Free":1383,"Heap":161,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D43C482D","000000B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
19:41:48.067 MQT: stat/Shelly2PM-2/STATUS5 = {"StatusNET":{"Hostname":"Shelly-Plus-2PM-2","IPAddress":"192.168.178.88","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"44:17:93:A7:C2:A0","IP6Global":"fd00::4617:93ff:fea7:c2a0","IP6Local":"fe80::4617:93ff:fea7:c2a0","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
19:41:48.079 MQT: stat/Shelly2PM-2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.100","MqttPort":1883,"MqttClientMask":"Shelly2PM-2","MqttClient":"Shelly2PM-2","MqttUser":"myhome","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:41:48.086 MQT: stat/Shelly2PM-2/STATUS7 = {"StatusTIM":{"UTC":"2023-01-28T18:41:48","Local":"2023-01-28T19:41:48","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"08:25","Sunset":"17:40"}}
19:41:48.091 MQT: stat/Shelly2PM-2/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:41:48.110 MQT: stat/Shelly2PM-2/STATUS10 = {"StatusSNS":{"Time":"2023-01-28T19:41:48","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":24.3},"ENERGY":{"TotalStartTime":"2022-10-31T14:53:30","Total":[0.000,0.000],"Yesterday":[0.000,0.000],"Today":[0.000,0.000],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":223750,"Voltage":0,"Current":[0.000,0.000]},"VEML6070":{"UvLevel":65535,"UvIndex":99.00,"UvIndexText":"OoR","UvPower":2.475},"ESP32":{"Temperature":64.4,"HallEffect":1},"TempUnit":"C"}}
19:41:48.123 MQT: stat/Shelly2PM-2/STATUS11 = {"StatusSTS":{"Time":"2023-01-28T19:41:48","Uptime":"0T00:07:26","UptimeSec":446,"Heap":159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":2095,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"kein_Internet_","BSSId":"B0:F2:08:67:7F:20","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
12.3.1.4 no VEML6070 device
upgrade to 12.3.1.5
shows VEML6070 device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

- FIX: i2cdriver12 0 working
**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-01-28:
VEML6070 has been part of tasmota32 and tasmota32solo1 for the last 2 years
So you probably made something at some point during the upgrade, re-enabling I2CDriver12
Like a full erase/reset of the board or of the settings.

petermicuch, 2023-01-28:
Actually I noticed the same thing. I was advised to upgrade to latest development version 12.3.1.5 while investigating different [issue](https://github.com/arendst/Tasmota/issues/17811) and could see, that value of voltage/power were shown as `0` and then they again show some values. Also frequency was showing very strange values. Not sure if it is VEML6070 related, but something is wrong with values shown (appearing, disappearing, incorrect).

Joylancer, 2023-01-28:
full reset 
switch to template from blakadder:
{"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,32,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3456,0,0,0,0,0,0,0,4736,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,3350"}
the same, need to disable i2cdriver 12
[image]

tyeth, 2023-01-28:
I had problems with it showing, but I assumed it was related to the VEML7700 I had deliberately enabled. The driver device check for this is non-existent and it blocks the Veml7700. I may attempt to resolve it as having it in all official builds is a blocker for me.

barbudor, 2023-01-28:
@Joylancer 
> full reset

Restore I2CDriver to all drivers enabled (tasmota default)
Why reset for a simple upgrade ?
Just OTA and you keep your settings

@tyeth 
> The driver device check for this is non-existent 

There is unfortunately no standard for I2C device detection. I2C was designed for when engineers knows what they put on their boards, not for hobbyists.
Attempts to test if a device is X may also crash the device if it's Y.
May be the best would be to have all I2C drivers disabled by default ? This is what I have is my own builds

Joylancer, 2023-01-28:
full reset
-- only after your entry
Restore I2CDriver to all drivers enabled (tasmota default)
-- all i2cdriver enabled (22:51:25.933 MQT: stat/Shelly_2PM_WZ-TV/RESULT = {"I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,69"})

Why reset for a simple upgrade ?
Just OTA and you keep your settings
- after OTA the VEML was present, before not, with 12.3.1.4 it was also not disabled (i2cdriver12 1).
testet on 2 devices, same problem

Joylancer, 2023-01-28:
downgrade to 12.3.1.1
command: i2cdriver shows only:
23:02:03.041 MQT: stat/Shelly_2PM_WZ-TV/RESULT = {"I2CDriver":"7"}

update to 12.3.1.5
command: i2cdriver shows :
23:11:34.313 MQT: stat/Shelly_2PM_WZ-TV/RESULT = {"I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,69"}

Joylancer, 2023-01-28:
the problem is: i2cdriver 7 and 12 has the same address 0x38
23:13:46.957 MQT: stat/Shelly_2PM_WZ-TV/RESULT = {"I2CScan":"Device(s) found on bus1 at 0x38"}

7 | USE_ADE7953 | xnrg_07 | ADE7953 | 0x38 | Energy monitor (existent in Shelly Plus 2PM)

12 | USE_VEML6070 | xsns_11 | VEML6070 | 0x38 - 0x39 | Ultra violet light intensity sensor

Joylancer, 2023-01-28:
it the same problem when use AHT22 sensor, you need to disable driver 12
is no problem, if it is published with the templates (as a hint)

tyeth, 2023-01-28:
> There is unfortunately no standard for I2C device detection. I2C was designed for when engineers knows what they put on their boards, not for hobbyists.
> Attempts to test if a device is X may also [...]

Understood, but anything that has feature detection implicitly by checking a name or serial number or successfully retrieving some result more than just a single byte check of a register, should be able to know if the device at said address is actually that family of devices. If that's not possible then the driver doesn't deserve to be loaded by default in any of the firmware builds. 

It would also make more sense to include the latest generation of each family if one is to be included in builds by default (The AHT2x instead of AHT1x for example and VEML7700 instead of 6070 as they share addresses).

sfromis, 2023-01-28:
I also saw the VEML6070 wrongly appear after upgrading to 12.3.1.5, without doing anything about other config stuff.
Mitigated using `I2CDriver12 0`

sfromis, 2023-01-28:
When building binaries, I've made the choice of initially having all I2C drivers off, and only enabling what I need. With autoconfig or just templates, such enabling could be included per device type. Of course, such would still be a breaking change, but I suppose that selective enabling would give less pain than needing to selectively disable when something fails.

tyeth, 2023-01-29:
Sorry I forgot to answer @barbudor - I dislike the need to disable all drivers by default. I personally see tasmota as an I2C remote control / sensor platform more than a GPIO/relay trigger, but I'm pretty new to it. I haven't stuck it on any of my sonoffs because ewelink+google works fine for me. I'm looking at using Tasmota for my own open-source thing built with the LilyGo ESP32S3 T-Display. I was relieved to see that some I2C drivers are included by default in a lot of the larger builds like lvgl/display.

Sure it's okay to undefine/disable-by-default all drivers in your custom builds or as part of safeboot / minimal, but to do that to all builds due to a couple of bad drivers (don't check if a device is correct device) instead of undefining the miscreants/misbehavers seems like a backwards step. Ideally they should all fail gracefully or be undefined in all builds by default. Maybe turning off only the bad drivers makes sense, rather than undefining, but I prefer them all defined if safe or undefined if bad / fixed.
  I also accept that theoretically the engineer (us) should know their device, and potentially after optimisation should only include minimal drivers, and be happy compiling custom firmware whenever they want to change that initial decision or to disable/enable something that can't be done via the webUI.
 The reality of saving time and premature optimisation makes me say forget optimisation and initially include anything that doesn't swell the firmware ridiculously and wont break other things. I did also generate a load of I2cDriverXX script to turn off all drivers before knowing how to correctly produce more lean builds, but having to turn them on and off is a pain compared to when they work correctly automatically.

sfromis, 2023-01-29:
It is not due to "bad drivers", it is just not always feasible for each driver to positively verify what driver is physically wired up. If you want a build with lots of I2C drivers included, that is not a good match for wanting it to "always" work without configuring. Maybe more could be done in some drivers, but then comes another I2C device messing things up.

arendst, 2023-01-29:
I2C feature detection is nice. It's also an utopie especially while dealing with Tasmota I2C legacy support. 

To use an I2C bus the master has to go over the bus and look for devices it can support. In an ever increasing number of I2C drivers available duplicate addresses are expected. As the I2C protocol does no support device recognition the only way to detect devices is by polling their address and hoping to get an answer OR write to known registers and expect known values. The latter is likely causing issues on devices not supporting register requests they do not support and might even hang the bus. 

The way Tasmota works is that it scans the bus for I2C devices in order of the way drivers are inserted in the code. This means it first scans the Driver I2C devices (Energy then Display then other drivers with increasing xdrv number) followed by the Sensor I2C devices. It's first come, first serve so if an address is detected it claims it and disables further address use for devices further down the line.

To fix eager I2C devices the command ``I2CDriver`` is introduced to disable devices early in the foodchain to allow detection of devices lower in the I2C chain.

That being said I would expect the VEML not to pop up (as it's address must have been detected by the ADE7953 driver before and disabled further use )but as recently support for a second I2C bus has been added on ESP32 it might have introduced this behaviour.

As a workaround use command ``I2CDriver12 0`` to disable the VEML devive in favour of ADE7953.

Investigating...

arendst, 2023-01-29:
BTW I see your template also is wrong for the Shelly Plus 2PM.

This is the correct one:
```
{"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
```
Having the right order of the switches AND the correct IRQ number distinguising the correct ADE7953 for the Shelly Plus 2PM.

Joylancer, 2023-01-29:
> 

https://templates.blakadder.com/shelly_plus_2PM.html
{"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,32,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3456,0,0,0,0,0,0,0,4736,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,3350"}
i will test your templates.

Joylancer, 2023-01-29:
@arendst 
new template works, output mixed value also correct
need also to disable VEML - i2cdriver12 0
thanks for your work,
@blakadder Please Update the template to:
{"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
with commend to disable VEML (i2cdriver12 0)

closed 2023-01-30 by Joylancer

sfromis, 2023-01-31:
With most recent dev build, I'm able to have Shelly Plus 2PM running without the energy monitoring chip on I2C address 0x38 being detected as VEML6070, even when having the latter driver enabled with `I2Cdriver12 1`

Thus, the issue should now be fixed without any workaround needed, meaning no "surprises" to people upgrading devices.

Joylancer, 2023-02-01:
> With most recent dev build, I'm able to have Shelly Plus 2PM running without the energy monitoring chip on I2C address 0x38 being detected as VEML6070, even when having the latter driver enabled [...]

@arendst 
perfect, thank you for your work