arendst/Tasmota #17811: Shelly Plus 2PM output values mixed and physical switch 2 not working

petermicuch, 2023-01-28:
### PROBLEM DESCRIPTION
_Shelly Plus 2PM mixes output 1 and 2 power,voltage,current measurements and physical switch 2 does not work_

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Plus 2PM with ESP32 dual core
- [x] Tasmota binary firmware version number used: 12.3.1(bluetooth) 2022-12-16T10:31:12
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-flasher executable in windows 10
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:57:34.501 CMD: Backlog Template; Module; GPIO 255
20:57:34.539 RSL: RESULT = {"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,32,192,0,0,225,224,0,0,0,0,192,0,0,0,0,0,0,608,640,3456,0,0,0,0,0,0,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
20:57:34.764 RSL: RESULT = {"Module":{"0":"Shelly Plus 2PM PCB v0.1.9"}}
20:57:35.018 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"192":"Switch_n1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"192":"Switch_n1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"3456":"ADE7953 IRQ1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"4736":"ADC Temp1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
20:58:39.725 CMD: Backlog Rule1; Rule2; Rule3
20:58:39.762 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:58:39.985 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:58:40.234 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:59:26.597 CMD: Status 0
20:59:26.601 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_879FB8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:59:26.640 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"0T00:16:00","StartupUTC":"2023-01-28T19:43:26","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2023-01-28T19:15:04","SaveCount":90}}
20:59:26.679 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1(bluetooth)","BuildDateTime":"2022-12-16T10:31:12","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-U4WDH-D rev.3","CR":"425/699"}}
20:59:26.693 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["T-IVEPE",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
20:59:26.722 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1611,"Free":1856,"Heap":101,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7C7","00148001","000000CF","010013C0","C0000981","40004080","00200000","5400082C","00000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,82,85","Sensors":"1,2,3,5,6,52,62,127"}}
20:59:26.759 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-879FB8-8120","IPAddress":"192.168.1.34","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C0:49:EF:87:9F:B8","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:59:26.804 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_879FB8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:59:26.819 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-01-28T19:59:26","Local":"2023-01-28T20:59:26","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"17:40"}}
20:59:26.844 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:59:26.864 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-01-28T20:59:26","Switch1":"OFF","ANALOG":{"Temperature1":42.9},"ENERGY":{"TotalStartTime":"2023-01-28T19:15:04","Total":0.011,"Yesterday":0.000,"Today":0.011,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
20:59:26.893 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-01-28T20:59:26","Uptime":"0T00:16:00","UptimeSec":960,"Heap":99,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":44},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"T-IVEPE","BSSId":"88:0F:A2:32:43:A6","Channel":7,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:01:52.863 SRC: WebGui from 192.168.1.157
21:01:52.869 RSL: RESULT = {"POWER1":"ON"}
21:01:52.871 RSL: POWER1 = ON
21:01:53.375 CFG: Saved, Count 92, Bytes 4096
21:01:53.845 ADE: ACCMODE 0x2C2C00, VRMS 6112479, 6116142, Period 4479, 4476, IRMS 1773, 27697, WATT 0, -398, VA 28, 435, VAR 0, 0
21:01:54.843 ADE: ACCMODE 0x2C2C00, VRMS 6116275, 6116264, Period 4479, 4477, IRMS 1773, 27755, WATT 0, -431, VA 27, 431, VAR 0, 0
21:01:55.843 ADE: ACCMODE 0x2C2C00, VRMS 6112605, 6110962, Period 4477, 4478, IRMS 1773, 27765, WATT 0, -431, VA 28, 431, VAR 0, 0
21:01:56.843 ADE: ACCMODE 0x2C2C00, VRMS 6117471, 6114243, Period 4478, 4477, IRMS 1773, 27747, WATT 0, -432, VA 27, 432, VAR 0, 0
21:01:57.849 ADE: ACCMODE 0x2C2C00, VRMS 6104048, 6106715, Period 4478, 4478, IRMS 1773, 27753, WATT 0, -432, VA 28, 433, VAR 0, 0
21:01:58.843 ADE: ACCMODE 0x2C2C00, VRMS 6097371, 6093526, Period 4478, 4477, IRMS 1773, 27754, WATT 0, -429, VA 27, 429, VAR 0, 0
21:01:59.843 ADE: ACCMODE 0x2C2C00, VRMS 6090097, 6087377, Period 4477, 4477, IRMS 1773, 27668, WATT 0, -428, VA 28, 428, VAR 0, 0
21:02:00.843 ADE: ACCMODE 0x2C2C00, VRMS 6096140, 6094733, Period 4478, 4478, IRMS 1773, 27737, WATT 0, -429, VA 27, 429, VAR 0, 0
21:02:01.148 BRY: GC from 8115 to 4061 bytes, objects freed 16/44 (in 0 ms) - slots from 49/122 to 46/122
21:02:01.843 ADE: ACCMODE 0x2C2C00, VRMS 6098487, 6099426, Period 4479, 4479, IRMS 1773, 27702, WATT 0, -430, VA 28, 430, VAR 0, 0
21:02:02.801 SRC: WebGui from 192.168.1.157
21:02:02.806 RSL: RESULT = {"POWER1":"OFF"}
21:02:02.807 RSL: POWER1 = OFF
21:02:02.846 ADE: ACCMODE 0x2C2C00, VRMS 6102854, 6100854, Period 4478, 4478, IRMS 1773, 24871, WATT 0, -429, VA 27, 429, VAR 0, 0
21:02:03.350 CFG: Saved, Count 93, Bytes 4096
[... 65 lines cut ...]

[... cut ...]
Switch1/Toggle1 should trigger output 1 and values of power and current should show on the left side in UI and calibration commands should also set for correct outputs i.e. PowerSet1 should influence Output 1 and PowerSet2 should influence Output 2.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Development branch without bluetooth Program Version 12.3.1.5(tasmota) Build Date & Time 2023-01-28T12:51:02 has the same issues (Even worse, as the power values sometimes reset to 0 and then come back again (but still mixed)_

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-01-29:
Your template is wrong.

This is the correct one:
```
{"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
```
Notice you have two equal Switch_N1!!!

Joylancer, 2023-01-29:
new template works for me.
need i2cdriver12 0
please update template https://templates.blakadder.com/shelly_plus_2PM.html

petermicuch, 2023-01-30:
@arendst thanks a lot for this. It does work now, I will request template update for https://templates.blakadder.com/shelly_plus_2PM.html.

Could you please tell me what does `,"CMND":"AdcParam1 2,10000,10000,3350"}` at the and of blakadder template means and what do you mean I do have two equal Switch_N1 (because in the UI I could see Switch_n and one had number 1 and the other 2).

closed 2023-01-30 by petermicuch