arendst/Tasmota #17824: ArtNet on esp32 is not available

dosipod, 2023-01-30:
[image]
Version  used  https://tasmota.github.io/install/
Program Version    12.3.1(tasmota)
Build Date & Time    2023-01-24T09:38:00
Core/SDK Version    2_0_5_3/v4.4.3

We have been told by the guys in discord to test on development as there could an issue with the release but we faced the same issue  when using the bins from here http://ota.tasmota.com/tasmota32/  . If you tested and that works  for you just link to bin used and no need for details

dosipod, 2023-01-30:
This bin seems to be the only one working at the monument  so might be others could be spared to effort to test 
http://ota.tasmota.com/tasmota32/tasmota32solo1.bin
[image]

Program Version	12.3.1.5(tasmota)
Build Date & Time	2023-01-28T12:51:02
Core/SDK Version	2_0_6/v4.4.3.20221227

For some reason you can not really tell which exact version is installed  but it is the solo1 bin

dosipod, 2023-01-30:
Just a correction that using the above bin did not work as the leds no longer works so might be hold on for a fix

s-hadinger, 2023-01-30:
I suggest that you start by explaining what you are trying to do, what is your setup and share the logs of Tasmota.

dosipod, 2023-01-31:
> I suggest that you start by explaining what you are trying to do, what is your setup and share the logs of Tasmota.

Oh might be the title of **ArtNet on esp32 is not available** is not clear ?!! If you read then the doc mentioned it is available by default but it is not . This has been confirmed on discord   https://discord.com/channels/479389167382691863/800751039283200050/1069306064572534804
So the answer is we want to use  ArtNet on esp32 with Tasmota as the doc stated  as we do with other firmware .

s-hadinger, 2023-01-31:
If you say so.

I just tried the precompiled Tasmota32 from https://ota.tasmota.com/tasmota32/release/tasmota32.bin and ArtNet is included in the firmware.

dosipod, 2023-02-01:
I am not saying so , the staff at tasmota discord as saying there is an issue with the release .No one said anything about compiling your own bin please try to take time to read the question and info provided as we toke the time to test that on multiple setups and provide bug report    . The release page  https://tasmota.github.io/install/ is what is refereed to in the docs  . If you tested from there then post the version tested . If you have ArtNet  working with your own compiled bin then also confirm that you have tested an app using ArtNet   with Tasmota and that it works 

At the end  the doc mentioned this is still experimental and would really help if you test the who steps else this is for sure a bug or two bugs

s-hadinger, 2023-02-01:
ArtNet is included by default in Tasmota32 precompiled binaries (for a few months at least). You don't need to compile your own version on ESP32. ArtNet is however not included in ESP8266 precompiled binaries.

If you flash the default version for ESP32 via https://tasmota.github.io/install/, I confirm that the binary includes ArtNet support.

dosipod, 2023-02-01:
Not true , prove evidence, technical details , version  and screen shots else what you are talking general things that do not really help and you are just adding noise to the post . Go to discord on the link in the post and then argue with the staff there that you do not see bug and again provide details if you can this time

arendst, 2023-02-01:
@dosipod you should have been very lucky to have a reply from our ArtNet programmer.

To advance your issue I suggest you provide all information in this issue and maybe he's willing to look into it again.

Rgrds, the Tasmota CEO/CIO.

dosipod, 2023-02-01:
We do not think we are that lucky , look here https://discord.com/channels/479389167382691863/800751039283200050/1069306064572534804 , This has being confirmed from your own team I believe hence we posted here . 
. We do not need any body resume and we have a very good team that being  using external control for many years now but unfortunately not with Tasmota  . As this is a new feature I think it would be best to provide details and address the issue rather then saying there is not issue

s-hadinger, 2023-02-01:
I was trying to help. You believe that what I say is "Not true" on a feature I developed, so be it. I won't argue.

This is my last message on this topic, you are on your own.

closed 2023-02-01 by Jason2866

dosipod, 2023-02-01:
No it is not it tasmota32 , so only our answers are correct  . Stop it all of you