arendst/Tasmota #17826: IP101 in tasmota esp32 does not work

davidgxj, 2023-01-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): esp32+ip101
- [ ] Tasmota binary firmware version number used: 12.3.1 (2_0_5_3/v4.4.3)[2023-01-24T09:38:00]
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA(https://tasmota.github.io/install/)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"ESP32-ip101test","GPIO":[1,1,5536,1,1,1,1,1,1,1,1,1,1,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
{"Module":{"0":"ESP32-ip101test"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"5536":"ETH POWER"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

RESULT = {"EthType":1}
RESULT = {"EthAddress":1}
RESULT = {"EthClockMode":0}
```
After restart 1, Ethernet cannot be started

If you use the pre-release-12.3 branch in github to compile. Try to comment out xdrv_ 82_ esp32_ 183-207 lines in the ethernet.ino file. Use the same configuration as before.

Ethernet IP101 works normally.

barbudor, 2023-01-30:
Do not use any of the pre-release branches. Those are internal

Have you tested with the latest development branch ?

davidgxj, 2023-01-30:
This bug has been around for a long time. The latest development branch has also been tested. The reason why I submit this bug is that I hope the Ethernet module can not change the xdrv part on the tasmota application

(Because tasmota ethernet is a very useful function. I have done the test of LAN8720A, and I am doing the test of rtl8201 recently.)

Jason2866, 2023-01-30:
This part of the code was added to get the LAN8720A reliable working, especially the Olimex PoE board. Removing is not an Option. It needs a different approach. Ideas or a PR is welcome.

davidgxj, 2023-01-31:
@Jason2866 
Thank you for your answer,
I understand what is happening now.

We track the Ethernet series products because they are more stable in IOT applications and convenient for application upgrading.

1. Can xdrv_ 82_ Ethernet.ino is more general. You can cancel the processing of different development boards (for example, WT32_ETH01)

2. Add the command of ETHPOE,
0, not enabled.
1，olimex poe
……

In this case, the universality of Ethernet will be strengthened. Most people who do not know self-compiled can quickly set it through template and Backlog.

I don't know whether such a proposal is feasible.

Jason2866, 2023-01-31:
No, not possible like you suggested. Tasmota has a more general approach and is not based on boards.

davidgxj, 2023-01-31:
Where can we learn, "Tasmota has a more general method."

Jason2866, 2023-01-31:
See the templates repo. There are the settings for every device. For example for the WT32-ETH01 https://templates.blakadder.com/wireless_tag_WT32-ETH01.html

davidgxj, 2023-01-31:
ths!

Before the problem occurred, the template method (wt32-eth01 and lilygo_T-Internet-POE) was used. The problem occurred after git submitted 2022-08-31 (Fix unstable ETH on Olimex POE). The ip101 doesn't work properly, so I have to comment out the code.

Jason2866, 2023-01-31:
The issue will be left open until hopefully! a fix is found to get all variants working.

Jason2866, 2023-01-31:
@davidgxj Which device you have which is not working? We have none with a IP101 PHY to do tests.

davidgxj, 2023-01-31:
For circuits using ip101, only

ESP32-Ethernet-Kit and the module in my hand. If you have time and need debugging, I can send you one.

[image]

s-hadinger, 2023-01-31:
> Before the problem occurred, the template method (wt32-eth01 and lilygo_T-Internet-POE) was used. The problem occurred after git submitted 2022-08-31 (Fix unstable ETH on Olimex POE). The ip101 doesn't work properly, so I have to comment out the code.

I suppose you mention this PR https://github.com/arendst/Tasmota/pull/16392

I didn't quite completely fix the Olimex POE problem although it seems to lower the chances of the problem to arise. I'm ok to revert it anyways, but I'm surprised it has any negative effect; especially after a power on. Doing a reset on an unconfigured GPIO shouldn't make any difference.

TD-er, 2023-01-31:
Maybe you need to add some time (400 ms) between pulling high the LAN pwr pin and calling ETH.begin()?
Or even make the call to pull this high before calling ETH.begin().

Which pins are used for this Ethernet adapter board?
Maybe some pins should be pulled down first before powering the LAN module or else it might be partially powered via pulled-up pins which are via a diode in the LAN chip connected to Vdd to protect those pins?

Jason2866, 2023-01-31:
@davidgxj we have removed the code part which breaks function of the PH IP101.

davidgxj, 2023-01-31:
@Jason2866 
tks!

Thank you for this adjustment. Darcy.gong and I will conduct a lot of tests on lan8720a, rtl8201 and other Ethernet in the near future, including corresponding POE support.

If you have better results, you will fork them.

Jason2866, 2023-01-31:
> Doing a reset on an unconfigured GPIO shouldn't make any difference.

The GPIO has a state before, by doing `gpio_reset_pin((gpio_num_t)GPIO_ETH_PHY_POWER);` you probably trigger something.

The delay mentioned from TD-er is probably needed, it is proposed fix in the PR https://github.com/espressif/arduino-esp32/pull/6188

davidgxj, 2023-01-31:
I have used the combination of esp32+ip101 for nearly a year. The code of 2022-08-31 has been stable since it was commented out.

Note: ip101 ref50m-esp32 gpio0

[image]

Jason2866, 2023-01-31:
@s-hadinger @TD-er What do you think about adding a `delay(400)` here https://github.com/tasmota/arduino-esp32/blob/Tasmota/207/libraries/Ethernet/src/ETH.cpp#L232
with a #ifdef around only for the LAN8720?

TD-er, 2023-01-31:
Why in the arduino-esp32 code?
Why not before making this call to ETH.begin()?

s-hadinger, 2023-01-31:
I think the key problem is to definitively fix the Olimex poe problem. I will leave it like that for a couple of weeks.

Note : I have old olimex poe that work perfectly well. The problem came only in a newer revision. I don't want to imply that Olimex is bad. They are awesome products

Jason2866, 2023-02-01:
Closing, since the PHY IP101 issue is solved.

closed 2023-02-01 by Jason2866