arendst/Tasmota #17850: ArtNet on esp32 is not available ( not solved even if closed , date is Feb 2nd, 2023 ) #17824

dosipod, 2023-02-01:
ArtNet on esp32 is not available ( not solved even if closed , date is Feb 2nd, 2023 ) #17824 
Do not close the issue until it is solved

arendst, 2023-02-01:
Tempting.....

Jason2866, 2023-02-01:
Since this PR https://github.com/arendst/Tasmota/pull/17104/files it is in the esp32 builds.
To be seen [here](https://github.com/s-hadinger/Tasmota/blob/5f0b92568ee88e77fd5c0f6b0a5dc6cb1c3769eb/tasmota/include/tasmota_configurations_ESP32.h#L562) for all [env] using the compile switch `-DFIRMWARE_TASMOTA32` in the `build_flags`
The solo1 env changed recently to the build `-DFIRMWARE_TASMOTA32`. Since that ArtNet support is there too.

closed 2023-02-01 by Jason2866

barbudor, 2023-02-01:
@dosipod There is no lack of profesional behavior
We are just facing a stuborn user that keeps on repeating the same thing without bringing any elements to the issue
We have templates that you are supposed to fill to provide information on you r setup but you just didn't care to fill it

Just flashed Tasmota release 12.3.1, configured the board and tried the commands as described in the doc:

[image]

dosipod, 2023-02-03:
Although we see no link to the fix mentioned and the staff on discord confirmed the issue   ,we will take your word for it 
and  many things to list that are not really professional. As the issue was closed before a solution is found we are no longer interested to use tasmota nor to talk to anyone who thinks he is god gift to the open source but does not know no one really care anymore  as we have plenty of options . Good luck to all of you