arendst/Tasmota #17854: Stored energy values get lost on power failure/no auto-save function working

ChrisDupu, 2023-02-02:
### PROBLEM DESCRIPTION
On a manual reboot of the controller the values for today/yesterday/total energy are stored within the flash and when the device comes back, the previous values are shown. Additionally the "Flash write count" value is increased.

According to https://github.com/arendst/Tasmota/discussions/17825#discussioncomment-4815553 the same should happen automatically on midnight. But this is not true. After running a device several days, the "Flash write count" does not change. Now when a power failure occurs, the values for today/yesterday/total energy fall back to some random, old value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): DLink WifI Wall Plug
- [X] Tasmota binary firmware version number used: 12.3.1.5
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
07:47:01.423 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:47:01.630 RSL: RESULT = {"Module":{"0":"Generic"}}
07:47:01.886 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2688":"HLW8012 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

07:47:37.976 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"","FriendlyName":["ShineBaby"],"Topic":"tasmota_D74227","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
07:47:37.980 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T10:59:31","StartupUTC":"2023-02-01T19:48:06","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2022-10-14T11:27:10","SaveCount":179,"SaveAddress":"FA000"}}
07:47:37.984 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1.5(tasmota)","BuildDateTime":"2023-01-28T12:48:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"345/699"}}
07:47:37.988 RSL: STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A002800000000","00000081","00006000","00004000","00000000"]}}
07:47:37.998 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":631,"Free":372,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
07:47:37.002 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D74227-0551","IPAddress":"","Gateway":"","Subnetmask":"255.255.255.0","DNSServer1":"","DNSServer2":"0.0.0.0","Mac":"","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:47:37.008 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-02-02T06:47:37","Local":"2023-02-02T07:47:37","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:19","Sunset":"17:48"}}
07:47:37.010 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:47:37.018 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-02-02T07:47:37","ENERGY":{"TotalStartTime":"2023-01-30T00:00:00","Total":10.841,"Yesterday":0.026,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
07:47:37.023 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-02-02T07:47:37","Uptime":"0T10:59:31","UptimeSec":39571,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":11,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:04"}}}

Note: for privacy reasons all sensitive information like BSSIDs, MACs and IPs have been removed here

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Wait until midnight

### EXPECTED BEHAVIOUR
The flash write count should increase

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-02-02:
With savedata 0 you're not storing anything except power changes.

Set savedata 1to fix your issue.

ChrisDupu, 2023-02-02:
The manual says, "savedata 1" causes a mode, where data are saved every second! This is definitely not what I want to have

arendst, 2023-02-02:
As I tried to explain in your other discussion the keyword here is SAVE CHANGES every second.

There only energy changes at midnight.