arendst/Tasmota #17869: Board definition doesn't match conditional compilation check

tonhuisman, 2023-02-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Boards definitions for `-DARDUINO_USB_CDC_ON_BOOT=0` don't match the conditional compilation `#ifdef ARDUINO_USB_CDC_ON_BOOT` in the source

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
n/a

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The use of `#if` instead of `#ifdef` in the source.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
n/a

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
n/a

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-02-03:
It is a platformio issue. Not working as it should. That's why we added `-DUSE_USB_CDC_CONSOLE` in boards definition.

Jason2866, 2023-02-09:
@tonhuisman Tbh, don't know what we should do with this issue.  Do you have a suggestion?

tonhuisman, 2023-02-09:
I haven't put much time into this yet, but I'll do some investigation to see if there is a solution/fix that would make this less confusing (mostly the `=0` part coinciding with `#ifdef`, as setting that define to 0 _does_ define it...)

tonhuisman, 2023-02-09:
That `#ifdef` check is used in:
- `lib/default/TasmotaSerial-3.6.0/src/TasmotaSerial.cpp` https://github.com/arendst/Tasmota/blob/development/lib/default/TasmotaSerial-3.6.0/src/TasmotaSerial.cpp#L231-L242
- `tasmota/tasmota_support/support.ino` https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support.ino#L1982-L1992

Both entries have comments suggesting possible unexpected behavior on ESP32, and IMHO it can't be excluded that this may be caused by having the wrong pre-processor check. Don't have the knowledge of Tasmota to actually test these intricacies, sorry.

I've seen weird/unexpected behavior of pre-processor directives in .ino files, but in .cpp files the `#if` and `#ifdef` will work differently.
Did try by replacing the `#ifdef` by `#if` in `support.ino` and added a few `#warning` messages to check, and the outcome _is_ different.

Jason2866, 2023-02-09:
@tonhuisman Thx for your thoughts/experiences with that. Now i do understand what you are alking about :-)
Keep you informed when finding "something"

tonhuisman, 2023-03-06:
unstale

void-spark, 2023-03-26:
Could this be why I'm having issues with SendSerial on esp32 at 9600 baud?
On a regular ESP32 with latest firmware (12.4.0.4), which I assume doesn't use some USB CDC, I get the message HWCDC supports 115200 bit/s only" when setting baudrate 9600.
Which seems to come from: https://github.com/arendst/Tasmota/blob/8e820e90c6db08be2e598b5da881281f57163b35/tasmota/tasmota_support/support.ino#L1981-L1993
Am I correct assuming it takes the wrong code path for a regular ESP32 here ?

arendst, 2023-03-26:
Well I guess if you load the wrong image on a ESP32 supporting hardware serial (with external USB/serial converter as the "older" ESP's do) then you can expect this message.

I don't see it on my "old" ESP32:
```
00:00:00.002-229/54 HDW: ESP32-D0WDQ6 
00:00:00.040-227/53 UFS: FlashFS mounted with 296 kB free
00:00:00.064-227/53 CFG: Loaded from File, Count 3763
00:00:00.073-227/53 QPC: Count 1
00:00:00.088-226/53 TFS: File 'mcp23x.dat' not found
00:00:00.089-226/53 MCP: Invalid template
00:00:00.093-226/53 ROT: Mode 1
00:00:00.256-221/52 BRY: Berry initialized, RAM used=3895 bytes
00:00:00.302-221/52 BRY: No 'preinit.be'
00:00:00.327-220/52 CFG: Energy loaded from file
00:00:00.333-220/52 SRC: Restart
00:00:00.338-220/51 Project tasmota - esp32d Version 12.4.0.4(theo)-2_0_7(2023-03-25T14:20:38)
00:00:00.339-220/52 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.557-219/51 BRY: No 'autoexec.be'
00:00:00.830-219/51 WIF: Attempting connection...
00:00:01.211-177/40 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as esp32d...
00:00:01.004-177/40 WIF: Attempting connection...
00:00:02.092-176/40 WIF: IPv4 192.168.2.115, mask 255.255.255.0, gateway 192.168.2.254
00:00:03.005-177/40 WIF: Connected
00:00:03.265-171/38 HTP: Web server active on esp32d with IP address 192.168.2.115
00:00:03.919-171/38 WIF: DNS resolved 'loghost' (192.168.2.24) in 13 ms
00:00:03.976-169/37 WIF: DNS resolved 'ntpserver1' (192.168.2.27) in 4 ms
00:00:03.993-169/37 RTC: UTC 2023-03-26T13:24:55, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
15:24:55.001-169/37 RTC: Synced by NTP
15:24:55.058-169/37 WIF: IPv6 Local fe80::9af4:abff:fe6d:2db4
15:24:55.251-169/37 MQT: Attempting connection...
15:24:55.259-169/37 WIF: DNS resolved 'mqttbroker' (192.168.2.24) in 7 ms
15:24:55.275-167/36 MQT: Connected
```

arendst, 2023-03-26:
Ah. bugger. You're right.

Let me check.....

arendst, 2023-03-26:
For the time being use ``SSerialSend`` instead. This will use another serial interface.

void-spark, 2023-03-26:
I was already trying that, but:
15:12:00.799 CMD: SBaudrate 9600
15:12:00.811 MQT: stat/tasmota_9BAD48/RESULT = {"Command":"Unknown"}
15:12:15.016 CMD: SBaudrate
15:12:15.027 MQT: stat/tasmota_9BAD48/RESULT = {"Command":"Unknown"}
15:16:33.381 CMD: SSerialConfig
15:16:33.392 MQT: stat/tasmota_9BAD48/RESULT = {"Command":"Unknown"}
15:32:56.760 CMD: SSerialSend
15:32:56.771 MQT: stat/tasmota_9BAD48/RESULT = {"Command":"Unknown"}

Is SSerial enabled on the default esp32 build?

arendst, 2023-03-26:
It should have been in there. I see on mine it's not too.....

arendst, 2023-03-26:
Did you configure SSerial using two SerialBridge GPIO's? If not, the commands won't work. The code is in the image but you'll have to enable it.

void-spark, 2023-03-26:
This is what I have set currently (documentation wasn't very clear here :) ):
[image]

Seeing your update, are SerBr Tx/Rx software serial specific? I saw them, but couldn't find the difference with the other ones, I thought they were maybe backwards compatibility

void-spark, 2023-03-26:
I'm seeing if I can build a custom version from master too :)
https://github.com/void-spark/Tasmota/actions/runs/4524924302
https://github.com/void-spark/Tasmota/commit/e9fd82a50e2d320de6da211d45442bdee5915074

arendst, 2023-03-26:
> Seeing your update, are SerBr Tx/Rx software serial specific? I saw them, but couldn't find the difference with the other ones, I thought they were maybe backwards compatibility

They are totally different. The ``SerBr`` ones use a different driver AND serial interface (UART) to interact. The ``Serial`` ones are the console serial interface and shouldn't be used on an ESP32 as there are more UARTS available to play with.

void-spark, 2023-03-26:
> They are totally different. The `SerBr` ones use a different driver AND serial interface (UART) to interact. The `Serial` ones are the console serial interface and shouldn't be used on an ESP32 as there are more UARTS available to play with.

Sounds like that explains some problems I'm having :)
So just to be sure, which gpio assignments should I use for esp32 'hardware' serial without touching the console uart but using the other uarts? :)

arendst, 2023-03-26:
SerBr

void-spark, 2023-03-26:
> SerBr

Thank you :)
With that and the 'serial' commands I get some responses with a lookpback wire atleast, progress! :)
But, I'm trying to do binary, and SSerialSend5 doesn't seem to work well (6 seems ok, but then I get very binary binary back, which I might be able to work with, but hex/decimal would be nicer :) ) :
```
17:00:36.185 CMD: SSerialSend6 2,82,5,6,2,0,83,3
17:00:36.192 MQT: stat/tasmota_9BAD48/RESULT = {"SSerialSend":"Done"}
17:00:36.330 MQT: tele/tasmota_9BAD48/RESULT = {"SSerialReceived":"RS"}
17:00:51.235 CMD: SerialSend5 0252050602005303
17:00:51.241 MQT: stat/tasmota_9BAD48/RESULT = {"SerialSend":"Done"}
17:01:04.736 CMD: SerialSend5 0252050602005303
17:01:04.742 MQT: stat/tasmota_9BAD48/RESULT = {"SerialSend":"Done"}
17:01:21.967 CMD: SerialSend5 02 52 05 06 02 00 53 03
17:01:21.974 MQT: stat/tasmota_9BAD48/RESULT = {"SerialSend":"Done"}
17:01:32.733 CMD: SSerialSend6 2,82,5,6,2,0,83,3
17:01:32.743 MQT: stat/tasmota_9BAD48/RESULT = {"SSerialSend":"Done"}
17:01:32.881 MQT: tele/tasmota_9BAD48/RESULT = {"SSerialReceived":"RS"}```

void-spark, 2023-03-26:
I guess my issue is now off topic for the OP one though, as the define doesn't affect the software serial :)

closed 2023-03-26 by Jason2866

void-spark, 2023-03-26:
Oh, this looks relevant: https://github.com/arendst/Tasmota/issues/18060 I guess I'll move there.
Looks like I need to use 254 serialdelimiter, which isn't documented yet :)

void-spark, 2023-03-26:
And last comment, Above I was using SerialSend5 instead of SSerialSend5, that's why that command didn't work.. XD
So using the correct command, with 254 as serial delimiter, should solve all my issues :)

PR #18269 (2023-03-26): rm `-Dxxxx=0`settings
## Description:

Fix #17869

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.7
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests.  [... cut]