arendst/Tasmota #17876: Berry Script: No return for light.get() ?

ozett, 2023-02-04:
### PROBLEM DESCRIPTION
i started to use berry script and expect to get a return for `light.get()`.
maybe my misreading of the documentation - or a bug?
maybe berry´s light is not applicable for a LED-configred GPIO?

docu: 
<kbd>[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:57:30.785 CMD: Backlog Template; Module; GPIO 255
10:57:30.812 MQT: stat/tasmota_3DF6D4/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
10:57:31.039 MQT: stat/tasmota_3DF6D4/RESULT = {"Module":{"1":"ESP32-DevKit"}}
10:57:31.294 MQT: stat/tasmota_3DF6D4/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"288":"Led1"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
10:58:18.202 CMD: Backlog Rule1; Rule2; Rule3
10:58:18.232 MQT: stat/tasmota_3DF6D4/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
10:58:18.458 MQT: stat/tasmota_3DF6D4/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
10:58:18.659 MQT: stat/tasmota_3DF6D4/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:58:39.520 CMD: Status 0
10:58:39.530 MQT: stat/tasmota_3DF6D4/STATUS = {"Status":{"Module":1,"DeviceName":"lolin-lite-flashled","FriendlyName":["lolin-lite-flashled"],"Topic":"tasmota_3DF6D4","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:58:39.540 MQT: stat/tasmota_3DF6D4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:19:42","StartupUTC":"2023-02-04T09:38:57","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-02-03T20:01:13","SaveCount":32}}
10:58:39.546 MQT: stat/tasmota_3DF6D4/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2023-01-24T09:38:00","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6-V3 rev.3","CR":"411/699"}}
10:58:39.551 MQT: stat/tasmota_3DF6D4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fo46io",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
10:58:39.569 MQT: stat/tasmota_3DF6D4/STATUS4 = {"StatusMEM":{"ProgramSize":1479,"Free":1400,"Heap":138,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D43C482D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
10:58:39.582 MQT: stat/tasmota_3DF6D4/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3DF6D4-5844","IPAddress":"192.168.14.15","Gateway":"192.168.14.9","Subnetmask":"255.255.255.0","DNSServer1":"192.168.14.9","DNSServer2":"8.8.8.8","Mac":"B4:8A:0A:3D:F6:D4","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:58:39.593 MQT: stat/tasmota_3DF6D4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.14.204","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3DF6D4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:58:39.606 MQT: stat/tasmota_3DF6D4/STATUS7 = {"StatusTIM":{"UTC":"2023-02-04T09:58:39","Local":"2023-02-04T10:58:39","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:16","Sunset":"17:51"}}
10:58:39.619 MQT: stat/tasmota_3DF6D4/STATUS10 = {"StatusSNS":{"Time":"2023-02-04T10:58:39"}}
10:58:39.632 MQT: stat/tasmota_3DF6D4/STATUS11 = {"StatusSTS":{"Time":"2023-02-04T10:58:39","Uptime":"0T00:19:42","UptimeSec":1182,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":44},"Wifi":{"AP":1,"SSId":"fo46io","BSSId":"F4:92:BF:A1:06:DD","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
10:58:39.520 CMD: Status 0
10:58:39.530 MQT: stat/tasmota_3DF6D4/STATUS = {"Status":{"Module":1,"DeviceName":"lolin-lite-flashled","FriendlyName":["lolin-lite-flashled"],"Topic":"tasmota_3DF6D4","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:58:39.540 MQT: stat/tasmota_3DF6D4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:19:42","StartupUTC":"2023-02-04T09:38:57","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-02-03T20:01:13","SaveCount":32}}
10:58:39.546 MQT: stat/tasmota_3DF6D4/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2023-01-24T09:38:00","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6-V3 rev.3","CR":"411/699"}}
10:58:39.551 MQT: stat/tasmota_3DF6D4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fo46io",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
10:58:39.569 MQT: stat/tasmota_3DF6D4/STATUS4 = {"StatusMEM":{"ProgramSize":1479,"Free":1400,"Heap":138,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D43C482D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24, [... cut ...]
type in berry console: light.get()

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
a return value like its happening with the power on berry-console
tasmota.get_power()
-> https://tasmota.github.io/docs/Berry/#lights-and-relays

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-04:
A led is not a light

Lights are:
- relay
- pwm
- ws2812

Led is just an indicator

ozett, 2023-02-04:
Yeah, just tried to switch to PWM and i got a return.

do you think thats a common understanting that `LED` is **not** a `LIGHT` ?

anyways . big thanks for the lightningspeed response. will dig further into berry... 😄 

<kbd>[image]

barbudor, 2023-02-04:
I believe the documentation provides quite some hints in what is Light for Tasmota
https://tasmota.github.io/docs/Lights/

Glad that solve your problem

For help request and general how-to, please start a Discussion
Issues should focus on qualified bugs
Thanks for your understanding