arendst/Tasmota #17879: Exception when setting SHUTTERPOSITION via MQTT

lngr, 2023-02-04:
### PROBLEM DESCRIPTION
 
Tasmota on a Shelly 2.5 device in Shutter mode restarts with an exception everytime I set SHUTTERPOSITION / SHUTTERPOSITION1 via MQTT.  It does *not* restart, if I type the command manually in the browser Console OR if I issue the command via HTTP. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [x] Tasmota binary firmware version number used: 12.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:00:45.020 CMD: Backlog Template; Module; GPIO 255
15:00:45.098 MQT: stat/esszimmer-rollladen/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
15:00:45.332 MQT: stat/esszimmer-rollladen/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
15:00:45.587 MQT: stat/esszimmer-rollladen/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:01:39.241 MQT: stat/esszimmer-rollladen/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"ON Time#Minute=60 do RESTART 1 ENDON"}}
15:01:39.459 MQT: stat/esszimmer-rollladen/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:01:39.660 MQT: stat/esszimmer-rollladen/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:02:05.053 MQT: stat/esszimmer-rollladen/STATUS = {"Status":{"Module":0,"DeviceName":"shelly2-esszimmer-rolladen","FriendlyName":["esszimmer-rolladen-up","esszimmer-rolladen-down"],"Topic":"esszimmer-rollladen","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:02:05.064 MQT: stat/esszimmer-rollladen/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:11:44","StartupUTC":"2023-02-04T13:50:21","Sleep":50,"CfgHolder":4617,"BootCount":102,"BCResetTime":"2021-05-02T11:39:57","SaveCount":2065,"SaveAddress":"F6000"}}
15:02:05.075 MQT: stat/esszimmer-rollladen/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"475/699"}}
15:02:05.083 MQT: stat/esszimmer-rollladen/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GW18",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A002800000000","40000080","00006000","00004000","00000000"]}}
15:02:05.099 MQT: stat/esszimmer-rollladen/STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
15:02:05.111 MQT: stat/esszimmer-rollladen/STATUS5 = {"StatusNET":{"Hostname":"esszimmer-rollladen-7484","IPAddress":"192.168.11.250","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.11.1","DNSServer2":"0.0.0.0","Mac":"8C:AA:B5:4C:BD:3C","Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
15:02:05.120 MQT: stat/esszimmer-rollladen/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.11.11","MqttPort":1883,"MqttClientMask":"shelly2-esszimmer-rollladen","MqttClient":"shelly2-esszimmer-rollladen","MqttUser":"gw18","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:02:05.132 MQT: stat/esszimmer-rollladen/STATUS7 = {"StatusTIM":{"UTC":"2023-02-04T14:02:05","Local":"2023-02-04T15:02:05","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T02:00:00","Timezone":99,"Sunrise":"08:07","Sunset":"17:27"}}
15:02:05.141 MQT: stat/esszimmer-rollladen/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:02:05.155 MQT: stat/esszimmer-rollladen/STATUS10 = {"StatusSNS":{"Time":"2023-02-04T15:02:05","ANALOG":{"Temperature":42.2},"ENERGY":{"TotalStartTime":"2021-05-02T11:39:57","Total":0.707,"Yesterday":0.001,"Today":0.006,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"TempUnit":"C"}}
15:02:05.168 MQT: stat/esszimmer-rollladen/STATUS11 = {"StatusSTS":{"Time":"2023-02-04T15:02:05","Uptime":"0T00:11:44","UptimeSec":704,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"GW18","BSSId":"6A:D7:9A:19:2D:92","Channel":11,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:05"}}}
15:02:05.181 MQT: stat/esszimmer-rollladen/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":450,"Close":450,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

This works:

```
$ curl -D - "http://192.168.11.250/cm?cmnd=SHUTTERPOSITION+50"
HTTP/1.1 200 OK
Content-Type: application/json
Server: Tasmota/12.3.1 (ESP8266EX)
Cache-Control: no-cache, no-store, must-revalidate
Pragma: no-cache
[... 80 lines cut ...]
15:06:19.211 ADE: ACCMODE 0x2D2800, VRMS 6081277, 6081429, Period 4471, 4472, IRMS 1778, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
15:06:19.609 CFG: Saved to flash at FB, Count 2068, Bytes 4096
15:06:20.177 MQT: tele/esszimmer-rollladen/STATE = {"Time":"2023-02-04T15:06:20","Uptime":"0T00:00:10","UptimeSec":10,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"GW18","BSSId":"6A:D7:9A:19:2D:92","Channel":11,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:06:20.193 MQT: tele/esszimmer-rollladen/SENSOR = {"Time":"2023-02-04T15:06:20","ANALOG":{"Temperature":43.3},"ENERGY":{"TotalStartTime":"2021-05-02T11:39:57","Total":0.709,"Yesterday":0.001,"Today":0.007,"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":90,"Direction":0,"Target":90,"Tilt":0},"TempUnit":"C"}
15:06:20.213 ADE: ACCMODE 0x2D2800, VRMS 6079721, 6083512, Period 4471, 4471, IRMS 1778, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0
15:06:21.212 ADE: ACCMODE 0x2D2800, VRMS 6073458, 6077113, Period 4472, 4472, IRMS 1778, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
15:06:22.212 ADE: ACCMODE 0x2D2800, VRMS 6068617, 6072416, Period 4471, 4471, IRMS 1778, 1773, WATT 0, 0, VA 27, 27, VAR 0, 0
[... cut ...] 0]},"Shutter1":{"Position":90,"Direction":0,"Target":90,"Tilt":0},"TempUnit":"C"},"ver":1} (retained)

This happens on all of my devices (tried it on 5 shelly 2.5 with the same configuration)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

MQTT Command should work the same way as HTTP or console command.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

barbudor, 2023-02-04:
I can't reproduce neither from http nor mqtt

Could you provide more details on your configuration ?

Could you run the same test, off AC and connected to a serial adapter to get the logs from serial ?

```
17:12:37.307 HTP: Command
17:12:37.312 SRC: WebCommand from 192.168.168.1
17:12:37.316 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 2, Pld 40, Data '40'
17:12:37.320 SRC: Shutter
17:12:37.327 MQT: stat/test/POWER2 = {"POWER2":"ON"}
17:12:37.333 MQT: stat/test/POWER2 = ON (retained)
17:12:37.336 SHT: Shtr1 Real 240000, Start 240000, Stop 160000, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
17:12:37.343 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":60,"Direction":-1,"Target":40,"Tilt":0}}
17:12:37.472 SHT: Shtr1 Real 236000, Start 240000, Stop 160000, Dir -1, Delay 0, Rtc 0.15 [s], Freq 0, PWM 0, Tilt 0
17:12:37.481 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":59,"Direction":-1,"Target":40,"Tilt":0}}
17:12:38.484 SHT: Shtr1 Real 196000, Start 240000, Stop 160000, Dir -1, Delay 0, Rtc 1.15 [s], Freq 0, PWM 0, Tilt 0
17:12:38.493 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":49,"Direction":-1,"Target":40,"Tilt":0}}
17:12:39.387 SHT: Stop 1 Mode 0 time 41
17:12:39.389 SRC: Shutter
17:12:39.398 MQT: stat/test/POWER2 = {"POWER2":"OFF"}
17:12:39.403 MQT: stat/test/POWER2 = OFF (retained)
17:12:39.406 SHT: Shtr1 Real 160000, Start 160000, Stop 160000, Dir 0, Delay 0, Rtc 2.05 [s], Freq 0, PWM 0, Tilt 0
17:12:39.413 MQT: stat/test/SHUTTER1 = 40 (retained)
17:12:39.418 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":40,"Direction":0,"Target":40,"Tilt":0}}
17:12:39.600 CFG: Saved to flash at F9, Count 27, Bytes 4096
17:13:23.941 SRC: MQTT
17:13:23.944 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 2, Pld 70, Data '70'
17:13:23.948 SRC: Shutter
17:13:23.954 MQT: stat/test/POWER1 = {"POWER1":"ON"}
17:13:23.957 MQT: stat/test/POWER1 = ON (retained)
17:13:23.960 SHT: Shtr1 Real 160000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
17:13:23.967 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":40,"Direction":1,"Target":70,"Tilt":0}}
17:13:24.493 SHT: Shtr1 Real 169000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 0.50 [s], Freq 0, PWM 0, Tilt 0
17:13:24.499 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":42,"Direction":1,"Target":70,"Tilt":0}}
17:13:25.497 SHT: Shtr1 Real 189000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 1.50 [s], Freq 0, PWM 0, Tilt 0
17:13:25.503 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":47,"Direction":1,"Target":70,"Tilt":0}}
17:13:26.502 SHT: Shtr1 Real 209000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 2.50 [s], Freq 0, PWM 0, Tilt 0
17:13:26.508 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":52,"Direction":1,"Target":70,"Tilt":0}}
17:13:27.476 SHT: Shtr1 Real 229000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 3.50 [s], Freq 0, PWM 0, Tilt 0
17:13:27.482 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":57,"Direction":1,"Target":70,"Tilt":0}}
17:13:28.455 SHT: Shtr1 Real 248000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 4.45 [s], Freq 0, PWM 0, Tilt 0
17:13:28.461 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":62,"Direction":1,"Target":70,"Tilt":0}}
17:13:29.457 SHT: Shtr1 Real 268000, Start 160000, Stop 280000, Dir 1, Delay 0, Rtc 5.45 [s], Freq 0, PWM 0, Tilt 0
17:13:29.462 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":67,"Direction":1,"Target":70,"Tilt":0}}
17:13:30.058 SHT: Stop 1 Mode 0 time 122
17:13:30.060 SRC: Shutter
17:13:30.066 MQT: stat/test/POWER1 = {"POWER1":"OFF"}
17:13:30.071 MQT: stat/test/POWER1 = OFF (retained)
17:13:30.074 SHT: Shtr1 Real 281000, Start 281000, Stop 280000, Dir 0, Delay 0, Rtc 6.10 [s], Freq 0, PWM 0, Tilt 0
17:13:30.081 MQT: stat/test/SHUTTER1 = 70 (retained)
17:13:30.086 MQT: stat/test/SHUTTER = {"Shutter1":{"Position":70,"Direction":0,"Target":70,"Tilt":0}}
17:13:30.569 CFG: Saved to flash at F8, Count 28, Bytes 4096
```

lngr, 2023-02-05:
> I can't reproduce neither from http nor mqtt

HTTP does work for me too, only MQTT causes the exceptions.

> Could you provide more details on your configuration ?

Sure, if you need any specifics, let me know.  In general, it's basically just a Shelly 2.5 with stock firmware in Shutter mode connected to the IOBroker [MQTT broker](https://github.com/ioBroker/ioBroker.mqtt) which is based on [MQTT.js](https://github.com/mqttjs/MQTT.js).  I use this configuration on all devices:

```
TIMEZONE 99
TimeDST 0,0,3,1,2,120
TimeSTD 0,0,10,1,2,60
latitude 51.3496951
longitude 6.4225623
WifiConfig 5
RULE 1
RULE1 ON Time#Minute=60 do RESTART 1 ENDON
SetOption80 1 
Interlock 1,2
Interlock ON
PowerOnState 0
Shutterrelay1 1
ShutterOpenDuration 30
ShutterCloseDuration 30
```

Oddly enough, the very same devices worked flowlessly before over MQTT when *not* configured in shutter mode, i.e., when I controlled POWER1/POWER2 over MQTT (with Interlock on 1,2), via sending to `cmnd/esszimmer-rollladen/POWER1` or `POWER2`.  I control several such Shelly 2.5 devices over iobroker/mqtt without problems.  I only recently changed their configuration to Shutter Mode to get more fine grained control over the shutter position, and that's when the problems began.

However, I just set `SetOption80 0` followed by a `Restart 1`, and now even `POWER1`/`POWER2` cause an exception.

> Could you run the same test, off AC and connected to a serial adapter to get the logs from serial ?

Unfortunately not, they are in the wall and I currently cannot remove them.  I might be able to do that as last resort.

Can you see from the exception stacktrace what function is causing the exception?

```15:06:16.107 MQT: tele/esszimmer-rollladen/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40254ecd","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40210261","4022d626","40290a83","4024ae88","4010193b","40101bc3","401054fa","4020e8d4","4010193b","4028b95e","4028b957","4020e905","4022d880","4025fd14","4025b3f2","402433ae","4024347e","4024a3f0","4024a541","40251c98","4024a91c","4010122a","4022b02a","4021fc66","40221d73","4023f84c","4023f8d0","40100aa0","4024eeb4","40101eb9"]},"BootCount":103}}
```

lngr, 2023-02-05:
Update: I think I have found the cause.  I connected the device to a different MQTT broker (Mosquitto), and here it works.

On mosquitto, I noticed `mosquitto_sub` to output:

    Client mosqsub|8478-mestus sending SUBSCRIBE (Mid: 1, Topic: cmnd/esszimmer-rollladen/SHUTTERPOSITION, QoS: 0)`

On the iobroker MQTT, the output is:

    Client mosqsub|8307-mestus received PUBLISH (d0, q0, r1, m0, 'cmnd.esszimmer-rollladen.SHUTTERPOSITION', ... (2 bytes))

i.e., it uses `.` instead of `/` for some reason.  I'm pretty sure that causes a array overflow in tasmota when parsing the message (it probably splits the string by `/`, I'm going to look this up.

barbudor, 2023-02-05:
I indeed noticed that some of your logs where showing some dots instead of slash but I thought it was your login tool

There are a few users out there using ioBroker with Tasmota so there should be a way for them to work together.
Definitively Tasmota expect the slash as separator

lngr, 2023-02-05:
After further investigation, I'm sure the underlying cause is a bug in the IOBroker MQTT-Broker that sends a message to a subscriber (Tasmota) for a topic that the subscriber did never subscribe for and the topic does not contain a `/`.  Maybe here:

https://github.com/arendst/Tasmota/blob/96b66225d2055b22aee6f70c03be0947b5ead118/tasmota/tasmota_support/support_command.ino#L388

Anyways, I'm closing this, since a non-compliant MQTT broker is not Tasmota's fault.  Thanks for the help!

closed 2023-02-05 by lngr

arendst, 2023-02-05:
Thx for investigating. The least I can do is bail out if no "/" is detected. I'll make it that way.

arendst, 2023-02-05:
You might want to try the fix as I cannot test it.

lngr, 2023-02-07:
Sure, I'll try it next weekend.

lngr, 2023-02-11:
Looks good
[image]