arendst/Tasmota #17892: Changing the flash/save frequency of Energy Total in Tasmota

kubedzero, 2023-02-06:
Hi there, I have a Sonoff S31 switch connected to a space heater. I move it around the house, unplugging it from mains power and plugging it back in a few times a day. 

I have a script collecting the "Energy Total" value in kWh every 60 seconds, which I use for energy visualization. 

I have an issue where the Energy Total seems to "reset" back to a lower value than the last reported value. I imagine Tasmota is storing Energy Total in RAM and writing it to non-volatile memory every once in a while.

See picture for a visualization of Energy Total over a two day period.

[image]

Is that save frequency documented anywhere, and is there any way to change it? I'd really love to avoid losing the Energy Total readings every time I unplug the device. 

Thanks!

kubedzero, 2023-02-06:
I found my answer in https://github.com/arendst/Tasmota/discussions/17825#discussioncomment-4815553 and https://github.com/arendst/Tasmota/issues/16791#issuecomment-1313584774

I can use Rules https://tasmota.github.io/docs/Commands/#rules to manually run the `savedata` command https://tasmota.github.io/docs/Commands/#management on a more frequent cadence 

`Rule1 on time#minute|60 do savedata endon` is an example Rule that apparently will save updated values every hour. I don't really get how that works, since `savedata 1` is apparently the default and as far as I can tell this is just calling the command with the default. Is there a side effect of calling this command that it works right away as well? Anyway, I am testing this now to confirm it works as expected.

arendst, 2023-02-06:
Yes. The side effect is that it requests drivers to save their changes to the settings RAM area and if there are any changes updates the flash accordingly.